MarlinFirmware/Marlin #27453: [BUG] MARLINUI doesn't change language with MKS_TS35_V2_0

anomalchik, 2024-10-03:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

UI language does not change

### Actual behavior

Running with English language.

### Steps to Reproduce

1. #define MKS_TS35_V2_0
2. #define TFT_COLOR_UI
3. #define DISPLAY_CHARSET_HD44780 CYRILLIC
4. #define LCD_LANGUAGE ru
5. uncomment CONFIG_EXPORT in Configuration_adv
6. compile firmware
7. check exported configuration and you will see that the DISPLAY_CHARSET_HD44780 and LCD_LANGUAGE parameters have not been exported

### Version of Marlin Firmware

bugfix-2.1.x (2024-09-29)

### Printer model

Custom

### Electronics

BOARD_MKS_ROBIN_NANO_V3

### LCD/Controller

MKS_TS35_V2_0

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[config.zip](https://github.com/user-attachments/files/17241884/config.zip)

ellensp, 2024-10-03:
HD44780 is for character LCD displays so this is not used
#define DISPLAY_CHARSET_HD44780

but your correct this is not working.
#define LCD_LANGUAGE ru

Marlin has this line which incorrectly removes LCD_LANGUAGE

```cpp
#if NONE(HAS_MARLINUI_U8GLIB, HAS_TFT_LVGL_UI, DGUS_LCD_UI_E3S1PRO)
  #undef LCD_LANGUAGE
#endif
```

update Marlin/src/inc/Conditionals-2-LCD.h with this code seems like a good fix.

```cpp
#if NONE(HAS_MARLINUI_U8GLIB, HAS_TFT_LVGL_UI, TFT_COLOR_UI, DGUS_LCD_UI_E3S1PRO)
  #undef LCD_LANGUAGE
#endif
```

ellensp, 2024-10-03:
I have created a PR to fix this going forward.

closed 2024-10-03 by ellensp