MarlinFirmware/Marlin #27460: [BUG] To fix FSMC in STM32F1xx clones it has been broken in the original STM32F1xx

SMH17, 2024-10-07:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I faced issues with DMA of an STM32F1xx based boards and FSMC with latest version of Marlin, after investigating the issue I have seen that [this code](https://github.com/jmz52/Marlin/commit/1940418bbe89d07863ed05c6cddb1edf285a5a31) has been added, this change the DMA channel used by FSMC(that was correctly set to DMA2_Channel1 before) and it seems intended to fix the analogous issue on GD32Fxx clones, but since GD32 clones aren't exactly drop in replacements for STM32F1xxx and there are [design differences](https://easyelectronics.ru/img/ARM_kurs/STvsGD/Migration/compatibility%20sumup%20between%20GD32%20and%20STM32_V2.0.pdf), attempting to fix the issue on GD32Fxxx reckless merging code in STM32 HAL without any proper distinction, inevitably will cause problems on the regular STM32Fxxx microcontrollers and hard to debug mess to codebase. 

I haven't checked if similar approach has been used also for other GD32 MCUs. 

### Version of Marlin Firmware

2.1.x

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

thinkyhead, 2024-10-07:
I went ahead and reverted that particular change, which may leave a conflicting DMA in place between FSMC and some other functions. If there is real interest in sorting this out and figuring out a good way to ensure there are never conflicting or shared DMA channels, let's aim for that. The solution was originally suggested in #27385 by @konradmb.

SMH17, 2025-01-11:
keep it opened

closed 2025-04-28 by github-actions[bot]