MarlinFirmware/Marlin #27465: Russia has banned and blocked Discord:    Russian users can get top priority by posting in this thread.

Roxy-3D, 2024-10-09:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Discord is Marlin's primary user support mechanism.     Because Russia has banned and blocked Discord within Russia, all of our Russian users are left with no support.

I don't expect we will get many problems reported by our Russian users.  After all...   They are a very smart and independent bunch of people.     But with that said...

If you are Russian....   And you need help with a Marlin issue....     Please post your problem here.   I will make sure it gets the attention it deserves.

### Version of Marlin Firmware

.

### Printer model

.

### Electronics

.

### LCD/Controller

.

### Other add-ons

.

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Slic3r

### Host Software

Pronterface

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

thisiskeithb, 2024-10-09:
I don't think we need a dedicated Issue for this since we have several other resources for users including places to ask for help:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube

Similar to users in China, Russians are using VPNs to get around blocks and can still access Discord.

closed 2024-10-09 by Roxy-3D

thisiskeithb, 2024-10-09:
Support links are also available in the README: https://github.com/MarlinFirmware/Marlin?tab=readme-ov-file#marlin-support