MarlinFirmware/Marlin #27475: [BUG] Y-Axis Motor Shake with ZV Input Shaping Enabled Only on Y-Axis

johnnychang924, 2024-10-13:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

When ZV input shaping is enabled only on the Y-axis (with the X-axis shaping disabled), the Y-axis motor experiences strong shaking and skips steps during movement. The issue does not occur when input shaping is enabled on both axes or when it is completely disabled.

### Bug Timeline

new bug

### Expected behavior

Smooth motion of the Y-axis with accurate positioning when input shaping is applied only to the Y-axis.

### Actual behavior

The Y-axis motor shakes violently and skips steps, resulting in significant inaccuracies in positioning. This makes the printer unusable.

### Steps to Reproduce

1. Configure Marlin firmware to enable ZV input shaping only on the Y-axis:
```INPUT_SHAPING_X_AXIS``` disabled
```INPUT_SHAPING_Y_AXIS``` enabled

2. Start any movement or home the printer.

### Version of Marlin Firmware

2.1.2.4

### Printer model

Ender-3 Pro with switchwire mod

### Electronics

creality 4.2.2

### LCD/Controller

cr-10 stock

### Other add-ons

I use stock extruder motor as y-axis motor.

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

I am using a CoreXZ 3D printer, and in Marlin, CoreXZ configurations do not support input shaping on the X-axis. Therefore, I need to use input shaping only on the Y-axis.
[Configuration.zip](https://github.com/user-attachments/files/17356063/Configuration.zip)

SMH17, 2024-10-14:
I'm facing this critical issue too, and mostly the shift happens on Y axis although some prints show problem on X at same time, but it happens with input shaping enabled on both axis. 

It seems working normal then suddenly it goes crazy: a stepper rattling noise with a shift to a significantly lower than normal Y value, continuing to print "normally" in a then heavily misaligned position without any gradual shifting till the next recurrence of the problem. 

- The problem represents itself also with low print speed and without mechanical problems as loose belts etc.
- Reducing or bumping stepper voltage doesn't seem to have any effect on the frequency of the issue.
- Enabling or disabling S_CURVE_ACCELERATION doesn't affect the issue too.

I haven't tried disabling STEALTHCHOP yet, that is the next in my list of suspects.

I have seen many issues related to shifting that are closed due to inactivity, without any related fix, I think the autoclosing of issues like this should be prevented till a proper determining of the cause of the problem and fix, otherwise these remains hidden and not properly addressed.

moriamoria, 2024-10-19:
Hello,

I have connected a scope to Z set pin and Z2 step pin. The result is clear

With // #define EDGE_STEPPING deactivated, I have the same signal on Z and Z2  (on skr 1.3 motherboard E1)

[image]

With #define EDGE_STEPPING activated, I have only a step signal on Z2 ! 

[image]

That explain why the Z2 stepper force on Z.

Now I have the root problem, tel me if you want me to do other test.

But Dr Murphy is there, now when I print with  // #define EDGE_STEPPING deactivated, my print stop after the brim and the begining of the print, I think I will open a new issue, not merging all problem together.

Regards to all

ellensp, 2024-10-19:
Please test the [`bugfix-2.1.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

there is also #define INPUT_SHAPING_Z

johnnychang924, 2024-10-20:
As I am unable to access my 3D printer right now, I will perform further tests next weekend and update you on the results.

moriamoria, 2024-10-20:
Hello,

I have change the branch to 2.1.2.4, I have enable #define SQUARE_WAVE_STEPPING

And I have same result, on Z2 step signal, nothing on Z

[image]

without define SQUARE_WAVE_STEPPING I have good signal on Z and Z2

[image]

What do you want me to do ?

Regards

moriamoria, 2024-10-20:
Hello again,

I notice when I compile marlin, i have following comment 👍 

Marlin\src\inc\Warnings.cpp:206:6: warning: #warning "Note: Auto-assigned Z2 STEP/DIR/ENABLE_PINs to unused En_STEP/DIR/ENABLE_PINs. (Define NO_AUTO_ASSIGN_WARNING to suppress this warning.)" [-Wcpp]
  206 |     #warning "Note: Auto-assigned Z2 STEP/DIR/ENABLE_PINs to unused En_STEP/DIR/ENABLE_PINs. (Define NO_AUTO_ASSIGN_WARNING to suppress this warning.)"

in the configuration i have define following configuration 👍 
#define X_DRIVER_TYPE  TMC2209

#define Z_DRIVER_TYPE  TMC2209

#define Z2_DRIVER_TYPE TMC2209

#define E0_DRIVER_TYPE TMC2209

Perhaps I have to declare Z2 somewhere else ?!

Regards to all

ellensp, 2024-10-21:
@moriamoria your issue seems completely different to the originals posters issue.

You MUST test bugfix 2.1.x before making bug reports.  As that is the most updated code.

The warning is just that a warning, its is telling you that Z2 in firmware will use the unused Extruder stepper driver of your controller (E1 in most cases).

moriamoria, 2024-10-22:
Hello,

What I have found is that issue is present on all bugfix-2.1.x branch, and that also is was present on 2.1.2.4 branch.

I agreed that the issue name is incorrect, I will create a new issue.

If I can help to solve this issue, I can do my best, but it is not easy to explorer the souce code.

Regards

closed 2024-10-26 by johnnychang924