MarlinFirmware/Marlin #27477: [BUG] Compile of G61.cpp fails with zero extruders

Kramax, 2024-10-15:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Compilation of the unit G61.cpp fails if no extruders are configured. This is because a single line of code is not surrounded by the "  #if HAS_EXTRUDERS" cpp command. The function GcodeSuite::G61() has this line:
`float epos = stored_position[slot].e;`
which needs to have the guard of HAS_EXTRUDERS. 

### Bug Timeline

New bug introduced on July 26, 2024 by commit 78e111a4d04b3e64077da769c03178d6185f69f7 for issue #27281.

### Expected behavior

I expect the compile to succeed when no extruders are defined.

### Actual behavior

The compile fails with unknown symbols since the "e" field of the saved position struct is missing.

### Steps to Reproduce

1. Create a Configuration.h with EXTRUDERS defined to be 0.
2. Make sure SAVED_POSITIONS in Configuration_adv.h is greater than 0.
3.  Try to build
4.  The build fails

### Version of Marlin Firmware

Bugfix-2.1.x

### Printer model

NA

### Electronics

BTT Octopus V1.1

### LCD/Controller

None

### Other add-ons

None

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration.zip](https://github.com/user-attachments/files/17372078/Configuration.zip)

Kramax, 2024-10-15:
This can be fixed with this patch:
diff --git a/Marlin/src/gcode/feature/pause/G61.cpp b/Marlin/src/gcode/feature/pause/G61.cpp
index 08b4a99b7d..c4d385b4ae 100644
--- a/Marlin/src/gcode/feature/pause/G61.cpp
+++ b/Marlin/src/gcode/feature/pause/G61.cpp
@@ -75,7 +75,10 @@ void GcodeSuite::G61(int8_t slot/*=-1*/) {

   // No XYZ...E parameters, move to stored position

+  #if HAS_EXTRUDERS
   float epos = stored_position[slot].e;
+  #endif
+
   if (!parser.seen_axis()) {
     DEBUG_ECHOLNPGM(STR_RESTORING_POSITION, slot, " (all axes)");
     // Move to the saved position, all axes except E

ellensp, 2024-11-09:
PR created

closed 2024-11-09 by ellensp