MarlinFirmware/Marlin #27481: [BUG] Possibly incorrect pin mapping for BLTouch pins and Filament Runout pin on Creality Ender 2 Pro v2.4.S4_170 (HC32f460kcta) board

calorifer, 2024-10-17:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

The pin definitions for SERVO0_PIN, Z_MIN_PROBE_PIN and FIL_RUNOUT_PIN are possibly incorrect in pins_CREALITY_ENDER2P_V24S4.h. Pin PB0 is also used twice for Z_STOP_PIN and SERVO0_PIN.

A detailed pinmap with images is here: https://gist.github.com/SteveGotthardt/33bdef27d1430f62b4fee5d723d0cde9

In that link pin PA4 is not specifically identified as the filament runout pin but the port they identified is the filament runout port (it's not specifically labeled as the filament runout port on the board)

Pin assignments that would work out of the box:
SERVO0_PIN PB1 (it's set to PB0 currently which is the z endstop pin)
Z_MIN_PROBE_PIN PB2 (it's set to PB1 atm)
FIL_RUNOUT_PIN PA4 (it's currently set to PC15 which i could find a reference for)

Changes that would work out of the box look like this:

```diff
index 1d142d87cc..11850604db 100644
--- a/Marlin/src/pins/hc32f4/pins_CREALITY_ENDER2P_V24S4.h
+++ b/Marlin/src/pins/hc32f4/pins_CREALITY_ENDER2P_V24S4.h
@@ -75,7 +75,7 @@
 // Servos
 //
 #ifndef SERVO0_PIN
-  #define SERVO0_PIN                        PB0   // BLTouch OUT *
+  #define SERVO0_PIN                        PB1   // BLTouch OUT *
 #endif

 //
@@ -83,17 +83,17 @@
 //
 #define X_STOP_PIN                          PA5
 #define Y_STOP_PIN                          PA6
-#define Z_STOP_PIN                          PB0   // BLTOUCH *
+#define Z_STOP_PIN                          PB0

 #ifndef Z_MIN_PROBE_PIN
-  #define Z_MIN_PROBE_PIN                   PB1   // BLTouch IN *
+  #define Z_MIN_PROBE_PIN                   PB2   // BLTouch IN *
 #endif

 //
 // Filament Runout Sensor
 //
 #ifndef FIL_RUNOUT_PIN
-  #define FIL_RUNOUT_PIN                    PC15  // "Pulled-high" *
+  #define FIL_RUNOUT_PIN                    PA4  // "Pulled-high" *
 #endif
```

I have tested these pin settings with a CRTouch and a cheap bltouch clone.

Labels on board for the bltouch connector:
[image]
BLTouch wiring:
[image]
CRTouch wiring:
[image]

### Bug Timeline

Since the file was first added

### Expected behavior

CRTouch/BLTouch would work by plugging the probes in the board and enabling BLTOUCH in the config.
Filament runout sensor would work by plugging into board and enabling in config.

### Actual behavior

Need to modify the pins for  SERVO0_PIN, Z_MIN_PROBE_PIN and FIL_RUNOUT_PIN

### Version of Marlin Firmware

Marlin 2.1.x-bugfix@e310f20 

### Printer model

Ender 2 Pro with v2.4.S4 board

### Other add-ons

BLTouch, Filament Runout Sensor

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

I haven't included Configuration.h and Configuration_adv.h since they are not really relevant to the issue.

I'm not sure how to provide tests or how to confirm whether the changes are correct.

classicrocker883, 2024-11-08:
this would be better as a Pull Request

calorifer, 2024-11-08:
Ok, i will link a pull request.

However, i found PA4 to be used as the X_MAX_PIN limit switch in the stock firmware ( https://github.com/CrealityOfficial/Ender-2-Pro/blob/b9046ba1466f3f3314f7e487475cdd33f882705d/source/Marlin/src/pins/stm32f1/pins_CREALITY_V428.h#L78 )  but the printer doesn't have an x_max limit switch, the connector is empty and the port has 3 pins compared to all the other limit switches on the printer which have only 2. The labeling on the board would match with this definition though (the pin is labeled X+ on the board).

Is it ok to define it as the filament runout pin in marlin for this board?

Another issue with the creality source code for this board is that the definitions for SERVO0_PIN and Z_MIN_PROBE_PIN are reversed there too compared to what they should be for the connectors to work. The pins are defined the same in platformio and the framework creality uses so it's not a pin naming issue.

Creality hasn't released a firmware with bltouch or filament runout enabled for this board and other configs for this board i found use the pins in the same way i found it to work ( example: https://github.com/internallabworks/ender2pro/blob/167829cdca9d896b3d6d3deb59a908345889044f/printer.cfg#L78 )

classicrocker883, 2024-11-09:
> However, i found PA4 to be used as the X_MAX_PIN limit switch in the stock firmware but the printer doesn't have an x_max limit switch, the connector is empty and the port has 3 pins compared to all the other limit switches on the printer which have only 2.
> Creality hasn't released a firmware with bltouch or filament runout enabled for this board and other configs for this board i found use the pins in the same way i found it to work ( example: https://github.com/internallabworks/ender2pro/blob/167829cdca9d896b3d6d3deb59a908345889044f/printer.cfg#L78 )

that is interesting. since there are 3 pins for the connector, perhaps it would be better suited as a filament runout sensor - or for a bltouch like you said.

thinkyhead, 2024-11-12:
Has any of this been tested? It's not clear what pins file is authoritative from Creality, and their repository has changed a lot between [CrealityOfficial/Ender-2-Pro/b9046ba146](https://github.com/CrealityOfficial/Ender-2-Pro/blob/b9046ba1466f3f3314f7e487475cdd33f882705d/source/Marlin/src/pins/stm32f1/pins_CREALITY_V428.h) and today. We need feedback from someone who can build Marlin for Ender-2 Pro with board 2.4 HC32 — with the current and changed pins.

calorifer, 2024-11-12:
I found this issue setting up an ender-2 pro with this board, so i have built and ran this, but i don't know exactly how to find someone trusted by the Marlin project to build, run and confirm that these pins are correct.

calorifer, 2024-11-13:
> Has any of this been tested? It's not clear what pins file is authoritative from Creality, and their repository has changed a lot between [...]

Hey, i tested the changes before creating this issue with a crtouch and a bltouch clone but that basically leaves it at "it works on my machine".

As far as i can tell the Creality repo for this printer and board hasn't changed in almost 2 years.

The 2.4.S4 HC32 board is present in the newer Ender-2 Pro printers, previously they used a relatively standard Creality V4.2.* board. For this board they have a different branch: [MCU_HDSC ](https://github.com/CrealityOfficial/Ender-2-Pro/tree/MCU_HDSC) that looks very different than what a normal Marlin repo would look because at the time they released it Creality was not building it with platformio. Support in Marlin for this chip was added here: [Add HAL for HC32](https://github.com/MarlinFirmware/Marlin/pull/26414) and for the board was added here: [[FR] (Unkown Mainboard with HC32F460 MCU)
](https://github.com/MarlinFirmware/Marlin/issues/24338). About a year after the last firmware release from Creality.

Any info i could find that would be authoritative (the creality ender-2 pro repo for this version of the printer) seems to be wrong and the labeling on the board seems to be reversed compared to what i would expect it to be (BLTOUCH IN for input from the bltouch, BLTOUCH OUT for servo control) which is why i named this issue as 'possibly incorrect' because the available authoritative info doesn't actually work for the machine i'm using and i found other people using the same printer to have set the same pins as i did (though for klipper, not marlin).

As far as i can tell, the [MCU_HDSC ](https://github.com/CrealityOfficial/Ender-2-Pro/tree/MCU_HDSC) branch is the source for the latest official firmware release available for this printer with this board because the binary committed to their repo is the same as the one currently available for download from creality:
- in their repo: https://github.com/CrealityOfficial/Ender-2-Pro/tree/MCU_HDSC/workspace/bin
- on the creality site: https://www.crealitycloud.com/software-firmware/firmware/ender-series

Both files (Ender-2pro-hw2.4.s4xx-sw2.2.41_165.bin) are identical.

None of their releases have bltouch or filament runout enabled so the pin definitions in their repo for these features are not actually used.

thinkyhead, 2024-11-14:
Thanks for the info. So, I doubt they accidentally left `M43` turned on, but that would be the ultimate way to check, if they had.

Our pins were borrowed from Klipper, which means that any functions that used to connect to the mainboard under Marlin are probably taken over by pins on the Pi side. And, if those functions were never used, makes sense they should be "whatever" in Creality's own files….

We can work with this!

closed 2024-11-14 by thinkyhead

thinkyhead, 2024-11-25:
Hi! We've been busy getting releases and builds together, but we're making progress!

So, in general we want board definitions to reflect how the pins on the board are labeled (when that labeling is correct, and we can add comments if pins are labeled badly) and how the pins on the board would be used if you moved the board to some generic 3D printer, laser, or CNC.

Some pins files (Trigorilla AVR) have special mappings for the machines they ship in, but if pins aren't overriddein or flags for those machines aren't set (in the Configurations files) then we get a generic mapping. We can take the same approach for this board, providing tweaks to the usual pin functions when the board is used in a shipping Ender-2 Pro, and then expose a more generic mapping otherwise.