MarlinFirmware/Marlin #27486: [FR] Support for compiling on Raspberry Pi

thijstriemstra, 2024-10-20:
### Is your feature request related to a problem? Please describe.

When I try to compile Marlin on a Raspberry Pi 3b+ (debian bookworm, 64bit) it fails with:

```
Processing STM32F446ZE_btt (board: marlin_BTT_Octopus_v1; platform: ststm32@~12.1; framework: arduino)
----------------------------------------------------------------------------------------------------------------
Platform Manager: Installing ststm32 @ ~12.1
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Platform Manager: ststm32@12.1.1 has been installed!
Tool Manager: Installing platformio/toolchain-gccarmnoneeabi @ ~1.90201.0
UnknownPackageError: Could not find the package with 'platformio/toolchain-gccarmnoneeabi @ ~1.90201.0' requirements for your system 'linux_aarch64'
```

### Describe the feature you want

I would like to be able to compile Marlin on a Raspberry Pi.

### Additional context

This issue was raised before (#23815) but it was closed with rather vague message about it being a platfomio issue. What issue specifically? I think it's more an issue with Marlin using an ancient `ststm32@~12.1` which doesn't support `linux_aarch64`?

The-EG, 2024-10-20:
It's a platformio issue because it depends on the compiler toolchain that platformio provides. As your error message indicates, platformio does not provide the proper toolchain for aarch64, which is needed for an arm64 such as a raspberry pi.

This isn't something Marlin can address directly. It might be possible to update to another version of ststm32, which would presumably might offer aarch64 support, but that would also likely require fixing up all the different variants and such along with it. It's not just a trivial version bump.

thijstriemstra, 2024-10-20:
I see there's an alternative `platformio/toolchain-gccarmnoneeabi` that presumably has linux_aarch64 support: https://github.com/platformio/platform-ststm32/commit/2429457be5e22a157f9cb7f6b0acdd83675e7ec1 

maybe there's a way to useststm32 12.1 with this `toolchain-gccarmnoneeabi ~1.100301.0`?

thijstriemstra, 2024-11-09:
> maybe there's a way to use ststm32 12.1 with this toolchain-gccarmnoneeabi ~1.100301.0 so these variants don't have to change? And what does it entail, "fixing up the different variants"?

Well turned out that was simple.... Tested on RPI5 bookworm 12 with patch:

```diff
diff --git a/ini/stm32-common.ini b/ini/stm32-common.ini
index ec058d1..e3cabbd 100644
--- a/ini/stm32-common.ini
+++ b/ini/stm32-common.ini
@@ -30,6 +30,7 @@ custom_marlin.TFT_TOUCH_DEVICE_GT911 = build_src_filter=+<src/HAL/STM32/tft/gt91
 # STM32 board based on a variant.
 #
 [stm32_variant]
+platform_packages = toolchain-gccarmnoneeabi@~1.100301.0
 extends       = common_stm32
 extra_scripts = ${common_stm32.extra_scripts}
                 pre:buildroot/share/PlatformIO/scripts/generic_create_variant.py
```

Result without patch:

```
Processing STM32F446ZE_btt (board: marlin_BTT_Octopus_v1; platform: ststm32@~12.1; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------
Tool Manager: Installing platformio/toolchain-gccarmnoneeabi @ ~1.90201.0
UnknownPackageError: Could not find the package with 'platformio/toolchain-gccarmnoneeabi @ ~1.90201.0' requirements for your system 'linux_aarch64'
```

Result with patch:

```
Checking size .pio/build/STM32F446ZE_btt/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]   8.2% (used 10696 bytes from 131072 bytes)
Flash: [===       ]  34.1% (used 178696 bytes from 524288 bytes)
Building .pio/build/STM32F446ZE_btt/firmware.bin
================================================================= [SUCCESS] Took 145.02 seconds =================================================================

Environment      Status    Duration
---------------  --------  ------------
STM32F446ZE_btt  SUCCESS   00:02:25.022
```