MarlinFirmware/Marlin #27489: #define EDGE_STEPPING remove step signal on Z, but works on Z2, result is noise and missing step.

moriamoria, 2024-10-22:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Hello,

I have an ortur 3d printer with an upgrade skr 1.3 motherboard.

Z uses driver 2209, Z2 stepper uses E1 stepper driver 2209.

When  #define EDGE_STEPPING, I have a good signal on Z2, but nothing on Z.

Result Z2 stepper force on Z and create noise, and missing step.

Following the data scope from Z in yellow, Z2 in green 

[image]

if I comment //#define EDGE_STEPPING then I have good signal, but not square

[image]

notice that on branch 4.2.1.2, when I activate #define SQUARE_WAVE_STEPPING, I have the same result, no signal on Z scope signal.

Thanks helping me to solve this issue.

Regards to all

### Bug Timeline

all bugfix 2.1.X commit suffer from this issue

### Expected behavior

Move Z and Z2 synchrone.

### Actual behavior

The Z2 stepper force on Z stepper

### Steps to Reproduce

#define SQUARE_WAVE_STEPPING
move Z
[Marlin.zip](https://github.com/user-attachments/files/17482744/Marlin.zip)

### Version of Marlin Firmware

Bugfig 2.1.2. [cron] Bump distribution date (2024-10-21)

### Printer model

ortur, Z and Z2 connected

### Electronics

skr 1.3, 2209 on Z,Z2, 

### LCD/Controller

REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

### Other add-ons

2209

### Bed Leveling

ABL Linear grid

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2024-10-23:
Provided Configs has //#define Y_DRIVER_TYPE  TMC2209 which mean that Y is disabled, is this deliberate?

anomalchik, 2024-10-23:
I think this should have been written in the previous thread, and not created a new one.

moriamoria, 2024-10-23:
Hello,

I have recreate a new issue because in the LIN_ADVANCE issue I have raise before, someone ask me to recreate an issue with the good subject.

Regarding Y, I use a close loop driver mks 42C

[image]

Regards

ellensp, 2024-10-23:
issue https://github.com/MarlinFirmware/Marlin/issues/27341 does seem to be the same issue
https://github.com/MarlinFirmware/Marlin/issues/27457 however is specifically talking about LIN_ADVANCE

anomalchik, 2024-10-23:
@tombrazier I set #define OLD_ADAPTIVE_MULTISTEPPING and it didn't bring any results.

tombrazier, 2024-11-04:
> issue #27341 does seem to be the same issue #27457 however is specifically talking about LIN_ADVANCE vs EDGE_STEPPING

@moriamoria, the OP of #27457, discovered that LIN_ADVANCE as not the cause after all. This issue, therefore, is a duplicate but without any confusing references to LIN_ADVANCE. I plan to close #27457 unless anyone objects.

#27341 does also seem to be a duplicate.

thisiskeithb, 2024-11-05:
Closing this issue in favor of https://github.com/MarlinFirmware/Marlin/issues/27341 since it's older & there's more discussion over there.

closed 2024-11-05 by thisiskeithb