MarlinFirmware/Marlin #27490: [BUG] enabled LCD_BED_TRAMMING not working

freek-heining, 2024-10-23:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I moved from regular 2.1 to bugfix 2.1 and compiled my firmware succesfully for my Ender 3 Pro. 
Sadly the only thing not working is the bed tramming wizard in the menu, which I used a lot in the regular 2.1.
The wizard/option is just nowhere to be found. I reflashed several times and resetted the EEPROM etc.

// Add a menu item to move between bed corners for manual bed adjustment
#define LCD_BED_TRAMMING

#if ENABLED(LCD_BED_TRAMMING)
  #define BED_TRAMMING_INSET_LFRB { 30, 30, 30, 30 } // (mm) Left, Front, Right, Back insets
  #define BED_TRAMMING_HEIGHT      0.4        // (mm) Z height of nozzle at leveling points
  #define BED_TRAMMING_Z_HOP       4.0        // (mm) Z height of nozzle between leveling points
  #define BED_TRAMMING_INCLUDE_CENTER       // Move to the center after the last corner
  //#define BED_TRAMMING_USE_PROBE
  #if ENABLED(BED_TRAMMING_USE_PROBE)
    #define BED_TRAMMING_PROBE_TOLERANCE 0.1  // (mm)
    #define BED_TRAMMING_VERIFY_RAISED        // After adjustment triggers the probe, re-probe to verify
    //#define BED_TRAMMING_AUDIO_FEEDBACK
  #endif
    (examples omitted)
  #define BED_TRAMMING_LEVELING_ORDER { LF, RF, RB, LB }
#endif

### Bug Timeline

New bug. Today

### Expected behavior

The bed tramming wizard appearing in the main menu.

### Actual behavior

It's not showing up.

### Steps to Reproduce

Uncomment #define LCD_BED_TRAMMING to enable

### Version of Marlin Firmware

2.1.2.4

### Printer model

Ender 3 Pro

### Electronics

SKR E3 mini v2.0

### Bed Leveling

No Bed Leveling

### Your Slicer

Cura

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin.zip](https://github.com/user-attachments/files/17498153/Marlin.zip)

thisiskeithb, 2024-10-24:
This was changed in https://github.com/MarlinFirmware/Marlin/pull/27393/

You need to enable `MESH_BED_LEVELING` for the menu to show up.

freek-heining, 2024-10-24:
> This was changed in #27393
> 
> You need to enable `MESH_BED_LEVELING` for the menu to show up.

Thank you it worked! 
I Googled a lot before asking this question but found nothing. Also it was not clear for me by reading the comments in the config file.

thisiskeithb, 2024-10-24:
This needs to be fixed or sanity checked, so the issue should remain open.

closed 2024-11-23 by thinkyhead