MarlinFirmware/Marlin #27496: [BUG] Rebooting while homing

AndreySamokhin, 2024-10-26:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

The board reboots while homing the Z-axis.

The issue occurs inside the `do_homing_move()` function with the `planner.synchronize();` being the last executed line. Further investigation showed that the `Planner::busy()` function never returns `0` because the `Planner::has_blocks_queued()` function consistently returns `1`.

The minimum required changes of default configuration files:
* Add a fourth axis.
* Use the TB6560 driver for X, Y, or I axis.
* Perform homing at high speed.

[configuration.zip](https://github.com/user-attachments/files/17529676/configuration.zip)

### Bug Timeline

New bug.

### Expected behavior

When homing with the Z endstop disconnected, the expected behavior is to receive the `Printer halted. kill() called!` error. This is achievable if the configuration specifies the A4988 driver for the I axis rather than TB6560.

[image]

### Actual behavior

The board reboots a few seconds after initiating the homing routine along the Z-axis.

[image]

### Steps to Reproduce

The bug can be reproduced using only motherboard without connecting any sensors, motors, etc.

1. Modify the default configuration files to add a fourth axis, set the TB6560 driver for the I axis and perform homing at high speed.

```
// Configuration.h
#ifndef MOTHERBOARD
  #define MOTHERBOARD BOARD_MKS_GEN_L
#endif
#define BAUDRATE 115200
#define I_DRIVER_TYPE  TB6560
#define DEFAULT_AXIS_STEPS_PER_UNIT   { 80, 80, 4200, 80, 500 }
#define DEFAULT_MAX_FEEDRATE          { 300, 300, 5, 300, 25 }
#define DEFAULT_MAX_ACCELERATION      { 3000, 3000, 100, 3000, 10000 }
#define I_ENABLE_ON 0
#define INVERT_I_DIR false
#define I_HOME_DIR -1
#define I_MIN_POS 0
#define I_MAX_POS 50
#define HOMING_FEEDRATE_MM_M { (50*60), (50*60), (5*60), (50*60) }

// Configuration_adv.h
#define HOMING_BUMP_MM      { 5, 5, 2, 5 }
#define HOMING_BUMP_DIVISOR { 2, 2, 4, 2 }
#define AXIS_RELATIVE_MODES { false, false, false, false, false }

// pins_MKS_GEN_L.h
#define I_STEP_PIN 49
#define I_DIR_PIN 50
#define I_ENABLE_PIN 51
#define I_MIN_PIN 52
```

2. Establish communication with the board via UART.
3. Send the `M119` command to confirm that the Z-axis endstop is not triggered.
4. Start homing routine using the `G28 Z` command.
5. After a few seconds observe that the the board reboots.

### Version of Marlin Firmware

bugfix-2.1.x (d2bda12); 2.1.2.4

### Printer model

DIY

### Electronics

Board - MKS GEN L V1.0 (ATmega2560); X, Y and Z drivers - A4988; I driver - TB6560

### LCD/Controller

LCD is not used.

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

classicrocker883, 2024-10-30:
I had an issue with homing because of variables declared like so:
```y
  float parkingposx[2],          
        parkinggrabdistance;
```

while this snippet doesnt really have to do with homing it is just an example.

for instance:
```diff
- uint32_t max_acceleration_mm_per_s2[DISTINCT_AXES],
-          min_segment_time_us;
            
+ uint32_t max_acceleration_mm_per_s2[DISTINCT_AXES],
+ uint32_t min_segment_time_us;
```

there are a few files which should have this explicitly defined otherwise it can cause issues.

plampix, 2024-10-30:
Why do you think this results in different code. Because it doesn't.

classicrocker883, 2024-11-01:
@plampix because when I had changed[ this code](https://github.com/classicrocker883/MRiscoCProUI/issues/159#issuecomment-2391995690), it resulted in erratic behavior when homing.

even if you ask ChatGPT what issues may happen if you group them when declaring:
```
This difference would cause issues if you tried to compile the first code snippet, as you cannot declare the same variable name with different data types in the same scope. The compiler would throw an error.
```

AndreySamokhin, 2024-11-02:
@classicrocker883, you could have mentioned the [Dangerous behavior when homing](https://github.com/classicrocker883/MRiscoCProUI/issues/159) issue in your first message.
Your explanation sounds unrealistic. As @plampix pointed out, comma-separated declarations are valid in C/C++. The actual problem with the code you modified is different: you changed the order of variables in the `planner_settings_t` structure, which really matters.
Next time, please do not refer to ChatGPT as a reliable source without double-checking its answers or at least carefully read it responses, as what it said ("... you cannot declare the same variable name with different data types...") is inconsistent with your explanation.

classicrocker883, 2024-11-09:
it wasnt the "order of variables" in the struct that mattered.  `feedrate_t` is literally the same as  `float`. declaring the variable inline as a different type was the cause.

AndreySamokhin, 2024-11-17:
@classicrocker883, I'd like to conclude this off-topic and ultimately pointless discussion. Are you aware that settings are typically loaded from EEPROM at startup? Do you understand how this process works under the hood? If not, I recommend examining the `MarlinSettings::_load()` function. Based on your responses, it seems you haven’t conducted any research of your own, so here is a brief summary for you. For the `planner_settings_t` structure, variables are loaded from EEPROM sequentially, in the same order they were originally defined. Now, consider what will happen when someone decides to reorder them to ["organize/simplify some code"](https://github.com/classicrocker883/MRiscoCProUI/issues/159#issuecomment-2391995690). Can you guess the consequences of such a change?

ellensp, 2025-02-20:
I cannot replicate this exactly, but I can get weird results using a ramps 1.4

I setup test axis on X,Y,Z and A on E1  with endstops plugged into X-MIN,Y-MIN,Z-MIN and A is on Z-MAX
X,Y,Z and E DRIVER_TYPE are all A4988,  I_DRIVER_TYPE is  TB6560

Power up and Send G28

Under normal conditions this is what happens
1) Z will raise
2) X will home
3) Y will home
4) Z will home
5) Z will home again at slower speed
6) A will home
7) Z will raise

When it fails it stops while doing step 4

Things I found from experimentation

DEFAULT_AXIS_STEPS_PER_UNIT plays a part.
With #define DEFAULT_AXIS_STEPS_PER_UNIT   { 80, 80, 2700, 80, 500 }  and #define I_DRIVER_TYPE  TB6560  It works as expected

but  
with #define DEFAULT_AXIS_STEPS_PER_UNIT   { 80, 80, 2750, 80, 500 } and #define I_DRIVER_TYPE  TB6560  it breaks.
The user provided DEFAULT_AXIS_STEPS_PER_UNIT   { 80, 80, 4200, 80, 500 } is way over the breaking point.

Also 
with #define DEFAULT_AXIS_STEPS_PER_UNIT   { 80, 80, 2750, 80, 500 } and #define I_DRIVER_TYPE  A4988  it works as expected.

To me this looks like some sort of timing issue with stepper pulse generation and steps/mm.

With TB6560 defined the following are defined

    #define MINIMUM_STEPPER_POST_DIR_DELAY 15000
    #define MINIMUM_STEPPER_PULSE_NS 30000
    #define MAXIMUM_STEPPER_RATE 15000

I suspect that MAXIMUM_STEPPER_RATE is to slow, it is impossible to generate the required number of steps pules in the time required with this constraint.   

These are my ramps test configs 

[Ramps test Configuration.zip](https://github.com/user-attachments/files/18886140/Ramps.test.Configuration.zip)

thinkyhead, 2025-02-20:
Well, here's what I can tell you differs in Marlin for TB6560 versus other drivers: surprisingly not much! Just these values:

```cpp
    #define MINIMUM_STEPPER_POST_DIR_DELAY 15000
    #define MINIMUM_STEPPER_PULSE_NS 30000
    #define MAXIMUM_STEPPER_RATE 15000
```

- The `MINIMUM_STEPPER_POST_DIR_DELAY` and `MINIMUM_STEPPER_PULSE_NS` are the highest of all driver types and the `MAXIMUM_STEPPER_RATE` is the lowest, so it really constrains the speeds you can reach and the number of axes. The DIR delay, in particular, requires a pause to occur inside the Stepper ISR for every direction change, and this can cause a watchdog reset without too much strain.

I wouldn't say we have a bug in Marlin so much as a case where we're not able to handle the given parameters within an interrupt context. You can try lowering some of those delays just to the point where DIR changes are still reliable.

Another option would be to try using `FT_MOTION` which does not use blocking DIR delays.

thinkyhead, 2025-02-20:
As a general test condition, you should also try turning off the `USE_WATCHDOG` setting, to see if it still resets. That will tell us that the reset is indeed caused by the ISR using up all the available cycles.

The solution within our existing planner would be to take account of very long DIR delays and exit the ISR for a while to allow the main loop to proceed, then return to continue further steps later, but this would require some re-architecture of the way it now works.

So until we get that sorted out your best option is to use `FT_MOTION`, but be aware that it is new and not widely tested and may require additional tuning.

AndreySamokhin, 2025-03-27:
With `USE_WATCHDOG` disabled, the board does not reboot. However, its behavior is still abnormal, with the `echo:busy: processing` message being printed slowly, letter by letter, in the terminal.

I believe that unexpected behavior is the worst outcome; a compile-time error would be a much better alternative.

@thinkyhead, would adding a few additional lines to *SanityCheck.h* be a good solution to address this issue? If so, I would be happy to submit a pull request.

```
// Both 'sanity_arr_1[]' and 'sanity_arr_2[]' are already defined in
// 'SanityCheck.h'. They correspond to 'DEFAULT_AXIS_STEPS_PER_UNIT' and
// 'DEFAULT_MAX_FEEDRATE' respectively

#ifdef MAXIMUM_STEPPER_RATE
  #if HAS_X_AXIS
    static_assert(sanity_arr_1[X_AXIS] * sanity_arr_2[X_AXIS] <= MAXIMUM_STEPPER_RATE,
    "DEFAULT_AXIS_STEPS_PER_UNIT[X_AXIS] * DEFAULT_MAX_FEEDRATE[X_AXIS] > MAXIMUM_STEPPER_RATE");
  #endif
  // Apply the same check for all other axes.
#endif
```

AndreySamokhin, 2025-03-28:
I prepared a pull-request, but one of the tests failed because for the applied configuration (i.e., */Creality/Ender-5 Plus/BigTreeTech SKR 3/Configuration.h*), the product of `DEFAULT_AXIS_STEPS_PER_UNIT` and `DEFAULT_MAX_FEEDRATE` exceeded `MAXIMUM_STEPPER_RATE`.

To evaluate the scope of the issue, I decided to parse all *Configuration.h* files from the [Configurations](https://github.com/MarlinFirmware/Configurations/tree/import-2.1.x) repository. To simplify the parsing routine, I excluded files from consideration if:

* `DEFAULT_AXIS_STEPS_PER_UNIT` or ` DEFAULT_MAX_FEEDRATE` lines were duplicated (12 files);
* additional macros were used instead of specifying values explicitly (47 files);
* the number of axes specified in `DEFAULT_AXIS_STEPS_PER_UNIT` and `DEFAULT_MAX_FEEDRATE` were not identical;
* 'non-standard' drivers were specified (4 files).

The results were not as bad as I had expected. Among 314 considered files (out of 379), only 6 failed the condition mentioned above:

* */Creality/Ender-5 Plus/BigTreeTech SKR 3/Configuration.h*
* */Creality/Ender-5 Plus/CrealityV1 (ORIGIN UI)/Configuration.h*
* */Creality/Ender-5 Plus/CrealityV1 (RELOADED UI)/Configuration.h*
* */Velleman/K8400/Dual-head/Configuration.h*
* */Velleman/K8400/Single-head/Configuration.h*
* */Wanhao/Duplicator i3 2.1/Configuration.h*

It should be noted that in all cases, unrealistic feedrates were specified.

```
// …/Creality/Ender-5 Plus/BigTreeTech SKR 3/Configuration.h*
// …/Creality/Ender-5 Plus/CrealityV1 (ORIGIN UI)/Configuration.h*
// …/Creality/Ender-5 Plus/CrealityV1 (RELOADED UI)/Configuration.h*
#define DEFAULT_AXIS_STEPS_PER_UNIT   { 80, 80, 800, 93.02 }
#define DEFAULT_MAX_FEEDRATE          { 8000, 8000, 15, 15000 }

// …/Velleman/K8400/Dual-head/Configuration.h
// …/Velleman/K8400/Single-head/Configuration.h
#define DEFAULT_AXIS_STEPS_PER_UNIT   { 134.74, 134.74, 4266.66, 148.7 }
#define DEFAULT_MAX_FEEDRATE          { 160, 160, 10, 10000 }

// …/Wanhao/Duplicator i3 2.1/Configuration.h
#define DEFAULT_AXIS_STEPS_PER_UNIT   { 80, 80, 400, 100 }
#define DEFAULT_MAX_FEEDRATE          { 700, 700, 100, 10000 }
```

That said, implementing additional sanity checks would require a bigger undertaking than I initially anticipated. While I do not mind fixing the six files mentioned above, it is not a priority and would require some effort from the community. @thinkyhead, what do you think, would it be worth it?

thinkyhead, 2025-04-02:
> implementing additional sanity checks would require a bigger undertaking than I initially anticipated

If I apply the new sanity checks and run the script to build all examples it will reveal all the example configs that have bad defaults. If you have a list of configs to patch already, please submit those and that will save us some time in the upcoming deployment.

AndreySamokhin, 2025-04-02:
Six files with unrealistic feed rates were [fixed](https://github.com/MarlinFirmware/Configurations/pull/1137). Of the 381 configuration files, 336 were checked automatically. The remaining 45 use nested macros and require a more sophisticated parser for automated processing.

closed 2025-04-07 by thinkyhead