MarlinFirmware/Marlin #27500: build warnings

thijstriemstra, 2024-10-27:
### Is your feature request related to a problem? Please describe.

I'm seeing a lot of warnings after upgrading to:
```
commit d2bda128abb4b08738d8ead651affc93e2821775 (HEAD -> bugfix-2.1.x, origin/bugfix-2.1.x, origin/HEAD)
Author: thinkyhead <thinkyhead@users.noreply.github.com>
Date:   Mon Oct 21 00:28:30 2024 +0000

    [cron] Bump distribution date (2024-10-21)
```

```
In file included from Marlin/src/lcd/menu/menu_main.cpp:31:
Marlin/src/lcd/menu/menu_main.cpp: In function 'void custom_menus_main()':
Marlin/src/lcd/menu/../../inc/../../Configuration_adv.h:4050:33: warning: unknown escape sequence: '\&'
 4050 |   #define MAIN_MENU_ITEM_3_DESC "Home X\&Y" // "Preheat for " PREHEAT_2_LABEL
      |                                 ^~~~~~~~~~~
Marlin/src/lcd/menu/menu_item.h:288:25: note: in definition of macro '_MENU_INNER_F'
  288 |   FSTR_P const flabel = FLABEL;                                \
      |                         ^~~~~~
Marlin/src/lcd/menu/menu_item.h:424:57: note: in expansion of macro '_MENU_ITEM_F'
  424 | #define MENU_ITEM_F(TYPE, FLABEL, V...)                 _MENU_ITEM_F(TYPE, false, FLABEL, ##V)
      |                                                         ^~~~~~~~~~~~
Marlin/src/lcd/menu/menu_item.h:464:58: note: in expansion of macro 'MENU_ITEM_F'
  464 | #define SUBMENU_F(FLABEL, DEST)                          MENU_ITEM_F(submenu, FLABEL, DEST)
      |                                                          ^~~~~~~~~~~
Marlin/src/lcd/menu/menu_main.cpp:141:7: note: in expansion of macro 'SUBMENU_F'
  141 |       SUBMENU_F(F(MAIN_MENU_ITEM_##N##_DESC), []{ \
      |       ^~~~~~~~~
/home/workshop/.platformio/packages/framework-arduinoststm32/cores/arduino/WString.h:38:74: note: in expansion of macro 'PSTR'
   38 | #define F(string_literal) (reinterpret_cast<const __FlashStringHelper *>(PSTR(string_literal)))
      |                                                                          ^~~~
Marlin/src/lcd/menu/menu_main.cpp:141:17: note: in expansion of macro 'F'
  141 |       SUBMENU_F(F(MAIN_MENU_ITEM_##N##_DESC), []{ \
      |                 ^
Marlin/src/lcd/menu/menu_main.cpp:141:19: note: in expansion of macro 'MAIN_MENU_ITEM_3_DESC'
  141 |       SUBMENU_F(F(MAIN_MENU_ITEM_##N##_DESC), []{ \
      |                   ^~~~~~~~~~~~~~~
Marlin/src/lcd/menu/menu_main.cpp:152:9: note: in expansion of macro '_CUSTOM_ITEM_MAIN_CONFIRM'
  152 |         _CUSTOM_ITEM_MAIN_CONFIRM(N); \
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~
Marlin/src/lcd/menu/menu_main.cpp:164:7: note: in expansion of macro 'CUSTOM_ITEM_MAIN'
  164 |       CUSTOM_ITEM_MAIN(3);
      |       ^~~~~~~~~~~~~~~~
Marlin/src/lcd/menu/menu_main.cpp: In lambda function:
Marlin/src/lcd/menu/menu_main.cpp:144:41: warning: unknown escape sequence: '\&'
  144 |             F(MAIN_MENU_ITEM_##N##_DESC "?")      \
      |                                         ^~~
Marlin/src/lcd/menu/menu_item.h:290:39: note: in definition of macro '_MENU_INNER_F'
  290 |     MenuItem_##TYPE::action(flabel, ##V);                      \
      |                                       ^
Marlin/src/lcd/menu/menu_item.h:424:57: note: in expansion of macro '_MENU_ITEM_F'
  424 | #define MENU_ITEM_F(TYPE, FLABEL, V...)                 _MENU_ITEM_F(TYPE, false, FLABEL, ##V)
      |                                                         ^~~~~~~~~~~~
Marlin/src/lcd/menu/menu_item.h:464:58: note: in expansion of macro 'MENU_ITEM_F'
  464 | #define SUBMENU_F(FLABEL, DEST)                          MENU_ITEM_F(submenu, FLABEL, DEST)
      |                                                          ^~~~~~~~~~~
Marlin/src/lcd/menu/menu_main.cpp:141:7: note: in expansion of macro 'SUBMENU_F'
  141 |       SUBMENU_F(F(MAIN_MENU_ITEM_##N##_DESC), []{ \
      |       ^~~~~~~~~
/home/workshop/.platformio/packages/framework-arduinoststm32/cores/arduino/WString.h:38:74: note: in expansion of macro 'PSTR'
   38 | #define F(string_literal) (reinterpret_cast<const __FlashStringHelper *>(PSTR(string_literal)))
      |                                                                          ^~~~
Marlin/src/lcd/menu/menu_main.cpp:144:13: note: in expansion of macro 'F'
  144 |             F(MAIN_MENU_ITEM_##N##_DESC "?")      \
      |             ^
Marlin/src/lcd/menu/menu_main.cpp:152:9: note: in expansion of macro '_CUSTOM_ITEM_MAIN_CONFIRM'
  152 |         _CUSTOM_ITEM_MAIN_CONFIRM(N); \
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~
Marlin/src/lcd/menu/menu_main.cpp:164:7: note: in expansion of macro 'CUSTOM_ITEM_MAIN'
[... 41 lines cut ...]
  446 | #define ACTION_ITEM_F(FLABEL, ACTION)                    MENU_ITEM_F(function, FLABEL, ACTION)
      |                                                          ^~~~~~~~~~~
Marlin/src/lcd/menu/menu_main.cpp:139:34: note: in expansion of macro 'ACTION_ITEM_F'
  139 |     #define _CUSTOM_ITEM_MAIN(N) ACTION_ITEM_F(F(MAIN_MENU_ITEM_##N##_DESC), GCODE_LAMBDA_MAIN(N));
      |                                  ^~~~~~~~~~~~~
/home/workshop/.platformio/packages/framework-arduinoststm32/cores/arduino/WString.h:38:74: note: in expansion of macro 'PSTR'
   38 | #define F(string_literal) (reinterpret_cast<const __FlashStringHelper *>(PSTR(string_literal)))
      |                                                                          ^~~~
Marlin/src/lcd/menu/menu_main.cpp:139:48: note: in expansion of macro 'F'
  139 |     #define _CUSTOM_ITEM_MAIN(N) ACTION_ITEM_F(F(MAIN_MENU_ITEM_##N##_DESC), GCODE_LAMBDA_MAIN(N));
      |                                                ^
Marlin/src/lcd/menu/menu_main.cpp:139:50: note: in expansion of macro 'MAIN_MENU_ITEM_3_DESC'
  139 |     #define _CUSTOM_ITEM_MAIN(N) ACTION_ITEM_F(F(MAIN_MENU_ITEM_##N##_DESC), GCODE_LAMBDA_MAIN(N));
      |                                                  ^~~~~~~~~~~~~~~
Marlin/src/lcd/menu/menu_main.cpp:154:9: note: in expansion of macro '_CUSTOM_ITEM_MAIN'
  154 |         _CUSTOM_ITEM_MAIN(N); \
      |         ^~~~~~~~~~~~~~~~~
Marlin/src/lcd/menu/menu_main.cpp:164:7: note: in expansion of macro 'CUSTOM_ITEM_MAIN'
  164 |       CUSTOM_ITEM_MAIN(3);
      |       ^~~~~~~~~~~~~~~~
```

### Describe the feature you want

Get a clean build again.

thisiskeithb, 2024-10-27:
Change `#define MAIN_MENU_ITEM_3_DESC "Home X\&Y"` to `#define MAIN_MENU_ITEM_3_DESC "Home X&Y"`

---

This is not a Feature Request.

Next time, please use ["🪲 Report a bug" when reporting a potential bug](https://github.com/MarlinFirmware/Marlin/issues/new/choose) and be sure to fill out the entire report, including attaching your configs.

[image]

---
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.com/servers/marlin-firmware-461605380783472640) Discord server

closed 2024-10-27 by thisiskeithb

thijstriemstra, 2024-10-27:
Oops sorry, I'll keep that mind, thanks @thisiskeithb