MarlinFirmware/Marlin #27505: [BUG] All Z Offset Values Result in Same Behavior - No Effect

xaviervilla, 2024-10-30:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Z Offset with ABL (BIQU_MICROPROBE_V2) has no effect on the physical height of the nozzle at specified Z Values.
i.e. Offset of -2 and -10  both result in nozzle crashing into bed at Z = 1.

This is a separate issue but I will mention it here because it is possibly related: I tried to use PROBE_OFFSET_WIZARD but it gave strange undefined behavior even after wiping EEPROM and reinitializing so I disabled it. I would use it to find a z offset around -1.4 and when I hit done the Z Offset would then show ~36.x offset.. All subsequent "G29" procedures after this action resulted in "Probe Failed". 

### Bug Timeline

Oldest commit I saw this behavior was 70ef7978b418691904b1fe70f6c1009295eb5f7f

### Expected behavior

I expect that with a z probe offset of -2, my nozzle is too high off the build plate when I do `G0 Z0` but Nozzle crashes at Z = 1. I would expect that with a Z probe offset of -5, my nozzle is WAY too high off the build plate when I do `G0 Z0` but nozzle crashes at Z = 1 again. I have `ENABLE_LEVELING_AFTER_G28` defined, and it appears to work as intended because bed leveling always shows "on" in the menus.

### Actual behavior

Different Z probe offset values make no difference to height of nozzle at Z=0.

### Steps to Reproduce

```
;  ##### Starting from a fresh state:

M502
//action:notification Hardcoded Default Settings Loaded
echo:Hardcoded Default Settings Loaded
ok

M500
echo:Settings Stored (682 bytes; crc 28459)
//action:notification Settings Stored
ok

; ##### Retrieving from EEPROM to make sure settings look good from config.h
M501
echo:V90 stored settings retrieved (682 bytes; crc 2845941
//action:notification Stored settings retrieved
echo:; Linear Units:
echo:  G21 ; (mm)
echo:; Temperature Units:
echo:  M149 C ; Units in Celsius
echo:; Filament settings (Disabled):
echo:  M200 S0 D1.75
echo:; Steps per unit:
echo:  M92 X80.00 Y80.00 Z400.00 E430.17
echo:; Max feedrates (units/s):
echo:  M203 X800.00 Y800.00 Z5.00 E25.00
echo:; Max Acceleration (units/s2):
echo:  M201 X800.00 Y800.00 Z100.00 E5000.00
echo:; Acceleration (units/s2) (P<print-accel> R<retract-accel> T<travel-accel>)              :
echo:  M204 P800.00 R500.00 T800.00
echo:; Advanced (B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> J              <junc_dev>):
echo:  M205 B20000.00 S0.00 T0.00 J0.08
echo:; Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:; Auto Bed Leveling:
echo:  M420 S0 Z10.00 ; Leveling OFF
echo:; Material heatup parameters:
echo:  M145 S0 H185.00 B45.00 F255
echo:  M145 S1 H240.00 B110.00 F255
echo:; Hotend PID:
echo:  M301 P21.73 I1.54 D76.55
echo:; Bed PID:
echo:  M304 P50.71 I9.88 D173.43
echo:; Controller Fan:
[... 253 lines cut ...]
### Other add-ons

BIQU MicroProbe V2, Sprite Direct Drive Extruder, Phaetus Dragon UHF

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

Other (explain below)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin.zip](https://github.com/user-attachments/files/17568114/Marlin.zip)

I embedded a bunch of GCode above that can be utilized for testing (in steps to reproduce section).

Nuck-TH, 2024-10-31:
I reproduce it on Kingroon KP3s with current bugfix.
Changing offset directly, using wizard or babystepping it doesn't change first layer thickness at all, it is always the same.

xaviervilla, 2024-11-01:
Interesting. Babystep z offset live during the first layer **did** work for me, but saving to EEPROM after the print did not solve the issue when I went to reprint. 

I spent a significant amount of time trying to figure out if it was me doin something wrong but I’m pretty confident now that there is a bug here. This is my first time ever using ABL though so maybe I’m missing something. 

In the meantime, I may try manually moving to the Z height I need in my g code and then setting the current position to zero before my print. I believe that should be a temporary workaround for people who need to print.

ThomasToka, 2024-11-01:
Have you tried to enable those two altready?

```
#define BABYSTEP_ALWAYS_AVAILABLE
#define BABYSTEP_ZPROBE_OFFSET
```

Nuck-TH, 2024-11-01:
BABYSTEP_ZPROBE_OFFSET was already enabled, BABYSTEP_ALWAYS_AVAILABLE did nothing to change situation.

xaviervilla, 2024-11-01:
Apologies, I was adjusting babystep during my first layer, **not** `BABYSTEP_ZPROBE_OFFSET`. I have `BABYSTEP_ZPROBE_OFFSET` disabled.  I could enable it and try that too but I would need to make some custom GCODE because I've pretty much destroyed my bed trying to sort this out.. I have a little time to dig into the code and check recent commits now.

xaviervilla, 2024-11-01:
Looks like this was removed. Looking into why.
```
        // Unapply the offset because it is going to be immediately applied
        // and cause compensation movement in Z
        const float fade_scaling_factor = TERN(ENABLE_LEVELING_FADE_HEIGHT, planner.fade_scaling_factor_for_z(current_position.z), 1);
        current_position.z -= fade_scaling_factor * bilinear_z_offset(current_position);
```

xaviervilla, 2024-11-01:
Okay after looking through the G29 code, I decided to try `G29 Z5` and that made the nozzle 5MM off the bed when Z showed zero on the build plate. And my Mesh shows:

``` 
      0      1      2      3      4
 0 +5.307 +5.170 +5.150 +5.148 +5.280
 1 +5.175 +5.035 +5.010 +5.028 +5.090
 2 +5.145 +5.015 +5.012 +4.990 +5.035
 3 +5.193 +5.052 +5.043 +5.045 +5.080
 4 +5.195 +5.160 +5.145 +5.133 +5.140
```

I believe this is to be expected from the documentation I read.

I don't see anywhere in the G29 code that looks like it is applying the offset from the EEPROM/config value, but only from the gcode command itself. Is this expected?

I'm thinking I should look into the M851 code now.

robherc, 2024-11-02:
> I expect that with a z probe offset of -2, my nozzle is too high off the build plate when I do G0 Z0 but Nozzle crashes at Z = 1. I would expect that with a Z probe offset of -5, my nozzle is WAY too high off the build plate when I do G0 Z0 but nozzle crashes at Z = 1 again.

I haven't read through your configs yet, but I ### did###  notice a slight problem with your logic here. The more negative you set your PROBE_Z_OFFSET, the further BELOW the prob's trigger-point you're telling Marlin to move to get the nozzle to Z=0.

Will you please attempt setting your Z_PROBE_OFFSET to +2mm; then re-home; then `G1 X110 Y110 Z0.1 F1500`
If the Z_PROBE_OFFSET is being applied, that should most DEFINITELY end motion with ths nozzle a few mm above your bed surface.

xaviervilla, 2024-11-02:
> Will you please attempt setting your Z_PROBE_OFFSET to +2mm; then re-home; then `G1 X110 Y110 Z0.1 F1500` If the Z_PROBE_OFFSET is being applied, that should most DEFINITELY end motion with ths nozzle a few mm above your bed surface.

I did try positive values as well, but I’ll give it a shot again and respond with what happens. Thanks for the feedback

xaviervilla, 2024-11-02:
@robherc 
I initially followed your instructions and then got the same behavior I mentioned. 

```
; Probe the bed
G29 
...
      0      1      2      3      4
 0 +0.253 +0.138 +0.123 +0.112 +0.255
 1 +0.148 +0.018 -0.005 -0.007 +0.065
 2 +0.120 -0.005 +0.003 -0.048 -0.018
 3 +0.140 +0.035 +0.028 +0.000 +0.060
 4 +0.235 +0.135 +0.118 +0.128 +0.125

X:235.00 Y:221.00 Z:11.55 E:0.00 Count X:18800 Y:17680 Z:4620
ok

; Set Positive Z probe offset
M851 Z2
ok

M500
echo:Settings Stored (682 bytes; crc 49771)
//action:notification Settings Stored
ok

; Move to bed center
G1 X110 Y110 Z5 F1500
echo:busy: processing
ok

; Incrementally lower nozzle
G1 Z3
ok
G1 Z2
ok
G1 Z1
; Nozzle is already pressed into the bed
```

And then I re-read what you said. You said to **home again** after setting the Z offset. I tried again, homing right after setting the Z offset and to my disbelief it worked. Persists reboots as well. The final z probe offset value I needed was -1.4. First print is going down great now.

### Retrospective:
I feel very dumb for not figuring that out on my own. Looking through my command history and even the gcode in the original post, I was not homing each time between my tests. Oddly, the prints that tried trying were homing so I don't know why those changes didn't seem like they were applying, but it was probably something I was doing wrong. My other source of confusion is that I thought I was babystepping Z probe offset and storing to EEPROM during my prints. But I was actually just adjusting regular babystep Z which is different and does not get stored to EEPROM at all.

### Summary:
Users must home after setting the z probe offset for it to be applied and take effect. While I am tempted to close this issue, I'll leave it open for a dev to close at their discretion with this closing remark:

On the [Marlin Documentation on M851](https://marlinfw.org/docs/gcode/M851.html), there is no mention that homing is required after setting the Z offset. In fact, in the videos I watched online, there was also no mention of this. Is this new? If this is the desired behavior I believe that it should definitely be documented better on the M851 page. Otherwise, I think it makes the most sense for M851 to automatically apply the Z offset immediately.

Thank you to all who helped.

robherc, 2024-11-02:
> Users must home after setting the z probe offset for it to be applied and take effect.
Actually, this is because you disabled the babystepping for z_probe_offset. When that is enabled, while it may still be recommendable to re-home after the adjustment, it is generally not necessary, as the printer is babystepping to "live-apply" the new Z-Offset as you adjust the setpoint.

closed 2024-11-02 by ellensp