MarlinFirmware/Marlin #27506: [FR] (feature summary) Support for urboot on AVRs

Laikulo, 2024-10-31:
### Describe the feature you want

PIO environments for urboot for at least melzi, and a bump to the version of avrdude specified (to one that support urclock protocol)

### Additional context

In the quest to cram too many features on a creality melzi board, urboot saves 256b over optiboot, and 1.75k over the stock arduino (sanguino) configuration.

ellensp, 2024-10-31:
URBOOT uses minimal 256 bytes, max 512 bytes. 

But the smallest boot section you can define with the 1284p micro controller fuses bits is 512 words, ie 1K

So there is zero benefit to use this bootloader you cannot use the 'saved' space as it is locked out by the fuse bits.

closed 2024-10-31 by ellensp

ellensp, 2024-10-31:
the atmega2560 is the same

[image]