MarlinFirmware/Marlin #27512: [FR] ESP32-S2 SUPPORT

ahakkoca07, 2024-11-04:
### Is your feature request related to a problem? Please describe.

I want to compile Marlin for ESP32-S2 Mini board, but it seems like marlin doesn't officially supports it.

### Are you looking for hardware support?

ESP32-S2 Mini

### Describe the feature you want

ESP32-S2 Mini support.

### Additional context

I searched the web for ESP32-S2 support on Marlin, and i only found an [open issue](https://github.com/MarlinFirmware/Marlin/issues/23507) on it, with no updates. I already wrote the pins file for the ESP32-S2, and [someone on the other issue](https://github.com/MarlinFirmware/Marlin/issues/23507#issuecomment-1022979050) already started writing a HAL for it.

ellensp, 2024-11-04:
Please don't open up duplicate feature requests.

closed 2024-11-04 by ellensp

ahakkoca07, 2024-11-04:
Sorry, i saw the other issue was inactive, so i thought i have to open a new issue.