MarlinFirmware/Marlin #27513: [BUG] Menu homing feedrate displays incorrectly

ellensp, 2024-11-05:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

The Menu homing feedrate displays incorrectly

### Bug Timeline

since https://github.com/MarlinFirmware/Marlin/pull/27456 

### Expected behavior

This menus should list each axis and its feedrate using " LSTR MSG_HOMING_FEEDRATE_N            = _UxGT("@ Homing Feedrate");"

### Actual behavior

This menus list the same axis over but lists the correct feedrate
I've seen 3  lines of the label X  Homing Feedrate, and 3 lines of the label Y  Homing Feedrate

### Steps to Reproduce

1.  Install Emulator example config
2. Add #define EDITABLE_HOMING_FEEDRATE
3. Build and run
4. look at the menu, not it is not displaying the axis label correctly.

### Version of Marlin Firmware

Bugfix-2.1.x

### Printer model

Simulated

### Electronics

Simulatoed

### LCD/Controller

REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Sim Configuration.zip](https://github.com/user-attachments/files/17626888/Sim.Configuration.zip)

ellensp, 2024-11-05:
a quick fix  (but I dropped the Inch support)  

```diff
diff --git a/Marlin/src/lcd/menu/menu_configuration.cpp b/Marlin/src/lcd/menu/menu_configuration.cpp
index d7eeab3cb3..3042e2f7b5 100644
--- a/Marlin/src/lcd/menu/menu_configuration.cpp
+++ b/Marlin/src/lcd/menu/menu_configuration.cpp
@@ -408,22 +408,8 @@ void menu_advanced_settings();
     START_MENU();
     BACK_ITEM(MSG_HOMING_FEEDRATE);
 
-    #if ENABLED(MENUS_ALLOW_INCH_UNITS)
-      #define _EDIT_HOMING_FR(A) do{ \
-        const float maxfr = MMS_TO_MMM(planner.settings.max_feedrate_mm_s[_AXIS(A)]); \
-        editable.decimal = A##_AXIS_UNIT(homing_feedrate_mm_m.A); \
-        EDIT_ITEM(float5, MSG_HOMING_FEEDRATE_N, &editable.decimal, \
-          A##_AXIS_UNIT(10), A##_AXIS_UNIT(maxfr), []{ \
-          homing_feedrate_mm_m.A = parser.axis_value_to_mm(_AXIS(A), editable.decimal); \
-        }); \
-      }while(0);
-    #else
-      #define _EDIT_HOMING_FR(A) do{ \
-        EDIT_ITEM(float5, MSG_HOMING_FEEDRATE_N, &homing_feedrate_mm_m.A, 10, MMS_TO_MMM(planner.settings.max_feedrate_mm_s[_AXIS(A)])); \
-      }while(0);
-    #endif
-
-    MAIN_AXIS_MAP(_EDIT_HOMING_FR);
+    LOOP_NUM_AXES(a)
+      EDIT_ITEM_FAST_N(float5, a, MSG_HOMING_FEEDRATE_N, &homing_feedrate_mm_m[a], 10, MMS_TO_MMM(planner.settings.max_feedrate_mm_s[a]));
 
     END_MENU();
   }
```

ellensp, 2024-11-05:
refactor idea.   

There a number of strings of the form _X, _Y, _Z,_E   and _N   

But if we have _N  why do we have the rest when they could be created in code?

classicrocker883, 2024-11-06:
I thought it was `homing_feedrate_mm_m.a`, not `homing_feedrate_mm_m[a]`, how does that work?

ellensp, 2024-11-06:
union

```cpp
template<typename T>
struct XYZval {
  union {
    #if NUM_AXES
      struct { NUM_AXIS_CODE(T x, T y, T z, T i, T j, T k, T u, T v, T w); };
      struct { NUM_AXIS_CODE(T X, T Y, T Z, T I, T J, T K, T U, T V, T W); };
      struct { NUM_AXIS_CODE(T a, T b, T c, T _i, T _j, T _k, T _u, T _v, T _w); };
      struct { NUM_AXIS_CODE(T A, T B, T C, T II, T JJ, T KK, T UU, T VV, T WW); };
    #endif
    T pos[NUM_AXES];
  };
```

and homing_feedrate_mm_m.a cannot work as a is an interger 

no such things as homing_feedrate_mm_m.0 or homing_feedrate_mm_m.1 etc
its homing_feedrate_mm_m.x, homing_feedrate_mm_m.y etc

classicrocker883, 2024-11-08:
or how about:

```y
      #define _EDIT_HOMING_FR(A)  \
        EDIT_ITEM_FAST_N(float5, _AXIS(A), MSG_HOMING_FEEDRATE_N, &homing_feedrate_mm_m.x, MMS_TO_MMM(planner.settings.min_feedrate_mm_s), MMS_TO_MMM(planner.settings.max_feedrate_mm_s[_AXIS(A)]));
    #endif
    ```

ellensp, 2024-11-08:
that uses &homing_feedrate_mm_m.x, so only the x homing feedrate value is edited

classicrocker883, 2024-11-08:
thanks, with that edit it works perfectly

ellensp, 2024-11-08:
Extended it to also work in inch mode and PR created

closed 2024-11-08 by ellensp