MarlinFirmware/Marlin #27517: [BUG] M851 Z-Offset is ignored

rotarypower, 2024-11-07:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I'm upgrading a old printer with BLtouch and a BTT SKR1.4 Board.
No matter what I set my NOZZLE_TO_PROBE_OFFSET to the Z value is always ignored. When changing X and Y changes are applied but no matter what I set the Z value to the bed will always move to the same physical location.
[Config.zip](https://github.com/user-attachments/files/17669542/Config.zip)

### Version of Marlin Firmware

2.1.2.4

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2024-11-08:
Please test the [`bugfix-2.1.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

rondlh, 2024-11-08:
> no matter what I set the Z value to the bed will always move to the same physical location.

Where is this location?

rotarypower, 2024-11-08:
It is 3.8mm below the nozzle. When sending M114 the Z value reads 10

ellensp, 2024-11-08:
NOZZLE_TO_PROBE_OFFSET values are stored in eeprom,

When the firmware is first run the eeprom is populated with the default firmware settings at that time.

Since you dont have EEPROM_INIT_NOW set, any attempts to update NOZZLE_TO_PROBE_OFFSET in configuration.h are ignored. The first set eeprom values take precedence. 

Either enable EEPROM_INIT_NOW so that the eeprom data is updated to Config file default every time you upload new firmware

or

Force a eeprom to new config values with M502 - Factory Reset then M500 - Save Settings after every new firmware upload.

rotarypower, 2024-11-08:
I'm sorry I forgot to mention that I did M502 -> M500 after the firmware update.
But changing the offset via M851 has no effect either, altough values are updated correctly when checking with M503

rondlh, 2024-11-08:
Your Z-offset is "#define NOZZLE_TO_PROBE_OFFSET { -10, 0, -3.8 }"
So I would expect that your bed moves to 3.8mm below the nozzle (like mentioned), then the Z-probe is triggered, and Z homing is complete. Afterwards the bed will move up a bit (Z_AFTER_PROBING 10). So everything seems to work as expected.

The key now is, where does the bed go when you move to Z=0?

Typically the procedure I use is like this:
1. Set Z-offset to 0: M851 Z0
2. Heat the bed and nozzle to typical printing temperatures
3. Home the printer: G28
4. Move the probe to the centre of the bed (if not there already after G28)
5. Slowly bring the nozzle and bed closer with babystepping (M290Z-0.01) (Use a sheet of paper)
6. You should see you Z-offset changing (M851) because you have BABYSTEP_ZPROBE_OFFSET enabled.
7. Now save your Z-offset (M500)

rotarypower, 2024-11-13:
I'm sorry for not getting back to you for a couple of days.
The issue has been resolved, I re-did the configurations from scratch, that solved it. Must have had a mistake in there

closed 2024-11-13 by rotarypower