MarlinFirmware/Marlin #27524: [BUG] M221 / Flow Rate setting lower then 100 sets the flow way too low

EvgenyD, 2024-11-09:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Setting flow rate value (via M221 or LCD interface) lower then 100 results in the extrusion rate being extremely slow.
Flow rate higher then 100 work correctly.

### Expected behavior

Setting flow rate to 99 should not produce noticible change in flow rate.

### Actual behavior

Setting flow rate to 99 result in a flow rate way too low, with extrusion motor nearly not moving.
it feels like the flow rate is set to 0.99% instead of 99%.

### Steps to Reproduce

1. start print with flow rate set to 100
2. while printing set flow rate to a vlau lower then 100 (e.g. 99) through the menu or via M221 command
3. almost no material coming out of nozzle and the extruder motor not moving. 

### Version of Marlin Firmware

bugfix-2.1.x

### Printer model

Kossel Delta

### Electronics

Trigorilla upgraded for TMC drivers

### LCD/Controller

default, REPRAP_DISCOUNT_SMART_CONTROLLER

### Other add-ons

TMC2208 drivers (Hybrid Mode) for all the motors. UART control.

### Bed Leveling

MBL Manual Bed Leveling

### Your Slicer

Prusa Slicer

### Host Software

Pronterface

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin.zip](https://github.com/user-attachments/files/17689581/Marlin.zip)

ellensp, 2024-11-10:
As start provided config has a few issues, 

has #define MANUAL_PROBE_START_Z 0.5  and    #define MANUAL_PROBE_START_Z 0.8
causing thousands of redefinition errors 

change #define MANUAL_PROBE_START_Z 0.5 to
```
  #ifndef MANUAL_PROBE_START_Z
    #define MANUAL_PROBE_START_Z 0.5  // (mm) Comment out to use the last-measured height
  #endif

```

This line is invalid as HOMING_FEEDRATE_Z is not defined.
#define Z_PROBE_FEEDRATE_FAST (HOMING_FEEDRATE_Z / 2) //(4*60)

this is wrong
```
// #if PROBE_SELECTED && !IS_KINEMATIC
  #define PROBING_MARGIN_LEFT 50//PROBING_MARGIN
  #define PROBING_MARGIN_RIGHT 10//PROBING_MARGIN
  #define PROBING_MARGIN_FRONT 10//PROBING_MARGIN
  #define PROBING_MARGIN_BACK 10//PROBING_MARGIN
// #endif
```
Since this  Machine is IS_KINEMATIC these defines are also removed by marlin in  https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/Marlin/src/inc/Conditionals-5-post.h#L3265-L3269

There is no need disable the additional axis stuff,  with the driver disabled it is just ignored.

Here are updated and cleaned configs 
[Cleaned Configuration.zip](https://github.com/user-attachments/files/17689861/Cleaned.Configuration.zip)

ellensp, 2024-11-10:
"Setting flow rate value (via M221 or LCD interface) " 

There is no way to set flow rate on this lcd

Not without setting 
    //#define ULTIPANEL_FEEDMULTIPLY  // Encoder sets the feedrate multiplier on the Status Screen
    #define ULTIPANEL_FLOWPERCENT // Encoder sets the flow percentage on the Status Screen

Which is not in your config.

Fr on main display is feedrate multiplier

ellensp, 2024-11-10:
I setup a test rig, used your config as much as possible (only changed motherboard to BOARD_RAMPS_14_EFB (same chipset) and change steps/mm for my hardware)

I made a simple bash script to generate test gcode  

```bash
E_VALUE=200
flowrate=100
echo "M302 S0; Allow cold extrusion"
echo "G28; Home all axes"
echo "G1 Z5 F1000; Move nozzle to 5mm above bed"
for i in {1..200}
do
  echo "M117 Flow set to $flowrate"
  echo "M221 S$flowrate"
  for j in {1..1}
  do
    echo "G1 X50 Y50 E$E_VALUE  F1000"
    E_VALUE=$((E_VALUE + 200))
    echo "G1 X-50 Y-50 E$E_VALUE F1000"
    E_VALUE=$((E_VALUE + 200))
  done
  echo "G92 E0; Reset extruder position"
  E_VALUE=200
  flowrate=$((RANDOM % 101 + 50))
done
```  

It generated lots of gcode like this but longer  (DO NOT RUN THIS ON A REAL PRINTER) 

```
M302 S0; Allow cold extrusion
G28; Home all axes
G1 Z180 F1000; Move nozzle to 180mm above bed
M117 Flow set to 100
M221 S100
G1 X50 Y50 E200  F1000
G1 X-50 Y-50 E400 F1000
G92 E0; Reset extruder position
M117 Flow set to 121
M221 S121
G1 X50 Y50 E200  F1000
G1 X-50 Y-50 E400 F1000
G92 E0; Reset extruder position
M117 Flow set to 140
M221 S140
G1 X50 Y50 E200  F1000
G1 X-50 Y-50 E400 F1000
G92 E0; Reset extruder position
M117 Flow set to 87
M221 S87
G1 X50 Y50 E200  F1000
G1 X-50 Y-50 E400 F1000

```
And M221 does just what it was meant to, it adjusted the extruder flow rate by a percentage.

EvgenyD, 2024-11-10:
Thank you for looking into the issue.
I am still working on configuration files - i'm trying to upgrade from 1.9 version to 2.1 - so there are still some junk left here and there, and I appreciate the help with the config files.
for the flow rate issue here is the video comparison of the extruder motor speed setting 99-100-101% from LCD menu:
https://github.com/user-attachments/assets/f4b95cef-bac5-4ebd-9f20-e2dfe2abedc0

I am not sure which variable exactrly is set through the menu, it is called "Flow" and the same behaviour occures using M221 command (M221 S99 and so on), so i assumed it is the flow rate:

[image]

ellensp, 2024-11-10:
Ok that one mystery solved

The tune menu is only available while printing.  
Need to add M75 to my test gcode to enable that

Tested on my test rig using Flow menu item still works as expected.

I cannot replicate this issue

closed 2025-02-23 by github-actions[bot]