MarlinFirmware/Marlin #27525: Anycubic Kobra Neo Firmware files

DecXdrones, 2024-11-10:
### Is your feature request related to a problem? Please describe.

I've been searching for a month now how to upgrade an customize my Anycubic Kobra Neo firmware but can't seem to find any set of files that I can compile. There are currently no files available to work with the TriGorilla GenV3.0.6 board online.

### Are you looking for hardware support?

Board is the TriGorilla Gen V3.0.6

### Describe the feature you want

I would like a Marlin configuration with which I can work and change the configuration.h file to modify my firmware

thisiskeithb, 2024-11-10:
Duplicate of https://github.com/MarlinFirmware/Configurations/issues/649

closed 2024-11-10 by thisiskeithb

ellensp, 2024-11-10:
and duplicate of https://github.com/MarlinFirmware/Marlin/issues/25266

ellensp, 2024-11-10:
You cannot create config files for hardware that is not yet supported.

DecXdrones, 2024-11-10:
I am sorry if my question seems stupid but my knowledge around programming and firmware is very limited. Could you someone explain how come this hardware isn't supported ? or what it would take for this board to be supported by Marlin ? This board comes stock with a Marlin config so I thought it would be possible to just change the configuration.h file and compile to get a running firmware.bin

ellensp, 2024-11-10:
Stock Marlin uses platformio to build firmware. Platformio has extremely  limited (community added) support for  Hc32f460kct6 processor. 

At present only two board are supported with this chip
BOARD_CREALITY_ENDER2P_V24S4  and BOARD_AQUILA_V101

Anycubic got around this by using the commercial compiler Keil and just bypassed platformio entirely.

So firstly this board needs to be added to Marlin, this includes the boards pin.h file (which is provided)
And it requires a build environment like these https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/ini/hc32.ini
which may or may not require building an entire framework and aurduno such as these
https://github.com/shadow578/platform-hc32f46x
https://github.com/shadow578/framework-hc32f46x-ddl
https://github.com/shadow578/framework-arduino-hc32f46x

I know at present that these do not support tmc stepper drivers or FLASH_EEPROM_EMULATION, your board needs both of these, so minimally these features need added.

Then there is the display, once again this is a non supported display. The entire display code needs to be backported into main marlin branch

The machine also seems to have custom leveling hardware.. this would also need backported into stock marlin.

Only once this is done can a working Configuration files be added

DecXdrones, 2024-11-10:
Okay! thank you for the detailed answer.

I did try to compile with Keil compiler and managed to do so but the printer started having weirds behaviours once I installed the firmware such as requiring a minimum temp of 230 C before extruding eventhough the configuration.h file clearly asks for 170 C. At that point I figured it was a problem with Keil and that VSCode was a better option. I guess we're back to the drawing board.

Thanks again