MarlinFirmware/Marlin #27526: [BUG] Unable to build after commit #27502

Crazy-Charles, 2024-11-10:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Build does not complete

### Bug Timeline

After changes to generic_create_variant.py in commit #27502

### Expected behavior

Linker to finish and produce a bin file

### Actual behavior

Compiles fine but fails on linker with....
```
arm-none-eabi-g++: error: D:/Github: No such file or directory
arm-none-eabi-g++: error: Repositories/Marlin/buildroot/share/PlatformIO/variants/MARLIN_F103Rx/ldscript.ld: No such file or directory
*** [.pio\build\STM32F103RE_creality\firmware.elf] Error 1
```

First error is correct as my path is actually D:\Github Repositories\

### Steps to Reproduce

Download current bugfix
Use the Ender 3 V2 config files (no changes made)
Build

### Version of Marlin Firmware

Bugfix 2.1.x

### Printer model

Ender 3 V2

### Electronics

Creality 4.2.7

### LCD/Controller

Dwin

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

Config files from https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.1.x with no changes made

If i comment out the last line of generic_create_variant.py it will complete the build.

`board.update("build.variants_dir", "buildroot/share/PlatformIO/variants");`

Crazy-Charles, 2024-11-10:
Still getting the same error

thisiskeithb, 2024-11-10:
Which OS and specific version are you running?

Crazy-Charles, 2024-11-10:
Windows 11 Pro 23H2 build 22631.4391

It would seem that the script doesn't like the space in the folder name, if i remove it then the build completes but this would cause a great inconvenience as that structure has been fine for the last few years!

Crazy-Charles, 2024-11-10:
@thisiskeithb 
If i change your script at the end to 
```
current_working_directory = Path.cwd()
board.update("build.variants_dir", current_working_directory / os.path.join("buildroot", "share", "PlatformIO", "variants"))
```

The build completes ok for me but don't know of any ramifications this might cause.

thisiskeithb, 2024-11-10:
I can confirm that spaces in the build path break on Windows 10 & 11 22H2 as well.

fnsign, 2024-11-11:
The same bug occurs also on macOS (Sequoia 15.1)

thisiskeithb, 2024-11-11:
> The same bug occurs also on macOS (Sequoia 15.1)

Interesting. I can't reproduce it on Sonoma.

rondlh, 2024-11-11:
I have the same issue (Win 10), the problem started from a very recent update

thinkyhead, 2024-11-12:
It looks like the problem / solution relates to having a full path to the current folder. Is that all that is needed?

thinkyhead, 2024-11-12:
If the above (merged) PR fixes the issue then we can close this. If it is still a problem we can apply some more patches. I think it's already better than it was, either way.

thisiskeithb, 2024-11-13:
Please download [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code and let us know if you're still having this issue.

fnsign, 2024-11-13:
On macOS (Sequoia 15.1) it is fixed. Thanks!

Crazy-Charles, 2024-11-13:
Has fixed my issue on Win 11 too

closed 2024-11-13 by thisiskeithb