MarlinFirmware/Marlin #27529: [BUG] Inconsistent/suboptimal types in ExtUI

rondlh, 2024-11-13:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Not really a bug, but just something I found in the ExtUI code that could possibly be improved.
Some functions in ExtUI are inconsistent with the internal function calls used:

celsius_float_t getTargetTemp_celsius(const heater_t);
celsius_float_t getTargetTemp_celsius(const extruder_t);

Target temps are of type celsius_t(int16_t), not celsius_float_t.

  float getTargetFan_percent(const fan_t fan) {
    UNUSED(fan);
    return TERN0(HAS_FAN, thermalManager.fanSpeedPercent(fan - FAN0));
  }

  static uint8_t fanSpeedPercent(const uint8_t fan)          { return ui8_to_percent(fan_speed[fan]); }

So the function returns a float, but internally it just gets a uint8_t.

------------------------------------

  float getActualFan_percent(const fan_t fan) {
    UNUSED(fan);
    return TERN0(HAS_FAN, thermalManager.scaledFanSpeedPercent(fan - FAN0));
  }

   static uint8_t scaledFanSpeedPercent(const uint8_t fan)    { return ui8_to_percent(scaledFanSpeed(fan)); }

So the function returns a float, but internally it just gets a uint8_t.

I tested the code with the suggested changes, nothing breaks, everything works as expected.

### Bug Timeline

Not a bug

### Expected behavior

More efficient code

### Actual behavior

Odd code

### Steps to Reproduce

Just have a look at the code given above

### Version of Marlin Firmware

Lastest

### Printer model

Custom

### Electronics

MKS Monster8 V1.0

### LCD/Controller

BTT TFT35 V3.0,

### Other add-ons

MKS EBB42 CAN Toolhead, BLTouch 3.1

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

Just a standard config file, nothing changed in here.
[Dummy Config.zip](https://github.com/user-attachments/files/17728317/Dummy.Config.zip)

classicrocker883, 2024-11-13:
just a helpful hint, it would read better by using 
````
Three " ` " before and after blocks of code

```
float getActualFan_percent(const fan_t fan) {
UNUSED(fan);
return TERN0(HAS_FAN, thermalManager.scaledFanSpeedPercent(fan - FAN0));
}
```

Single " ` " before and after single line of code

`celsius_float_t getTargetTemp_celsius(const heater_t);`

````

---

result:
```
float getActualFan_percent(const fan_t fan) {
UNUSED(fan);
return TERN0(HAS_FAN, thermalManager.scaledFanSpeedPercent(fan - FAN0));
}
```

`celsius_float_t getTargetTemp_celsius(const heater_t);`

classicrocker883, 2024-11-13:
are these the changes you mean?

```diff
- celsius_float_t getActualTemp_celsius(const heater_t);
- celsius_float_t getActualTemp_celsius(const extruder_t);
- celsius_float_t getTargetTemp_celsius(const heater_t);
- celsius_float_t getTargetTemp_celsius(const extruder_t);
- float getActualFan_percent(const fan_t);
- float getTargetFan_percent(const fan_t);

+ celsius_t getActualTemp_celsius(const heater_t);
+ celsius_t getActualTemp_celsius(const extruder_t);
+ celsius_t getTargetTemp_celsius(const heater_t);
+ celsius_t getTargetTemp_celsius(const extruder_t);
+ uint8_t getActualFan_percent(const fan_t);
+ uint8_t getTargetFan_percent(const fan_t);
```

rondlh, 2024-11-13:
@classicrocker883 Thanks, great tip, I didn't know about the "y", nice!
I can make a pull request, but perhaps it's better to get some feedback first, because this change can affect lots of people.

@classicrocker883 Something like you mention, but getActualTemp_celsius should use floats.
> ```diff
> - celsius_float_t getActualTemp_celsius(const heater_t);
> - celsius_float_t getActualTemp_celsius(const extruder_t);
> ```

closed 2024-11-23 by thinkyhead