MarlinFirmware/Marlin #27530: [BUG] Not defining BABYSTEP_ALWAYS_AVAILABLE causes steppers and controller fan to run continueslly after baby stepping

rondlh, 2024-11-13:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

If BABYSTEP_ALWAYS_AVAILABLE is not defined, then in babystep.cpp the stepper timeout is not reset:

  TERN_(BABYSTEP_ALWAYS_AVAILABLE, gcode.reset_stepper_timeout());

This causes the steppers to stay enabled without timeout and the controller fan to run continuously after using baby stepping. So if you use baby stepping after a print to quickly adjust the Z-offset then there is a side effect.

### Bug Timeline

Unknown

### Expected behavior

Stepper timeout and controller fan timeout as defined, no unexpected side effects of apparently unrelated option (BABYSTEP_ALWAYS_AVAILABLE).

### Actual behavior

Stepper stay activated, controller fan run continuously after using baby stepping.

### Steps to Reproduce

1. Don't define BABYSTEP_ALWAYS_AVAILABLE (configuration_adv.h)
2. Define BABYSTEPPING (configuration_adv.h)
3. Define USE_CONTROLLER_FAN (use a low IDLE time: ### CONTROLLERFAN_IDLE_TIME 5 and DEFAULT_STEPPER_TIMEOUT_SEC 10). (configuration_adv.h)
4. Make some baby steps and check if the steppers stay active and the controller fan keeps running. 

The code shows what happens, babystep.cpp

TERN_(BABYSTEP_ALWAYS_AVAILABLE, gcode.reset_stepper_timeout());

The stepper timer will not be reset, which causes the steppers to stay enabled, and the controller fan to run continuously when BABYSTEP_ALWAYS_AVAILABLE is not defined.

### Version of Marlin Firmware

Latest bugfix

### Printer model

Custom

### Electronics

MKS Monster8 V1.0

### LCD/Controller

BTT TFT35 V3.0

### Other add-ons

MKS EBB42 CAN Toolhead, BLTouch 3.1

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

Simplified configuration for testing:

[Configuration_adv.zip](https://github.com/user-attachments/files/17728995/Configuration_adv.zip)

classicrocker883, 2024-11-13:
my guess is it cannot do babysteps after it has timed-out. that is why `BABYSTEP_ALWAYS_AVAILABLE` resets the stepper timeout.
because those functions have `if (!can_babystep(axis)) return;` 
and `can_babystep()` returns `(ENABLED(BABYSTEP_WITHOUT_HOMING) || !axis_should_home(axis));`

so in your testing, try having `BABYSTEP_WITHOUT_HOMING` enabled

rondlh, 2024-11-13:
@classicrocker883 Thanks for your response.
> my guess is it cannot do babysteps after it has timed-out. that is why `BABYSTEP_ALWAYS_AVAILABLE` resets the stepper timeout. because those functions have `if (!can_babystep(axis)) return;` and [...]

I just did a test with BABYSTEP_ALWAYS_AVAILABLE undefined:
1. Baby stepping doesn't work (as expected, BABYSTEP_WITHOUT_HOMING is not defined)
2. G28(home) then wait for the controller fan to switch off (after stepper timeout).
3. Baby stepping works fine, the controller fan and steppers are activated, but... they never switch off afterwards.

So it seems baby stepping works fine even after the stepper/controller fan timeout.

BABYSTEP_WITHOUT_HOMING doesn't make much sense in my book, but I guess there are use cases.

classicrocker883, 2025-02-20:
just checking up on how it's going, fixed or still issue?

closed 2025-06-05 by github-actions[bot]