MarlinFirmware/Marlin #27543: [BUG] Cannot build TFT_LVGL_UI with ADVANCED_PAUSE_FEATURE

ellensp, 2024-11-24:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

A basic config for TFT_LVGL_UI with ADVANCED_PAUSE_FEATURE will not build
It error with #error "ADVANCED_PAUSE_FEATURE requires a supported LCD controller (or EMERGENCY_PARSER)."
This is fair enough, but doesnt seem accurate, you look in 

Marlin/src/lcd/extui/mks_ui/draw_dialog.cpp or Marlin/src/lcd/extui/mks_ui/draw_pause_message.cpp
You find code references to ADVANCED_PAUSE_FEATURE

Anyway if you enable EMERGENCY_PARSER it bypasses this error only to give build errors

Either way this has ssues.

### Bug Timeline

Unknown

### Expected behavior

It should build

### Actual behavior

Marlin/src/lcd/extui/mks_ui/draw_dialog.cpp:133:7: error: 'pause_menu_response' was not declared in this scope; did you mean 'pause_menu_disp'?
  133 |       pause_menu_response = PAUSE_RESPONSE_EXTRUDE_MORE;
      |       ^~~~~~~~~~~~~~~~~~~

### Steps to Reproduce

1. Use provided Config files
2. build
3. since it has EMERGENCY_PARSER this will error.

### Version of Marlin Firmware

Bugfix-2.1.x

### Electronics

BOARD_MKS_ROBIN_NANO_V3_1

### LCD/Controller

MKS_TS35_V2_0

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration.zip](https://github.com/user-attachments/files/17892466/Configuration.zip)

ellensp, 2024-11-24:
untested hack that allows building

```diff
diff --git a/Marlin/src/feature/pause.cpp b/Marlin/src/feature/pause.cpp
index fd01e85284..d5b8728c80 100644
--- a/Marlin/src/feature/pause.cpp
+++ b/Marlin/src/feature/pause.cpp
@@ -86,10 +86,10 @@

 static xyze_pos_t resume_position;

-#if M600_PURGE_MORE_RESUMABLE
+//#if M600_PURGE_MORE_RESUMABLE
   PauseMenuResponse pause_menu_response;
   PauseMode pause_mode = PAUSE_MODE_PAUSE_PRINT;
-#endif
+//#endif

 #if ENABLED(CONFIGURE_FILAMENT_CHANGE)
   fil_change_settings_t fc_settings[EXTRUDERS];
diff --git a/Marlin/src/feature/pause.h b/Marlin/src/feature/pause.h
index 5ee2e50ea7..74880c4a85 100644
--- a/Marlin/src/feature/pause.h
+++ b/Marlin/src/feature/pause.h
@@ -56,7 +56,6 @@ enum PauseMessage : char {
   PAUSE_MESSAGE_COUNT
 };

-#if M600_PURGE_MORE_RESUMABLE
   // Input methods can Purge More, Resume, or request input
   enum PauseMenuResponse : char {
     PAUSE_RESPONSE_WAIT_FOR,
@@ -65,7 +64,7 @@ enum PauseMessage : char {
   };
   extern PauseMenuResponse pause_menu_response;
   extern PauseMode pause_mode;
-#endif
+//#endif

 typedef struct FilamentChangeSettings {
   #if ENABLED(CONFIGURE_FILAMENT_CHANGE)
diff --git a/Marlin/src/inc/Conditionals-4-adv.h b/Marlin/src/inc/Conditionals-4-adv.h
index b6981e59a2..60d72f5acd 100644
--- a/Marlin/src/inc/Conditionals-4-adv.h
+++ b/Marlin/src/inc/Conditionals-4-adv.h
@@ -988,7 +988,7 @@
 #if ANY(MIN_SOFTWARE_ENDSTOPS, MAX_SOFTWARE_ENDSTOPS)
   #define HAS_SOFTWARE_ENDSTOPS 1
 #endif
-#if ANY(EXTENSIBLE_UI, IS_NEWPANEL, EMERGENCY_PARSER, HAS_ADC_BUTTONS, HAS_DWIN_E3V2)
+#if ANY(EXTENSIBLE_UI, IS_NEWPANEL, EMERGENCY_PARSER, HAS_ADC_BUTTONS, HAS_DWIN_E3V2, HAS_TFT_LVGL_UI)
   #define HAS_RESUME_CONTINUE 1
 #endif

diff --git a/Marlin/src/lcd/extui/mks_ui/draw_dialog.cpp b/Marlin/src/lcd/extui/mks_ui/draw_dialog.cpp
index 4b2474a8e6..cc5e83e14c 100644
--- a/Marlin/src/lcd/extui/mks_ui/draw_dialog.cpp
+++ b/Marlin/src/lcd/extui/mks_ui/draw_dialog.cpp
@@ -46,7 +46,7 @@
   #include "../../../feature/powerloss.h"
 #endif

-#if ENABLED(PARK_HEAD_ON_PAUSE)
+#if ENABLED(PARK_HEAD_ON_PAUSE) || ENABLED(ADVANCED_PAUSE_FEATURE)
   #include "../../../feature/pause.h"
 #endif
```

thinkyhead, 2024-12-04:
Basically, `ADVANCED_PAUSE_FEATURE` is not compatible with MKS UI. They implemented their own handling for filament runout as part of MKS UI. I pointed out to @makerbase-mks or one of the other MKS contributors that this would break and would not get better with new updates, but it was never re-implemented properly. Although some changes might allow it to build with `ADVANCED_PAUSE_FEATURE`, it still will not work.

closed 2024-12-04 by thinkyhead