MarlinFirmware/Marlin #27545: [BUG] Strange sound from Z-axis after upgrading to 2.1.2.5

rmoravcik, 2024-11-24:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I decided upgrade from 2.1.2 to 2.1.2.5 and stepper motors on Z-axis started making weird sound, like running in different mode previously.

I'm using Z axis with two stepper drivers/motors (TMC2208).

Here is an video of moving Z-axis with 2.1.2 - https://photos.app.goo.gl/qAkcPh99wJj5B2p67 and with 2.1.2.5 - https://photos.app.goo.gl/w1JPVApw3Pvoa34r7
[Configuration.zip](https://github.com/user-attachments/files/17893830/Configuration.zip)

Did I miss some new configuration option?

Thank you,
Roman

### Version of Marlin Firmware

2.1.2.5

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

rmoravcik, 2024-11-24:
Here are my configuration files [Configuration.zip](https://github.com/user-attachments/files/17893831/Configuration.zip)

thisiskeithb, 2024-11-24:
Please download [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code and let us know if you're still having this issue.

rmoravcik, 2024-11-24:
> Please download [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code and let us know if you're still having this issue.

Sure, I will try.

I was just testing all versions from 2.1.2 to 2.1.2.5 and 2.1.2.3 is working fine. 2.1.2.4 started behaving strangely.

rmoravcik, 2024-11-24:
It seems my problem is caused by this commit https://github.com/MarlinFirmware/Marlin/pull/27149. So, it's duplicate of https://github.com/MarlinFirmware/Marlin/issues/27341

closed 2024-11-24 by rmoravcik