MarlinFirmware/Marlin #27552: [FR] Switch between thermistor temp tables "on-the-fly"

classicrocker883, 2024-11-30:
### Describe the feature you want

Basically a Gcode command and/or display menu option that can swap the extruder's thermistor.
Lets focus on 1 single extruder.
Load say 2 or more `temptable_*`s and swap them in `heater_ttbl_map`/`heater_ttbllen_map`
This may have to involve some tricks, like pretend there's 2 extruders.

### Additional context

the benefit is if you change a hotend and dont wish to recompile. that you have it already stored in memory.
only feasible if temp tables don't actually take up much room. can load 1 or 2 or more additional thermistors.

rondlh, 2024-12-05:
That would be a nice feature.
I have 2 printers with a CAN bus host and tool head, both host and tool head are running Marlin.
The tool head controls the hotend heater and reads measures the thermistor resistance, so the thermistor is defined in the tool head, This means that I have to update the tool head firmware if I change the thermistor to another type (different hotend). Doing this without recompiling and flashing the firmware would be much easier.

The storage required for a thermistor table depends on the number of ADC-temp pairs in the table. One lines takes 4 bytes (2 words). Thermistor 1 is 68 lines, which takes 272 bytes or program memory. This number could probably be reduced.

It seems Klipper is very efficient with the thermistor table, It only uses 3 temperatures and resistor values to characterize a thermistor (see temperature_sensors.cfg). They probably calculate the thermistor β (beta) from the 3 values to completely characterize the thermistor. I see they even have thermistor definitions that only have 1 entry (temp and resistance) and they provide the beta directly. So a complete thermistor table would only be a few bytes, math can do the rest.

classicrocker883, 2025-04-03:
anyone have any ideas of how to implement this?