MarlinFirmware/Marlin #27553: Build error using mks tinybee board

moriamoria, 2024-11-30:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Hello,

after my MKS skr 1.4 died, I have order a tinybee motherboard.

Whan I change the motherboard in Marlin, I have a buld error

  //#define MOTHERBOARD BOARD_RAMPS_14_EFB
  #define MOTHERBOARD BOARD_MKS_TINYBEE

PS C:\Users\mat\Documents\matthieu\3D\orthur\m2.1.bugfix.1\Marlin> platformio run --silent -e mks_tinybee
'"CC"' n'est pas reconnu en tant que commande interne
ou externe, un programme ex\x82cutable ou un fichier de commandes.
Error: Failed to parse Marlin features. See previous error messages.

Many thanks if you can help me to use my new tinybee board
[Marlin.zip](https://github.com/user-attachments/files/17965587/Marlin.zip)

### Bug Timeline

new bug

### Expected behavior

Buld should be ok and deliver a firmware.bin

### Actual behavior

error when building

### Steps to Reproduce

create folder ./marlin.2.1.bugfix
git clone https://github.com/MarlinFirmware/Marlin.git
open vscode
show ABM Panel

### Version of Marlin Firmware

[cron] Bump distribution date (2024-11-28)

### Printer model

ortur

### Electronics

tinybee

### LCD/Controller

REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

### Other add-ons

2209

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

I have done the same steap with https://github.com/makerbase-mks/MKS-TinyBee/tree/main

and I can build firmwre (but this is version 2.0)

moriamoria, 2024-11-30:
Hello,

Following previous request, I have by dichotomie try to find a point of failure:

1. Marlin 2.1 OK
2. Marlin 2.1.2  ok
3. Marlin 2.1.2.3 ok
4. Marlin 2.1.2.5 ok
5. 📝 Update HAL documentation, README 24/11/2024 07:00:32 ok
6. marlin  30/11/2024 21:25:55 📌 Update, pin LPC PIO framework (#27554) ko

For the moment, I don't find reason, look CC is c compiler ? 

Thanks for your help

Regards

ellensp, 2024-12-01:
seem to be windows specific, no issues on linux, but I can replicate on windows

ellensp, 2024-12-01:
In https://github.com/MarlinFirmware/Marlin/commit/f2d585ac7f5dad29aa57112ea327812f91924088

a section was added

```python
                # Skip '*-elf-g++' (crosstool-NG)
                if not gpath.stem.endswith('-elf-g++'):
                    gccpath = str(gpath.resolve())
                    break
```

But the compiler for esp32 is a -elf-g++   .platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-g++

So it cannot find the the compiler nd default to CC and fails.

@thinkyhead is there a more exact string you can look for to exclude this crosstool-NG

GonzoMDC, 2024-12-01:
Made above changes and now get Invalid syntax error. Latest Marlin 2.1.X bug fix, Windows 10, Vs code PIP and MAB all up to date. Compiling for Neptune 2S with MKS Tinybee.
[image]

ellensp, 2024-12-01:
the code should end up looking like 

```python
                # Skip '*-elf-g++' (crosstool-NG)
                if not "xtensa32" not in str(gpath) and gpath.stem.endswith('-elf-g++'):
                    gccpath = str(gpath.resolve())
                    break
```

GonzoMDC, 2024-12-02:
I mad a mistake on the additions to the code. This fix DOES work. Thank you so very much for your time and effort

ellensp, 2024-12-02:
A PR has been created to fix this.

closed 2024-12-02 by ellensp