MarlinFirmware/Marlin #27555: [BUG] (bug summary)Marlin 2.1.3 Beta 1 - MKS Robin Mini (Kingroon KP3) Z stepper motor moves with strong vibration and loud noise, but works normally with earlier version 2.0.9.1

Hokeric, 2024-11-30:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Z motor not moving as expected. Upon homing, after X and Y have homed, Z moves very slowly and with shaking loud noise even with the default motor strength 800. 

I tried to increase Z motor strength to 1000 and performed move Z axis from LCD control, but it did not move. I then increased motor strength to 1200 and then moved Z axis by 10mm via LCD control, it appeared to move correctly. However, after I stored the configuration, the Z motor began to make loud noise again. 

I tested the printer with older version 2.0.9.1 and it worked normally. Therefore, it is not hardware issue.

### Bug Timeline

New bug. This bug is observed in 2.1.3 beta 1, as well as in 2.1.2.5

### Expected behavior

Z motor should home normally, but it does not.

### Actual behavior

Z motor not moving as expected. Upon homing, after X and Y have homed, Z begins to move but with strong vibration and loud noise. 

### Steps to Reproduce

1. Perform a homing action.

### Version of Marlin Firmware

2.1.3 beta 1 and 2.1.2.5 both have the same problem

### Printer model

Kingroon KP3 (board is MKS Robin Mini)

### Electronics

Stock electronics

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration Files.zip](https://github.com/user-attachments/files/17966383/Configuration.Files.zip)

thisiskeithb, 2024-11-30:
> I here also upload config files of version 2.0.9.1 with works correctly

Have you tried any other versions between `2.0.9.1` & `2.1.2.5`?

Hokeric, 2024-12-01:
Yes, 2.0.9.1 has no problem, it works correctly. But 2.1.2.5 has the same problem as 2.3.1-b1

ellensp, 2024-12-01:
2.0.9.1 is from Jun 28, 2021 and 2.1.2.5 is a few selected patches off of 2.1.2.1 dated May 17, 2023, that's 2 years of code changes to go through.

ellensp, 2024-12-01:
you need to try a few versions in between 

2.0.9.1
2.0.9.2
2.0.9.3
2.0.9.4
2.0.9.5
2.0.9.6
2.0.9.7
2.0.9.8
2.1
2.1.0.1
2.1.0.2
2.1.1
2.1.1.1
2.1.1.2
2.1.2
2.1.2.1

Hokeric, 2024-12-01:
I tried 2.0.9.7 and 2.1.0.1, but was unable to compile. Could this be compatibility issue with VS Code? I do wish to find out since which version it has the problem.

Hokeric, 2024-12-01:
This is the compilation error while compiling 2.1.0.1

[image]

ellensp, 2024-12-01:
TEMP_SENSOR_IS_MAX is not in Configuration.h https://github.com/MarlinFirmware/Configurations/blob/2.1/config/examples/Kingroon/KP3/Configuration.h

It is however in https://github.com/MarlinFirmware/Configurations/blob/2.1.1/config/examples/Kingroon/KP3/Configuration.h, but this is the wrong Configuration file

Hokeric, 2024-12-01:
Used the 2.1 Configuration.h to compile, but has this error.
[image]

ellensp, 2024-12-01:
change ini/stm32f1.ini line 281  https://github.com/MarlinFirmware/Marlin/blob/2.1/ini/stm32f1.ini#L281 

"-DMCU_STM32F103VE -DTIMER_TONE=TIM3 -DTIMER_SERVO=TIM2" 
to
"-DMCU_STM32F103VE -DTIMER_TONE=TIM3 -DTIMER_SERVO=TIM1"

Hokeric, 2024-12-01:
Was to compile 2.1 after the .ini file change. The Z motor has the same problem.

It is likely that the error began after transition from 2.0.x to 2.1.x.

Tried to test 2.0.9.7, but had compilation error, as follows:
[image]

ellensp, 2024-12-01:
same error as 2.1

one line lower 
https://github.com/MarlinFirmware/Marlin/blob/2.0.9.7/ini/stm32f1.ini#L282

Hokeric, 2024-12-01:
Was able to compile 2.0.9.7 after the .ini file change, but Z motor has the same problem.
[image]

Tried to test with 2.0.9.2, but had compilation error, as follows:

Hokeric, 2024-12-01:
Was to to compile 2.0.9.5, and the Z motor has the same problem.

Hokeric, 2024-12-03:
I have narrowed down the troubleshooting to the extent that I can. 
Is there any other input that I can provide to help troubleshooting at your side?
Thank you.

Hokeric, 2025-03-13:
Added comment as a follow-up after receiving bot notification of stagnant closing.