MarlinFirmware/Marlin #27569: [BUG] (bug summary)SD card causes blank screen MKS Tinybee and Mini12864V3

chebhou, 2024-12-05:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

whenever i set the `#define SDCARD_CONNECTION ONBOARD`  and plug an SD card the lcd goes blank and the beeper silent, the board still functions normally through web interface.

If i dont set `#define SDCARD_CONNECTION ONBOARD` (any other option)  the LCD will still work if insert an SD card but the SD card will either fail to init, or fail during print, or fail to upload throught he web interface any file larger than ( 130kb so far).

sd card ( 2GB and 4GB) both formatted with the recommended sd formatter , and with windows (fat32  4byte/16/32) with no success.

the SD card slots on the board and on the LCD both act the same o matter what settings i put.

I've tried 2.1.3b1 and latest bugfix 

### Bug Timeline

multiple marlin versions

### Expected behavior

SD card to work with LCD

### Actual behavior

LCD blank when SD card inserted

### Steps to Reproduce

make these changes to the config files 

 ```
#define MOTHERBOARD BOARD_MKS_TINYBEE

#define BAUDRATE 115200
#define SERIAL_PORT_2 -1
#define BAUDRATE_2 115200

#define MKS_MINI_12864_V3

#define ESP3D_WIFISUPPORT
#define SDCARD_CONNECTION ONBOARD
```
compile, upload and test

### Version of Marlin Firmware

Bugfix-2.1.x

### Printer model

Custom

### Electronics

MKS TINYBEE

### LCD/Controller

MKS MINI 12864 V3

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin.zip](https://github.com/user-attachments/files/18020892/Marlin.zip)

KillerInk, 2025-01-09:
It do work here on a mks tinybee v1 and  Mini12864V3.
When i un/plug it the screen goes black and then return with the message sd card insert/removed.
Im using a sandisk sdhc 16gig class 4

pls check if you also enabled #define SDSUPPORT in Configuration.h

PanosPetrou, 2025-01-13:
I can confirm this too, I have to keep #define SDCARD_CONNECTION disabled, otherwise TinyBee hangs on boot

KillerInk, 2025-01-14:
I can confirm that too with `#define SDCARD_CONNECTION  ONBOARD` blacks out the screen after the Marlin Boot Logo
Using `LCD` does work.

closed 2025-08-08 by github-actions[bot]