MarlinFirmware/Marlin #27571: [FR] make Configuration.h and Configuration_adv.h "re-constructible" from an existing firmware .bin file.

gitcnd, 2024-12-05:
### Is your feature request related to a problem? Please describe.

I need to tweak some minor settings, but the original firmware source configuration files do not match the (mostly working) firmware I'm currently using.

In my case, my "stock" firmware ends at 0x384a0 but my rebuilt-from-source (identical version) is 0x2e354, indicating some significant differences; if re-creating of the config files was possible from my existing firmware image (stm32 extracted via j-link), I'd know with certainty that  I'm not going to overwrite important things when I re-flash a new build...

### Are you looking for hardware support?

N/A

### Describe the feature you want

If all the Configuration.h and Configuration_adv.h variables were written to the end of the firmware file (when there's free space to accommodate them), it would be *extremely* helpful for several use-cases (debugging, rebuilding, tweaking, etc).

This could either be something "code intensive" that finds all `#define` etc settings, and puts them into the firmware, or even something as simple as literally doing a "zip" on the actual Configuration.h and Configuration_adv.h files and appending the resulting PK... binary onto the end of the firmware.  (currently, those files are 94kb zipped; which is less than half of the free space on my Monster8 board)...

### Additional context

I understand some boards might not have enough free flash space; but this feature could either be a new config setting itself, or it could be conditional on the resulting firmware having enough available free space to accommodate it.

ellensp, 2024-12-06:
@gitcnd this already exists

```
  /**
   * Enable this option if you have more than ~3K of unused flash space.
   * Marlin will embed all settings in the firmware binary as compressed data.
   * Use 'M503 C' to write the settings out to the SD Card as 'mc.zip'.
   * See docs/ConfigEmbedding.md for details on how to use 'mc-apply.py'.
   */
  //#define CONFIGURATION_EMBEDDING 
```

closed 2024-12-06 by ellensp

rondlh, 2024-12-06:
Have a look at CONFIG_EXPORT in Configuration_adv.h

I guess you want to attach a file like that.

gitcnd, 2024-12-06:
Pardon my ignorance!

So I guess what I'm really saying is, that the logic for "#define CONFIGURATION_EMBEDDING " needs to be reversed:

The people who need this, are everyone who does not have the original source code (not the people who built it to begin with), so when space permits, "CONFIGURATION_EMBEDDING" should be the **default** behavior, unless the developer has explicitly chosen to "#define OMIT_CONFIGURATION_EMBEDDING".

Observing reality: new dev boards and upgrades just keep coming - hardly anyone goes to the effort of porting new Marlin to old boards (and putting their work someplace permanent that others can easily find), so expecting this to remain the status-quo moving forwards, making "CONFIGURATION_EMBEDDING" the **default** action (where space permits; which will basically be "always", because new boards have way more space than old ones used to) to retain these critical parameters means that everyone in future should have everything they need to be able to work out how to update their own boards.  "Future Proofing", as it were...

ellensp, 2024-12-06:
No.

We are not one of these awful companies that take and scrape all data they can from their users unless you take steps to opt out

If you want to embed your Config files, do so, you cannot  force this on others.