MarlinFirmware/Marlin #27574: Skr 1.3 and tinybee exp1, exp2 documentation fault.

moriamoria, 2024-12-08:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Hello,

After a falure of my motherboard, I have migrate from SKR 1.3 to tinybee.

My REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER is working with SKR 1.3 and not on tinybee.

The reason is the inversion of the rubun IDC 10 pins needed  for tinybee.

Is it possible to add a comment in Marlin\src\pins\esp32\pins_MKS_TINYBEE.h line 123 that REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER ribon need inversion ?

Is it possible to change in Marlin\src\pins\lpc1768\pins_BTT_SKR_V1_3.h line 202 the definition of the idc pinour from : 

/**
 *                 ------                                     ------
 *  (BEEPER) 1.30 | 1  2 | 0.28 (BTN_ENC)        (MISO) 0.17 | 1  2 | 0.15 (SCK)
 *  (LCD_EN) 1.18 | 3  4 | 1.19  (LCD_RS)     (BTN_EN1) 3.26 | 3  4 | 0.16 (SD_SS)
 *  (LCD_D4) 1.20   5  6 | 1.21  (LCD_D5)     (BTN_EN2) 3.25   5  6 | 0.18 (MOSI)
 *  (LCD_D6) 1.22 | 7  8 | 1.23  (LCD_D7)   (SD_DETECT) 1.31 | 7  8 | RESET
 *            GND | 9 10 | 5V                            GND | 9 10 | --
 *                 ------                                     ------
 *                  EXP1                                       EXP2
 */

to

/**
 *                 ------                                     ------
 *  (BEEPER) 1.30 | 1  2 | 0.28 (BTN_ENC)        (MISO) 0.17 | 1  2 | 0.15 (SCK)
 *  (LCD_EN) 1.18 | 3  4 | 1.19  (LCD_RS)     (BTN_EN1) 3.26 | 3  4 | 0.16 (SD_SS)
 *  (LCD_D4) 1.20   5  6 | 1.21  (LCD_D5)     (BTN_EN2) 3.25   5  6 | 0.18 (MOSI)
 *  (LCD_D6) 1.22 | 7  8 | 1.23  (LCD_D7)   (SD_DETECT) 1.31 | 7  8 | RESET
 *            GND | 9 10 | 5V                            GND | 9 10 | --
 *                 ------                                     ------
 *                  EXP1                                       EXP2
 */

in any case, i am lost with the comments.

the only thing I know is that lcd connected to skr 1.3 and tinybee request to invert cable, what a mess.

[image]

### Bug Timeline

all the time

### Expected behavior

comments should follow pin numbet of idc 10 pin notation.

### Version of Marlin Firmware

any version

### Printer model

ortur 4 v1

### Electronics

skr 1.3 and tinybee

### LCD/Controller

REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2024-12-08:
Unfortunately its not that simple

If your lcd comes with right angle plugs, (and some do) then the lcd is correct for the TINYBEE, but not the BTT_SKR_V1_3

Sellers of these screens sell it with an appropriate adapter for the it to work on a ramps, beyond that it buyer beware.

thinkyhead, 2024-12-08:
Once we have the LCD Pins refactor completed there will be a warning anytime the plugs are reversed between the display and the board. That PR is pending and will be applied for the next release, after 2.1.3.

closed 2025-04-26 by github-actions[bot]