MarlinFirmware/Marlin #27575: [BUG] SCARA axis homing

Sejkorka, 2024-12-08:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Hi, I have a problem with marlin and its SCARA function, it's my graduation paper, I have already printed the complete scara, most things work, if not at least half.
But what doesn't work is the homing of the "X" axis, the limit switches work normally and M119 shows them as triggers when they should be. 
The "X" axis does not homing, it seems to me that once the "Y" axis is homed so is the "X" axis.

[Config adv.txt](https://github.com/user-attachments/files/18053981/Config.adv.txt)
[Config.txt](https://github.com/user-attachments/files/18053982/Config.txt)

### Version of Marlin Firmware

2.0.9

### Bed Leveling

None

### Your Slicer

None

### Host Software

Repetier Host

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

thinkyhead, 2024-12-09:
This part of the code doesn't get enough attention. When I was last working on a SCARA style robot it had a single free arm, so it had to home both its first and second joints. I've never seen the Morgan and MP SCARA robots do their homing, but the provided code is supposed to have done the right thing at one time.

Anyway, it sounds like you'll need to dig into the homing code and debug it to work for your machine, or recruit some help to figure it out. The Marlin Discord server is a pretty good place to find help. Once you have it sorted out it would be great if you could submit the fix to the main project. SCARA is a rare thing, but we do want it to work!

ellensp, 2024-12-10:
MP_SCARA is quite broken in bugfix 2.1.x

I Enabled debugging 
#define DEBUG_LEVELING_FEATURE
M111 S32  

All the following G28 test where done on a freshly reset board.  

G28 gives me this log on a standard G28, Z axis  moves up and then it errors..

G28 
18:54:25.501 > >>> G28  X0.00 Y199.07 Z0.00
18:54:25.501 > Machine Type: SCARA
18:54:25.501 > Probe: NONE
18:54:25.501 > remember_feedrate_scaling_off: fr=66.67 100%
18:54:25.501 > Raise Z before homing:
18:54:25.501 > do_z_clearance(5.00 [0.00 to 5.00], 0)
18:54:25.501 > do_blocking_move_to_z(5.00, 4.00)
18:54:25.501 > >>> do_blocking_move_to  X0.00 Y199.07 Z0.00
18:54:25.501 > >  X0.00 Y199.07 Z5.00
18:54:25.501 >   destination= X0.00 Y199.07 Z5.00 : prepare_fast_move_to_destination
18:54:25.501 >   destination= X0.00 Y199.07 Z5.00 : prepare_fast_move_to_destination
18:54:27.481 > echo:busy: processing
18:54:27.858 > <<< do_blocking_move_to  X0.00 Y199.07 Z5.00
18:54:27.858 >   current_position= X0.00 Y0.00 Z5.00 : sync_plan_position
18:54:27.858 > do_blocking_move_to_xy(-300.00, -300.00, 28.28)
18:54:27.858 > >>> do_blocking_move_to  X0.00 Y0.00 Z5.00
18:54:27.858 > >  X-300.00 Y-300.00 Z5.00
18:54:27.858 > <<< do_blocking_move_to  X0.00 Y0.00 Z5.00
18:54:27.858 > echo:Homing Failed
18:54:27.858 > Error:Printer halted. kill() called!

G28 Z works as expected

G28 Z
19:08:29.178 > >>> G28  X0.00 Y199.07 Z0.00
19:08:29.178 > Machine Type: SCARA
19:08:29.178 > Probe: NONE
19:08:29.178 > remember_feedrate_scaling_off: fr=66.67 100%
19:08:29.178 > >>> homeaxis(Z)
19:08:29.178 > Home Fast: -150.00mm
19:08:29.178 > >>> do_homing_move  X0.00 Y199.07 Z0.00
19:08:29.178 > ...(Z, -150.00, [4.00])
19:08:29.178 >   current_position= X0.00 Y199.07 Z0.00 : sync_plan_position
19:08:31.158 > echo:busy: processing
...
19:08:59.142 > echo:busy: processing
19:08:59.377 > <<< do_homing_move  X0.00 Y199.07 Z-150.00
19:08:59.378 > Move Away: 3.00mm
19:08:59.378 > >>> do_homing_move  X0.00 Y199.07 Z-150.00
19:08:59.378 > ...(Z, 3.00, [4.00])
19:08:59.378 >   current_position= X0.00 Y199.07 Z0.00 : sync_plan_position
19:09:00.928 > <<< do_homing_move  X0.00 Y199.07 Z3.00
19:09:00.928 > Re-bump: -6.00mm
19:09:00.928 > >>> do_homing_move  X0.00 Y199.07 Z3.00
19:09:00.928 > ...(Z, -6.00, 1.00)
19:09:00.928 >   current_position= X0.00 Y199.07 Z0.00 : sync_plan_position
19:09:01.151 > echo:busy: processing
19:09:03.148 > echo:busy: processing
19:09:04.182 > <<< do_homing_move  X0.00 Y199.07 Z-6.00
19:09:04.182 > >>> set_axis_is_at_home(Z)
19:09:04.182 >   current_position= X0.00 Y199.07 Z0.00 : 
19:09:04.182 > <<< set_axis_is_at_home(Z)
19:09:04.182 >   current_position= X0.00 Y199.07 Z0.00 : sync_plan_position
19:09:04.182 > <<< homeaxis(Z)
19:09:04.182 >   current_position= X0.00 Y199.07 Z0.00 : sync_plan_position
19:09:04.182 > >>> do_move_after_z_homing  X0.00 Y199.07 Z0.00
19:09:04.182 > do_z_clearance(5.00 [0.00 to 5.00], 1)
19:09:04.182 > do_blocking_move_to_z(5.00, 4.00)
19:09:04.182 > >>> do_blocking_move_to  X0.00 Y199.07 Z0.00
19:09:04.182 > >  X0.00 Y199.07 Z5.00
19:09:04.182 >   destination= X0.00 Y199.07 Z5.00 : prepare_fast_move_to_destination
19:09:04.201 >   destination= X0.00 Y199.07 Z5.00 : prepare_fast_move_to_destination
19:09:05.144 > echo:busy: processing
19:09:06.554 > <<< do_blocking_move_to  X0.00 Y199.07 Z5.00
19:09:06.554 > <<< do_move_after_z_homing  X0.00 Y199.07 Z5.00
19:09:06.554 > restore_feedrate_and_scaling: fr=66.67 100%
19:09:06.554 > X:0.00 Y:199.07 Z:5.00 E:0.00 Count A:2581B:2581 Z:8000
19:09:06.554 > SCARA Theta:90.01  Psi:90.01
19:09:06.554 > 
19:09:06.554 > <<< G28  X0.00 Y199.07 Z5.00
19:09:06.554 > ok

G28 Y  moves Z up 5mm as expected, but then does nothing. 

G28 Y
19:11:21.900 > >>> G28  X0.00 Y199.07 Z0.00
19:11:21.900 > Machine Type: SCARA
19:11:21.900 > Probe: NONE
19:11:21.900 > remember_feedrate_scaling_off: fr=66.67 100%
19:11:21.900 > Raise Z before homing:
19:11:21.900 > do_z_clearance(5.00 [0.00 to 5.00], 0)
19:11:21.900 > do_blocking_move_to_z(5.00, 4.00)
19:11:21.900 > >>> do_blocking_move_to  X0.00 Y199.07 Z0.00
19:11:21.900 > >  X0.00 Y199.07 Z5.00
19:11:21.900 >   destination= X0.00 Y199.07 Z5.00 : prepare_fast_move_to_destination
19:11:21.900 >   destination= X0.00 Y199.07 Z5.00 : prepare_fast_move_to_destination
19:11:23.865 > echo:busy: processing
19:11:24.254 > <<< do_blocking_move_to  X0.00 Y199.07 Z5.00
19:11:24.254 >   current_position= X0.00 Y199.07 Z5.00 : sync_plan_position
19:11:24.254 > restore_feedrate_and_scaling: fr=66.67 100%
19:11:24.254 > X:0.00 Y:199.07 Z:5.00 E:0.00 Count A:2581B:2581 Z:8000
19:11:24.254 > SCARA Theta:90.01  Psi:90.01
19:11:24.254 > 
19:11:24.254 > <<< G28  X0.00 Y199.07 Z5.00
19:11:24.254 > ok

G28 X  moves Z up 5mm as expected, but then does nothing. 

G28 X

19:18:41.893 > >>> G28  X0.00 Y199.07 Z0.00
19:18:41.893 > Machine Type: SCARA
19:18:41.893 > Probe: NONE
19:18:41.893 > remember_feedrate_scaling_off: fr=66.67 100%
19:18:41.893 > Raise Z before homing:
19:18:41.893 > do_z_clearance(5.00 [0.00 to 5.00], 0)
19:18:41.893 > do_blocking_move_to_z(5.00, 4.00)
19:18:41.893 > >>> do_blocking_move_to  X0.00 Y199.07 Z0.00
19:18:41.893 > >  X0.00 Y199.07 Z5.00
19:18:41.893 >   destination= X0.00 Y199.07 Z5.00 : prepare_fast_move_to_destination
19:18:41.893 >   destination= X0.00 Y199.07 Z5.00 : prepare_fast_move_to_destination
19:18:43.857 > echo:busy: processing
19:18:44.247 > <<< do_blocking_move_to  X0.00 Y199.07 Z5.00
19:18:44.247 >   current_position= X0.00 Y199.07 Z5.00 : sync_plan_position
19:18:44.247 > restore_feedrate_and_scaling: fr=66.67 100%
19:18:44.247 > X:0.00 Y:199.07 Z:5.00 E:0.00 Count A:2581B:2581 Z:8000
19:18:44.247 > SCARA Theta:90.01  Psi:90.01
19:18:44.247 > 
19:18:44.247 > <<< G28  X0.00 Y199.07 Z5.00
19:18:44.247 > ok

Im not sure that G28 X or Y should actually do anything since there isn't really a X or a Y endstop.
But G28 should most definely home all axies.

ellensp, 2024-12-10:
disabling #define VALIDATE_HOMING_ENDSTOPS  does allow G28 to complete, but does not move X or Y steppers

G28 
19:43:25.835 > >>> G28  X0.00 Y199.07 Z0.00
19:43:25.835 > Machine Type: SCARA
19:43:25.835 > Probe: NONE
19:43:25.835 > remember_feedrate_scaling_off: fr=66.67 100%
19:43:25.835 > Raise Z before homing:
19:43:25.835 > do_z_clearance(5.00 [0.00 to 5.00], 0)
19:43:25.835 > do_blocking_move_to_z(5.00, 4.00)
19:43:25.835 > >>> do_blocking_move_to  X0.00 Y199.07 Z0.00
19:43:25.835 > >  X0.00 Y199.07 Z5.00
19:43:25.835 >   destination= X0.00 Y199.07 Z5.00 : prepare_fast_move_to_destination
19:43:25.835 >   destination= X0.00 Y199.07 Z5.00 : prepare_fast_move_to_destination
19:43:27.800 > echo:busy: processing
19:43:28.194 > <<< do_blocking_move_to  X0.00 Y199.07 Z5.00
19:43:28.194 >   current_position= X0.00 Y0.00 Z5.00 : sync_plan_position
19:43:28.194 > do_blocking_move_to_xy(-300.00, -300.00, 28.28)
19:43:28.194 > >>> do_blocking_move_to  X0.00 Y0.00 Z5.00
19:43:28.194 > >  X-300.00 Y-300.00 Z5.00
19:43:28.194 > <<< do_blocking_move_to  X0.00 Y0.00 Z5.00
19:43:28.194 > >>> homeaxis(Z)
19:43:28.194 > Home Fast: -150.00mm
19:43:28.194 > >>> do_homing_move  X0.00 Y0.00 Z5.00
19:43:28.194 > ...(Z, -150.00, [4.00])
19:43:28.194 >   current_position= X0.00 Y0.00 Z0.00 : sync_plan_position
19:43:29.807 > echo:busy: processing
19:43:31.804 > echo:busy: processing
19:43:32.452 > <<< do_homing_move  X0.00 Y0.00 Z-150.00
19:43:32.453 > Move Away: 3.00mm
19:43:32.453 > >>> do_homing_move  X0.00 Y0.00 Z-150.00
19:43:32.453 > ...(Z, 3.00, [4.00])
19:43:32.453 >   current_position= X0.00 Y0.00 Z0.00 : sync_plan_position
19:43:33.810 > echo:busy: processing
19:43:34.003 > <<< do_homing_move  X0.00 Y0.00 Z3.00
19:43:34.003 > Re-bump: -6.00mm
19:43:34.003 > >>> do_homing_move  X0.00 Y0.00 Z3.00
19:43:34.003 > ...(Z, -6.00, 1.00)
19:43:34.003 >   current_position= X0.00 Y0.00 Z0.00 : sync_plan_position
19:43:35.808 > echo:busy: processing
19:43:37.172 > <<< do_homing_move  X0.00 Y0.00 Z-6.00
19:43:37.172 > >>> set_axis_is_at_home(Z)
19:43:37.172 >   current_position= X0.00 Y0.00 Z0.00 : 
19:43:37.172 > <<< set_axis_is_at_home(Z)
19:43:37.172 >   current_position= X0.00 Y0.00 Z0.00 : sync_plan_position
19:43:37.172 > <<< homeaxis(Z)
19:43:37.172 >   current_position= X0.00 Y0.00 Z0.00 : sync_plan_position
19:43:37.172 > >>> do_move_after_z_homing  X0.00 Y0.00 Z0.00
19:43:37.172 > do_z_clearance(5.00 [0.00 to 5.00], 1)
19:43:37.172 > do_blocking_move_to_z(5.00, 4.00)
19:43:37.172 > >>> do_blocking_move_to  X0.00 Y0.00 Z0.00
19:43:37.172 > >  X0.00 Y0.00 Z5.00
19:43:37.172 >   destination= X0.00 Y0.00 Z5.00 : prepare_fast_move_to_destination
19:43:37.191 >   destination= X0.00 Y0.00 Z5.00 : prepare_fast_move_to_destination
19:43:37.798 > echo:busy: processing
19:43:39.545 > <<< do_blocking_move_to  X0.00 Y0.00 Z5.00
19:43:39.545 > <<< do_move_after_z_homing  X0.00 Y0.00 Z5.00
19:43:39.545 > restore_feedrate_and_scaling: fr=66.67 100%
19:43:39.545 > X:0.00 Y:0.00 Z:5.00 E:0.00 Count A:-2147483648B:-2147483648 Z:8000
19:43:39.545 > SCARA Theta:-74893488.00  Psi:-74893488.00
19:43:39.545 > 
19:43:39.545 > <<< G28  X0.00 Y0.00 Z5.00
19:43:39.545 > ok

ellensp, 2024-12-10:
This answer that question, only G28 Z is action

```cpp
  void homeaxis(const AxisEnum axis) {

    #if ANY(MORGAN_SCARA, MP_SCARA)
      // Only Z homing (with probe) is permitted
      if (axis != Z_AXIS) { BUZZ(100, 880); return; }
```

ellensp, 2024-12-10:
I want back to 2.0.5 to find one that can home, and I notice that is_scara is not set for the mp_scara, 

ie 2.0.5 does not limit homeaxis to Z_AXIS
``` 
void homeaxis(const AxisEnum axis) {

  #if IS_SCARA
    // Only Z homing (with probe) is permitted
    if (axis != Z_AXIS) { BUZZ(100, 880); return; }
  #else
```

So  I regressed bugfix 2.1.x to just the following and allowed it to home X and Y

``` 
#if ENABLED(MORGAN_SCARA)  
```
And the large arm actually homed, but very very slowly. then after homing it flung itself wild to the other end of travel.
So it seems like the axis is not being setup correctly.   
... 

no solution yet.

shimaode, 2024-12-18:
I commented out the return code for SCARA
```
  void homeaxis(const AxisEnum axis) {

    #if ANY(MORGAN_SCARA, MP_SCARA)
      // Only Z homing (with probe) is permitted
      // if (axis != Z_AXIS) { BUZZ(100, 880); return; }
```
And in my case, the x axis can home in normal speed but it's not guaranteed to back to home position every time, sometimes it just stops before it even hit the endstop. And the y axis is going wild in an unnormal speed.

Looking into the homing code, this should be causing the problem:
```
const float move_length = 1.5f * max_length(TERN(DELTA, Z_AXIS, axis)) * axis_home_dir;
if (DEBUGGING(LEVELING)) DEBUG_ECHOLNPGM("Home Fast: ", move_length, "mm");
do_homing_move(axis, move_length, 0.0, !use_probe_bump);
```
This commands the head to move 1.5f * max_length of each axis

And in `do_homing_move`:
```
  #if IS_SCARA
    // Tell the planner the axis is at 0
    current_position[axis] = 0;
    sync_plan_position();
    current_position[axis] = distance;
    line_to_current_position(real_fr_mm_s);
```
`line_to_current_position` will call into `planner.buffer_line` and finally call into MP_SCARA's `inverse_kinematics` to translate this target position into MP_SCARA's arm angles.

The problem of the above code is that this homing target position calculated from 1.5 * max_length of axis is not guaranteed to get a valid arm angles after being sent into MP_SCARA's `inverse_kinematics`, it could be beyond the reach of the arms and get you a wrong arm angle, depends on your settings of the SCARA. 

That's why in my case, the x axis can do homing normally and the y axis can not, because this homing position for x axis happens to be in the reach of the arms but the y axis is not so lucky.

Also this homing position for x axis just rotate my arm for under 90 degrees in my case, so when the arm is too far away from the endstop, it just stops before hits the endstop.

I think for SCARA, homing should be just keep rotating the arm until the endstops are hit instead of trying to use a far away position and do IK to that position for homing.

Not sure my ideas are right, correct me if I'm wrong.   :-)

ellensp, 2024-12-18:
I have this homing but the arm must be positioned so the upper arm hits its endstop first, ie the forarm should point away from the endstops. 

Ascii art of the  mp_scara     (shoulder joint) upper arm (elbow joint) forearm with a fixed hand.
So we are using same terminology...

```diff
diff --git a/Marlin/src/module/motion.cpp b/Marlin/src/module/motion.cpp
index 820089d7ee..4e9cec1003 100644
--- a/Marlin/src/module/motion.cpp
+++ b/Marlin/src/module/motion.cpp
@@ -2467,7 +2467,7 @@ void prepare_line_to_destination() {
       #endif
     }

-    #if ANY(MORGAN_SCARA, MP_SCARA)
+    #if ENABLED(MORGAN_SCARA)
       // Tell the planner the axis is at 0
       current_position[axis] = 0;
       sync_plan_position();
@@ -2665,7 +2665,7 @@ void prepare_line_to_destination() {

   void homeaxis(const AxisEnum axis) {

-    #if ANY(MORGAN_SCARA, MP_SCARA)
+    #if ENABLED(MORGAN_SCARA)
       // Only Z homing (with probe) is permitted
       if (axis != Z_AXIS) { BUZZ(100, 880); return; }
     #else
@@ -3054,7 +3054,7 @@ void set_axis_is_at_home(const AxisEnum axis) {
     }
   #endif

-  #if ANY(MORGAN_SCARA, AXEL_TPARA)
+  #if IS_SCARA
     scara_set_axis_is_at_home(axis);
   #elif ENABLED(DELTA)
     current_position[axis] = (axis == Z_AXIS) ? DIFF_TERN(HAS_BED_PROBE, delta_height, probe.offset.z) : base_home_pos(axis);
```

Config files needs this added  #define FEEDRATE_SCALING

The homing procedure is it does a quick home on X and Y ie home both at once (this is required to keep the forearm pointing in the same direction during homing home) 

Then the upper arm is rehomed, then the forarm is homed.

finally scara_set_axis_is_at_home is called to set home position to homeposition = { SCARA_OFFSET_THETA1, SCARA_OFFSET_THETA2} from Configuration.h

Sejkorka, 2024-12-18:
> Ascii art of the  mp_scara     (shoulder joint)<==upper arm==>(elbow joint)<== forearm with a fixed hand==>
> So we are using same terminology...
> 
> I have this homing but the arm must be positioned so the upper arm hits its endstop first, ie the forarm should point away from the endstops.   (I can see no way around this as the upper arm must be in correct location before the forearm can get near its endstop) 
> 
> ```diff
> diff --git a/Marlin/src/module/motion.cpp b/Marlin/src/module/motion.cpp
> index 820089d7ee..4e9cec1003 100644
> --- a/Marlin/src/module/motion.cpp
> +++ b/Marlin/src/module/motion.cpp
> @@ -2467,7 +2467,7 @@ void prepare_line_to_destination() {
>        #endif
>      }
> 
> -    #if ANY(MORGAN_SCARA, MP_SCARA)
> +    #if ENABLED(MORGAN_SCARA)
>        // Tell the planner the axis is at 0
>        current_position[axis] = 0;
>        sync_plan_position();
> @@ -2665,7 +2665,7 @@ void prepare_line_to_destination() {
> 
>    void homeaxis(const AxisEnum axis) {
> 
> -    #if ANY(MORGAN_SCARA, MP_SCARA)
> +    #if ENABLED(MORGAN_SCARA)
>        // Only Z homing (with probe) is permitted
>        if (axis != Z_AXIS) { BUZZ(100, 880); return; }
>      #else
> @@ -3054,7 +3054,7 @@ void set_axis_is_at_home(const AxisEnum axis) {
>      }
>    #endif
> 
> -  #if ANY(MORGAN_SCARA, AXEL_TPARA)
> +  #if IS_SCARA
>      scara_set_axis_is_at_home(axis);
>    #elif ENABLED(DELTA)
>      current_position[axis] = (axis == Z_AXIS) ? DIFF_TERN(HAS_BED_PROBE, delta_height, probe.offset.z) : base_home_pos(axis);
> ```
> 
> Config files needs this added  #define FEEDRATE_SCALING  (without this the homing feed rate are all over the place) 
> Also disable #define VALIDATE_HOMING_ENDSTOPS
> 
> 
> The XY homing procedure is it does a quick home on X and Y ie both move at once (this is required to keep the forearm pointing in the same direction during moving home) 
> 
> Then the upper arm is re-homed, then the forarm is homed.
> 
> finally scara_set_axis_is_at_home is called to set home position to homeposition = { SCARA_OFFSET_THETA1, SCARA_OFFSET_THETA2} from Configuration.h 
> 
>  The homes reliably for me, but I have not tested anything else, especially not the sacra kinematics.  

[image]

This is my scara

ellensp, 2024-12-18:
This is the MP_SCARA https://www.thingiverse.com/thing:2487048 which is also what I tested on

Sejkorka, 2024-12-18:
> This is the MP_SCARA https://www.thingiverse.com/thing:2487048 which is also what I tested on 
> 

Technically they are the same.

ellensp, 2024-12-18:
MP_SCARA has endstops in a weird location

ie.

[image]

Yours is much more like https://howtomechatronics.com/projects/laser-engraving-with-diy-arduino-scara-robot-complete-guide/

The video https://www.youtube.com/watch?v=1QHJksTrk8s shows the endstops are independent. This is not a MP_SCARA machine.

Sejkorka, 2024-12-18:
> MP_SCARA has endstops in a weird location
> 
> ie.
> 
> [image]
> 
> 
> Yours is much more like https://howtomechatronics.com/projects/laser-engraving-with-diy-arduino-scara-robot-complete-guide/
> 
> The video https://www.youtube.com/watch?v=1QHJksTrk8s shows the endstops are independent. Technically or otherwise this is not a MP_SCARA machine.  
> 

Yes, i took inspo from video, do you think i need to move endstops?

Sejkorka, 2024-12-18:
> MP_SCARA má koncové zastávky na podivném místě
> 
> tzn.
> 
> [image]
> 
> 
> Váš je mnohem více podobný https://howtomechatronics.com/projects/laser-engraving-with-diy-arduino-scara-robot-complete-guide/
> 
> Video https://www.youtube.com/watch?v=1QHJksTrk8s ukazuje, že koncové zarážky jsou nezávislé. Technicky ani jinak se nejedná o stroj MP_SCARA.  
> 

 

 If I understand correctly, I need to achieve that the first and second axis are homed at the same time?

ellensp, 2024-12-18:
The patch I provided should make it work for you also.
You just don't have the issue where by the forearm needs to be out of the way, before homing.
 
It will still quick home until either  the upper arm or forearm endstop is triggered, then it will re home upper arm then forearm

Sejkorka, 2024-12-18:
> The patch I provided should make it work for you too.
> You just don't have the issue where the forearm needs to be out of the way before homing.
>  
> It will still quick home until either the upper arm or forearm endstop is triggered, then it will re home upper arm then forearm  
> 

 Great, I'll try it as soon as I get to work.

Sejkorka, 2024-12-18:
> I commented out the return code for SCARA
>
> ```
>   void homeaxis(const AxisEnum axis) {
>
>     #if ANY(MORGAN_SCARA, MP_SCARA)
>       // Only Z homing (with probe) is permitted
>       // if (axis != [...]

Which version of MARLIN do you use?

Sejkorka, 2024-12-18:
After some testing, I found that I need to force the X motor, or X axis, or as Marlin calls it, the A axis, to rotate in the opposite direction. But only when homing. I tried switching the homing direction and also reversing the motor's operation. I think there must be a mistake somewhere or rather a problem in the homing process. I will try more experiments tomorrow. Do you think you could provide me with your config file?

ellensp, 2024-12-18:
development work is done on bugfix 2.1.x

Sejkorka, 2024-12-19:
> development work is done on bugfix 2.1.x

Can you send me please your configuratio or full marlin?

ellensp, 2024-12-19:
[Configuration.zip](https://github.com/user-attachments/files/18198665/Configuration.zip)

ellensp, 2024-12-19:
In Marlin is https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/Marlin/src/inc/Conditionals-5-post.h#L288-L295

```CPP 
#if IS_SCARA
  #if ENABLED(AXEL_TPARA)
    #define PRINTABLE_RADIUS (TPARA_LINKAGE_1 + TPARA_LINKAGE_2)
  #else
    #define QUICK_HOME   //  <<------------------------  disable this line 
    #define PRINTABLE_RADIUS (SCARA_LINKAGE_1 + SCARA_LINKAGE_2)
  #endif
#endif
```

that forces quick home, I would disable this line for your thing.

Sejkorka, 2024-12-19:
How  did you calcu

> In Marlin is https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/Marlin/src/inc/Conditionals-5-post.h#L288-L295
> 
> ```c++
> #if IS_SCARA
>   #if ENABLED(AXEL_TPARA)
>     #define PRINTABLE_RADIUS (TPARA_LINKAGE_1 + TPARA_LINKAGE_2)
>   #else
>     #define QUICK_HOME   //  <<------------------------  disable this line 
>     #define PRINTABLE_RADIUS (SCARA_LINKAGE_1 + SCARA_LINKAGE_2)
>   #endif
> #endif
> ```
> 
> that forces quick home, I would disable this line for your thing.

how did you get THETA1 and THETA2?

Sejkorka, 2024-12-19:
Homing is working, thank you so much, but next problem is here, when i type G0 X0 Y0 scara moves oposite direction

Sejkorka, 2024-12-19:
so i randomlly fixed problem with moving, but now its laggy, its moving very slowly

shimaode, 2024-12-20:
> How did you calcu
>
>
> how did you get THETA1 and THETA2?

Here is a great article explaining how to get THETA1 and THETA2 in step 12
https://www.instructables.com/Single-Arm-SCARA-Plotter-With-HOMING/

Sejkorka, 2024-12-20:
> Here is a great article explaining how to get THETA1 and THETA2 in step 12 https://www.instructables.com/Single-Arm-SCARA-Plotter-With-HOMING/

okay, i get it, now... why its moving "laggy"?

shimaode, 2024-12-21:
I'm using bugfix2.1x branch

Homing function is back to normal after applying ellensp's patch. But the arms' moving is not working properly.
The following code in `scara.cpp` is wrong I think:

  ```
void scara_set_axis_is_at_home(const AxisEnum axis) {
    if (axis == Z_AXIS)
      current_position.z = Z_HOME_POS;
    else {
      // MP_SCARA uses arm angles for AB home position
      #ifndef SCARA_OFFSET_THETA1
        #define SCARA_OFFSET_THETA1  12 // degrees
      #endif
      #ifndef SCARA_OFFSET_THETA2
        #define SCARA_OFFSET_THETA2 131 // degrees
      #endif
      ab_float_t homeposition = { SCARA_OFFSET_THETA1, SCARA_OFFSET_THETA2 };
      //DEBUG_ECHOLNPGM("homeposition A:", homeposition.a, " B:", homeposition.b);

      inverse_kinematics(homeposition);
      forward_kinematics(delta.a, delta.b);
      current_position[axis] = cartes[axis];

      update_software_endstops(axis);
    }
  }

  void inverse_kinematics(const xyz_pos_t &raw) {
    const float x = raw.x, y = raw.y, c = HYPOT(x, y),
                THETA3 = ATAN2(y, x),
                THETA1 = THETA3 + ACOS((sq(c) + sq(L1) - sq(L2)) / (2.0f * c * L1)),
                THETA2 = THETA3 - ACOS((sq(c) + sq(L2) - sq(L1)) / (2.0f * c * L2));

    delta.set(DEGREES(THETA1), DEGREES(THETA2), raw.z);
  }
```

1. for code in `scara_set_axis_is_at_home`, why passing { SCARA_OFFSET_THETA1, SCARA_OFFSET_THETA2 } to inverse_kinematics, this just not make sense to me, because inverse_kinematics accept cartesian position instead of angles.
  So I simply set current_position to X_HOME_POS, Y_HOME_POS in `scara_set_axis_is_at_home`, like the lts-2.0.7 branch did.
2. for inverse_kinematics, position should minus scara_offset first, ie. raw - scara_offset. The code should be like `const xy_pos_t spos = raw - scara_offset; const float x = spos.x, y = spos.y`

After the changes above, my arm moves normally.

Sejkorka, 2024-12-21:
> I'm using bugfix2.1x branch
>
> Homing function is back to normal after applying ellensp's patch. But the arms' moving is not working properly.
> The following code in `scara.cpp` is wrong I think: [...]

Can you please share your modified file?

shimaode, 2024-12-22:
```
diff --git "a/Marlin/src/module/scara.cpp" "b/Marlin/src/module/scara.cpp"
index 0f00ab5643..9cad34a504 100644
--- "a/Marlin/src/module/scara.cpp"
+++ "b/Marlin/src/module/scara.cpp"
@@ -152,8 +152,8 @@ float segments_per_second = DEFAULT_SEGMENTS_PER_SECOND;
       ab_float_t homeposition = { SCARA_OFFSET_THETA1, SCARA_OFFSET_THETA2 };
       //DEBUG_ECHOLNPGM("homeposition A:", homeposition.a, " B:", homeposition.b);
 
-      inverse_kinematics(homeposition);
-      forward_kinematics(delta.a, delta.b);
+      //inverse_kinematics(homeposition);
+      forward_kinematics(SCARA_OFFSET_THETA1, SCARA_OFFSET_THETA2);
       current_position[axis] = cartes[axis];
 
       //DEBUG_ECHOLNPGM_P(PSTR("Cartesian X"), current_position.x, SP_Y_LBL, current_position.y);
@@ -162,7 +162,8 @@ float segments_per_second = DEFAULT_SEGMENTS_PER_SECOND;
   }
 
   void inverse_kinematics(const xyz_pos_t &raw) {
-    const float x = raw.x, y = raw.y, c = HYPOT(x, y),
+    const xy_pos_t spos = raw - scara_offset;
+    const float x = spos.x, y = spos.y, c = HYPOT(x, y),
                 THETA3 = ATAN2(y, x),
                 THETA1 = THETA3 + ACOS((sq(c) + sq(L1) - sq(L2)) / (2.0f * c * L1)),
                 THETA2 = THETA3 - ACOS((sq(c) + sq(L2) - sq(L1)) / (2.0f * c * L2));
```
here is my patch.  My branch is bugfix-2.1.x

Sejkorka, 2024-12-22:
moving is working, but i thing inverse kinematics not working

shimaode, 2024-12-23:
Probably because the inverse kinematics is left handed but you want right handed?

shimaode, 2024-12-24:
Maybe you can try to add some console output in `inverse_kinematics` in `Marlin/src/module/scara.cpp` to check the results of inverse kinematics. I myself use right handed single arm scara, so I rewrote the whole inverse kinematics function. 

I think maybe you need to makesure your version of inverse kinematics matches your machine, I'm not sure which kind of inverse kinematics you are using, left handed or right handed?

Sejkorka, 2024-12-24:
because if i send command G0 X0 Y0, scara moves to bad positon, i will attach two images, one where it goes and second is skecth where is working area and homeposition.
[image]

[image]

dekutree64, 2024-12-26:
I think shimaode is right. Your arm appears to be right-handed, whereas the MPSCARA code is left-handed. It's easy to change. In scara.cpp function inverse_kinematics, switch the signs on the ACOS terms in THETA1 and THETA2.

Sejkorka, 2024-12-29:
> I think shimaode is right. Your arm appears to be right-handed, whereas the MPSCARA code is left-handed. It's easy to change. In scara.cpp function inverse_kinematics, switch the signs on the ACOS terms in THETA1 and THETA2.

All of the signs?

dekutree64, 2024-12-29:
No, just the ACOS.

Left handed:
```
                THETA1 = THETA3 + ACOS((sq(c) + sq(L1) - sq(L2)) / (2.0f * c * L1)),
                THETA2 = THETA3 - ACOS((sq(c) + sq(L2) - sq(L1)) / (2.0f * c * L2));

```
Right handed:
```
                THETA1 = THETA3 - ACOS((sq(c) + sq(L1) - sq(L2)) / (2.0f * c * L1)),
                THETA2 = THETA3 + ACOS((sq(c) + sq(L2) - sq(L1)) / (2.0f * c * L2));
```

Sejkorka, 2024-12-29:
> No, just the ACOS.
>
> Left handed:
> ```
>                 THETA1 = THETA3 + ACOS((sq(c) + sq(L1) - sq(L2)) / (2.0f * c * L1)),
>                 THETA2 = THETA3 - ACOS((sq(c) + sq(L2) - sq(L1)) / (2.0f * c [...]

Okay, I tried it, but no success, still doing the same thing...

dekutree64, 2024-12-30:
It probably just has mismatched angles and cartesian coordinates in its mind.

In the angular coordinate system, 0 degrees points along positive X axis and 90 degrees points along positive Y axis (positive angle rotates counterclockwise when viewed from above). delta.a is the angle of the first arm segment, delta.b is the second arm segment. Both are relative to the global X axis, so moving delta.a alone should cause the physical elbow angle to change, to keep the second arm segment at the same angle relative to global X.

I recommend enabling BED_CENTER_AT_0_0. Then SCARA_OFFSET_X is 0, and SCARA_OFFSET_Y is the distance from bed center to shoulder axis (should be negative).

Then for testing, try manually positioning the arm so it points down the center of the bed, along the global Y axis. Set delta.a=delta.b=90, current_position[X_AXIS]=cartes[X_AXIS]=0, current_position[Y_AXIS]=cartes[Y_AXIS]=SCARA_OFFSET_Y+L1+L2;
Then try sending some G0 commands and see if it moves right. If that works, then you can work on automatic homing. It's basically the same thing. You just need to get a consistent set of delta,current_position,cartes corresponding to the physical position of the arm after it hits the limit switches.

Sejkorka, 2024-12-30:
> It probably just has mismatched angles and Cartesian coordinates in its mind.
> 
> In the angular coordinate system, 0 degrees points along the positive X axis and 90 degrees points along the positive Y axis (positive angle rotates counterclockwise when viewed from above). delta.a is the angle of the first arm segment, delta.b is the second arm segment. Both are relative to the global X axis, so moving delta.a alone should cause the physical elbow angle to change, to keep the second arm segment at the same angle relative to global X.
> 
> I recommend enabling BED_CENTER_AT_0_0. Then SCARA_OFFSET_X is 0, and SCARA_OFFSET_Y is the distance from bed center to shoulder axis (should be negative).
> 
> Then for testing, try manually positioning the arm so it points down the center of the bed, along the global Y axis. Set delta.a=delta.b=90, current_position[X_AXIS]=cartes[X_AXIS]=0, current_position[Y_AXIS]=cartes[Y_AXIS]=SCARA_OFFSET_Y+L1+L2;
> Then try sending some G0 commands and see if it moves right. If that works, then you can work on automatic homing. It's basically the same thing. You just need to get a consistent set of delta, current_position, maps corresponding to the physical position of the arm after it hits the limit switches.

 I hope I understood correctly, but homing is working fine now, but movement after homing not so much, so could this help with movement issues?

dekutree64, 2024-12-30:
Yes, it's just a reliable way to get the angular and cartesian coordinates reasonably well matched with the physical position of the arm.

Sejkorka, 2024-12-30:
> Yes, it's just a reliable way to get the angular and cartesian coordinates reasonably well matched with the physical position of the arm.

Okay, so... you think i have right handed scara? And i need to set right XY coordinates?

Sejkorka, 2025-01-02:
> SCARA_OFFSET_Y

Okay, it moves to bed zero, but i need to set maximum rotation, or angle of arms, because secondarm is hitting max position ad try to move to the construction
i will attach photo.
[image]

Sejkorka, 2025-01-11:
> Yes, it's just a reliable way to get the angular and cartesian coordinates reasonably well matched with the physical position of the arm.

Do you have any idea?

dekutree64, 2025-01-11:
I've done all I can remotely. You'll have to put some debug prints in the code to monitor the angular and cartesian coordinates to figure out what it's thinking.

One more thing just to be clear, the cartesian coordinate space is from the robot's perspective. Imagine you are the robot and it is your right arm. +Y is forward, +X is to your right.

Sejkorka, 2025-01-16:
> I've done all I can remotely. You'll have to put some debug prints in the code to monitor the angular and cartesian coordinates to figure out what it's thinking.
> 
> One more thing just to be clear, the cartesian coordinate space is from the robot's perspective. Imagine you are the robot and it is your right arm. +Y is forward, +X is to your right.
> 
> Here's a diagram from my wall-mounted SCARA printer that might help. [image]

and is there any way to help via discord via camera? or something like that? to see exactly what he's doing?

Sejkorka, 2025-01-18:
I'm starting to think that the firmware is ignoring the set zero distance of the x-axis from the zero point of the workspace. It's just a guess, but no matter what I try it still hits the structure with the other arm, I tried to recalculate the steps and it didn't help, after setting a smaller number of steps it doesn't hit anymore, but that's because it doesn't hit the given distance and at that moment the distance traveled doesn't correspond to reality.

Sejkorka, 2025-01-23:
So i get it to moving and engraving, but there is little problem, lines are not straight, and i set home offset to right cornet but scarais moving to left corner

[image]
[image]

Sejkorka, 2025-01-23:
> I've done all I can remotely. You'll have to put some debug prints in the code to monitor the angular and cartesian coordinates to figure out what it's thinking.
>
> One more thing just to be clear, [...]

do you have idea, which can cause this problem?

dekutree64, 2025-01-24:
Oh, I think I figured it out. You most likely need to subtract the shoulder angle from the elbow angle in inverse_kinematics,
`delta.set(DEGREES(THETA1), DEGREES(THETA2 - THETA1), raw.z);`
What I said before about delta.x and y being relative to global X axis was not quite correct. It's only THETA1,2,3 that are always relative to global X. The original MPSCARA uses a belt system that results in the second arm segment maintaining its orientation relative to the global X axis when the shoulder axis is moved, so in that case delta.y is also relative to global X. But in your case the elbow motor rides along with the shoulder rotation, so delta.y is relative to the first arm segment.

I think forward_kinematics will need to be modified too.
```
   const float a_sin = sin(RADIANS(a)) * L1,
                a_cos = cos(RADIANS(a)) * L1,
                b_sin = sin(RADIANS(b+a)) * L2,
                b_cos = cos(RADIANS(b+a)) * L2;
```
This is why we don't have easy-to-use SCARA code :) There are too many variations in arm construction, so even if we did have code for all of them, it would be confusing to figure out exactly which one you need. Mine is the worst of all, because it has both motors stationary like MPSCARA, but different reduction ratios on the shoulder and elbow so it doesn't perfectly cancel out the shoulder rotation like MPSCARA. I solved it by adding a new #define SCARA_CROSSTALK_FACTOR and doing `delta.set(DEGREES(THETA1), DEGREES(THETA2 - THETA1 * SCARA_CROSSTALK_FACTOR), raw.z);` in inverse_kinematics, and `RADIANS(b + a * SCARA_CROSSTALK_FACTOR)` in forward_kinematics. Perhaps we should integrate it into the main branch. For mine it's 0.33333, but it would work for yours too with crosstalk factor 1, and 0 for MPSCARA.

Sejkorka, 2025-01-24:
> const float a_sin = sin(RADIANS(a)) * L1,
>                 a_cos = cos(RADIANS(a)) * L1,
>                 b_sin = sin(RADIANS(b+a)) * L2,
>                 b_cos = cos(RADIANS(b+a)) * L2;

Add to configuration.h or in which file?

Sejkorka, 2025-01-24:
[image]

[image]

[image]

Is this ok?

dekutree64, 2025-01-24:
Looks good. Give it a try and see if it works.

Sejkorka, 2025-01-24:
I will try it, when i get to scara

Sejkorka, 2025-01-24:
> Looks good. Give it a try and see if it works.

it doesn't work, it tries to go against the endstop and the square is the same shape just smaller in size, I'll attach a picture.

[image]

ellensp, 2025-01-24:
non straight lines are often  the result of incorrect machine arm measurements
arm measurements are between center points of rotation

Sejkorka, 2025-01-24:
> non straight lines are often the result of incorrect machine arm measurements
> arm measurements are between center points of rotation
> 
>  
> 
> 
> 
>  

 if L1 is the arm that is closer to the base then everything is correct, especially what happened now is that instead of going to the right corner, the scara goes somewhere in the middle and when engraving it goes against the limit switches

dekutree64, 2025-01-24:
Let's try this... mark bed zero with a piece of tape or something, home the printer, and take a photo from above. Print out delta.x, delta.y, current_position[X_AXIS] and current_position[Y_AXIS].

Then command it to move to some other position (anywhere that doesn't crash into itself) and take another photo from above and print out those variables.

Also post the whole SCARA section from Configuration.h.

That may be enough information for me to figure out what it's thinking.

Sejkorka, 2025-01-24:
> Let's try this... mark bed zero with a piece of tape or something, home the printer, and take a photo from above. Print out delta.x, delta.y, current_position[X_AXIS] and current_position[Y_AXIS]. [...]

 Maybe a stupid question but how do I print delta.x and delta ?

dekutree64, 2025-01-24:
Hmm, I can't remember how I did it before. Fishing through the code, I think M114 will work. It looks like it prints "Raw:" followed by the current_position values and "ScaraK:" followed by the delta values.

Sejkorka, 2025-01-25:
[image]
This is my setup, if i set it to marlin and upload it to scara, after this i sent G00 X0 Y0, the scara wants to move towards the limit switch

[68 more replies]