MarlinFirmware/Marlin #27577: [BUG] G29 S0 causes printer to hang with "echo:busy: processing" forever

mikef0, 2024-12-09:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I am trying to run the following sequence of commands. When it gets to G29 S0, it just says "echo:busy: Processing" indefinitely, forever. I confirmed this issue on Bugfix-2.1 from August 2023, and just updated to Bugfix-2.1 from December 8 2024 with no change. I have also tried saving to slot 1 with the save behavior.

I initially discovered the issue by using the Bed Visualizer plugin in Octoprint, but even if I manually run G29 followed by G29 S0 or G29 S1 from the printer LCD or by sending terminal commands, the behavior is the same

I am running a 5x5 mesh.

This is the sequence of commands I am ultimately trying to run.

G29 P0 T	   ; run bilinear probing
G29 P3 ;Smart Fill Repeat until all mesh points are filled in, Used to fill unreachable points.
G29 S0
M155 S3  ; reset temperature reporting
M140 S0 ; cooling down the bed
M500 ; store mesh in EEPROM
M420 S1

### Bug Timeline

Initially observed with bugfix 2.1 from August 2023 and still occurs on bugfix 2.1 from December 8, 2024

### Expected behavior

Successfully save mesh to slot in EEPROM

### Actual behavior

doesn't save mesh to slot, just gets stuck on "echo:busy: processing" indefinitely

### Steps to Reproduce

1. Home printer with G28,
2. Run G29 and wait for mesh to complete
3. When mesh completes, run G29 S0

### Version of Marlin Firmware

bugfix-2.1, December 8 2024

### Printer model

home built

### Electronics

SKR2 STM32F407VG

### LCD/Controller

BTT TFT3.5

### Other add-ons

Official Bltouch 3.1

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

Cura

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration.zip](https://github.com/user-attachments/files/18064727/Configuration.zip)

ellensp, 2024-12-09:
there are 5 different types of bed leveling, they all support different gcode parameters

You have
/#define AUTO_BED_LEVELING_3POINT
//#define AUTO_BED_LEVELING_LINEAR
#define AUTO_BED_LEVELING_BILINEAR
//#define AUTO_BED_LEVELING_UBL
//#define MESH_BED_LEVELING

For AUTO_BED_LEVELING_BILINEAR here are the supported commands https://marlinfw.org/docs/gcode/G029-abl-bilinear.html

so "G29 P0 T ; run bilinear probing" is not valid
and "G29 P3 ;Smart Fill Repeat until all mesh points are filled in, Used to fill unreachable points." is not valid
and "G29 S0" is valid but it is not save

you seem to be trying to run AUTO_BED_LEVELING_UBL gcodes https://marlinfw.org/docs/gcode/G029-ubl.html

thinkyhead, 2024-12-10:
Patch applied, and will likely get back-ported to the LTS branches RSN. Thanks for the report!