MarlinFirmware/Marlin #27578: tinybee and #define SERIAL_PORT_2 -1 provoque build error

moriamoria, 2024-12-09:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

hello,

i am sorry, but i stil try to configure my new tinybee with bugfix.2.1

but when i just set
 #define MOTHERBOARD BOARD_MKS_TINYBEE
 #define SERIAL_PORT_2 -1

i have many build error:

Marlin/src/HAL/shared/../../core/../inc/../core/macros.h:641:24: note: in expansion of macro 'EVAL2'
 #define EVAL4(V...)    EVAL2(EVAL2(V))

Marlin/src/HAL/shared/../../core/serial.h:94:27: note: in expansion of macro '_SERIAL_LEAF_2'
     #define SERIAL_LEAF_2 _SERIAL_LEAF_2
                           ^
Marlin/src/HAL/ESP32/../../inc/../core/serial.h:132:55: error: request for member 'write' in 'multiSerial', which is of non-class type 'SerialOutputT {aka int}' 
....

many thanks if you can have a look

regards
[Marlin.zip](https://github.com/user-attachments/files/18069174/Marlin.zip)

### Bug Timeline

all bugfix 2.1.X commit suffer from this issue

### Expected behavior

no buil error

### Steps to Reproduce

just buid

### Version of Marlin Firmware

origin/bugfix-2.1.x 

### Printer model

ortur 4 v1

### Electronics

tinybee

### LCD/Controller

REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2024-12-09:
Why are you setting up two serial ports?

moriamoria, 2024-12-10:
Hello,

that was the case when I have my skr 1.3 motherboard.

I have define SERIAL_PORT_2 to activate USB.

#define SERIAL_PORT_2 -1

Also in the firmware delivered by makerbase, i have the same settings

https://github.com/makerbase-mks/MKS-TinyBee/blob/main/firmware/mks%20tinybee%20marlin/Marlin/Configuration.h

#define SERIAL_PORT_2 -1
#define BAUDRATE_2 115200   // Enable to override BAUDRATE

regards

ellensp, 2024-12-10:
no,  you sequentially enable the number if serial ports you need 
SERIAL_PORT, SERIAL_PORT_2 then SERIAL_PORT_3
you assign what ever uart you want to use.  to any enabled serial_port   (there are some exceptions to this) 

Your board has two uarts: 
0 is the USB cable  
2 is on the exp port for serial displays

-1 is not valid, It is for chips with inbuilt usb in hardware, this board uses a ch340 

All you need is 
#define SERIAL_PORT 0
and comment out #define SERIAL_PORT_2

moriamoria, 2024-12-10:
Hello,

I am really lost.

Following settings gives bad result : 

#define SERIAL_PORT 0
#define BAUDRATE 115200

#define SERIAL_PORT_2 1
#define BAUDRATE_2 115200   // :[2400, 9600, 19200, 38400, 57600, 115200, 250000, 500000, 1000000] Enable to override 

I got following error : 

#define EVAL16(V...)   EVAL8(EVAL8(V))
                              ^
Marlin/src/HAL/shared/../../inc/../core/macros.h:630:24: note: in expansion of macro 'EVAL16'
 #define EVAL(V...)     EVAL16(V)

Marlin/src/HAL/ESP32/../../inc/../core/serial.h: In function 'void SERIAL_ECHO(T)':
Marlin/src/HAL/ESP32/../../inc/../core/serial.h:141:61: error: request for member 'print' in 'multiSerial', which is of 
non-class type 'SerialOutputT {aka int}'

Also I don't understand why makerbase Marlin firmware is not following your recomandation, but looks like it is version 2.0.9

Can you try on your side just to see if you can build ?

Regards

moriamoria, 2024-12-12:
Hello,

I have check other esp32 motherboard and firmware provided by constructor.

as example, please find configuration of FYSETC motherboard  : 
[[FYSETC-E4](https://github.com/FYSETC/FYSETC-E4/tree/main)
](https://github.com/FYSETC/FYSETC-E4/blob/main/firmware/Marlin/Marlin/Configuration.h)

in the configuration we can find many serial declared:

#define SERIAL_PORT 0
#define SERIAL_PORT_2 -1

I wait for your answer, but I have dougth about my tinybee motherboard and If I need to change it.

That is petty as many marlin user could be interested to by a esp32 based motherboard.

Let's wait if I will be able to use this tinybee with Marlin 2.1 bugfix.

**My goal is to use my tinybee with both REPRAP_DISCOUNT_SMART_CONTROLLER and MKS TFT24**

So any will be welcome

Regards

ellensp, 2024-12-12:
all boards are different.

FYSETC-E4 != TINYBEE

Also the linked file clearly has #define SERIAL_PORT_2 -1 commented out!

-1 is a special USB port, where usb d- and d+ pins are directly connected to the main processor chip
This is simply not possible on these esp32 chips.

What part of leave #define SERIAL_PORT_2 -1 disabled is so f**ken hard to comprehend!!!

DISABLE #define SERIAL_PORT_2 -1
DISABLE #define SERIAL_PORT_2 -1
DISABLE #define SERIAL_PORT_2 -1
DISABLE #define SERIAL_PORT_2 -1
DISABLE #define SERIAL_PORT_2 -1
DISABLE #define SERIAL_PORT_2 -1
DISABLE #define SERIAL_PORT_2 -1
DISABLE #define SERIAL_PORT_2 -1
DISABLE #define SERIAL_PORT_2 -1
DISABLE #define SERIAL_PORT_2 -1
DISABLE #define SERIAL_PORT_2 -1
DISABLE #define SERIAL_PORT_2 -1
DISABLE #define SERIAL_PORT_2 -1
DISABLE #define SERIAL_PORT_2 -1
DISABLE #define SERIAL_PORT_2 -1
DISABLE #define SERIAL_PORT_2 -1
DISABLE #define SERIAL_PORT_2 -1
DISABLE #define SERIAL_PORT_2 -1
DISABLE #define SERIAL_PORT_2 -1
DISABLE #define SERIAL_PORT_2 -1
DISABLE #define SERIAL_PORT_2 -1
DISABLE #define SERIAL_PORT_2 -1
DISABLE #define SERIAL_PORT_2 -1
DISABLE #define SERIAL_PORT_2 -1
DISABLE #define SERIAL_PORT_2 -1
DISABLE #define SERIAL_PORT_2 -1
DISABLE #define SERIAL_PORT_2 -1

closed 2024-12-12 by ellensp