MarlinFirmware/Marlin #27585: [BUG] Unresponsive & Errorrenous print processor

gokhanozcetin, 2024-12-12:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I am running a custom Marlin build on a custom LPC1768 board. Following code causes the print job to become unresponsive and commands including Emergency Parser inputs. It also fires M11() (air_evac_disable) at a random point in G-Code

Bugging Gcode attached as Test1.txt

Other codes work fine, attached as Test3.txt

G21
G90
M10
G4 P4
M3 S50
G0 F10000
G1 F5000
M7
G0 X5.000 Y15.000
G1 X395.000 Y15.000
G0 X395.000 Y25.000
G1 X5.000 Y25.000
G0 X5.000 Y35.000
G1 X395.000 Y35.000
G0 X395.000 Y45.000
G1 X5.000 Y45.000
G0 X5.000 Y55.000
G1 X395.000 Y55.000
G0 X395.000 Y65.000
G1 X5.000 Y65.000
G0 X5.000 Y75.000
G1 X395.000 Y75.000
G0 X395.000 Y85.000
G1 X5.000 Y85.000
G0 X5.000 Y95.000
G1 X395.000 Y95.000
G0 X395.000 Y105.000
G1 X5.000 Y105.000
G0 X5.000 Y115.000
G1 X395.000 Y115.000
G0 X395.000 Y125.000
G1 X5.000 Y125.000
G0 X5.000 Y135.000
G1 X395.000 Y135.000
G0 X395.000 Y145.000
G1 X5.000 Y145.000
G0 X5.000 Y155.000
G1 X395.000 Y155.000
G0 X395.000 Y165.000
G1 X5.000 Y165.000
G0 X5.000 Y175.000
G1 X395.000 Y175.000
G0 X395.000 Y185.000
G1 X5.000 Y185.000
G0 X5.000 Y195.000
G1 X395.000 Y195.000
G0 X395.000 Y205.000
G1 X5.000 Y205.000
G0 X5.000 Y215.000
G1 X395.000 Y215.000
G0 X395.000 Y225.000
G1 X5.000 Y225.000
G0 X5.000 Y235.000
G1 X395.000 Y235.000
G0 X395.000 Y245.000
G1 X5.000 Y245.000
G0 X5.000 Y255.000
G1 X395.000 Y255.000
G0 X395.000 Y265.000
G1 X5.000 Y265.000
G0 X5.000 Y275.000
G1 X395.000 Y275.000
G0 X395.000 Y285.000
G1 X5.000 Y285.000
G0 X5.000 Y295.000
G1 X395.000 Y295.000
G0 X395.000 Y305.000
G1 X5.000 Y305.000
G0 X5.000 Y315.000
G1 X395.000 Y315.000
G0 X395.000 Y325.000
G1 X5.000 Y325.000
G0 X5.000 Y335.000
G1 X395.000 Y335.000
G0 X395.000 Y345.000
G1 X5.000 Y345.000
G0 X5.000 Y355.000
G1 X395.000 Y355.000
G0 X395.000 Y365.000
G1 X5.000 Y365.000
G0 X5.000 Y375.000
G1 X395.000 Y375.000
G0 X395.000 Y385.000
G1 X5.000 Y385.000
M11
M5
M9
G0 X0 Y0
M2

### Expected behavior

- Freeze pin works
- Spindle output works
- Coolant control works
- Steppers operate perfectly
- Heating works
- P000 and R000 commands works

### Actual behavior

- M524 does not work
- M112 does not work
- M commands through neither Nextion nor Serial works
- All commands sent via Nextion or serial are processed after print is completed
- A spurious M11 is fired in a random point in G-Code

### Steps to Reproduce

- Cura and SD printing results same
- Disabled motor drivers to eliminate EMI - same behaviour
- Disabled endstops in printing - same behaviour
- Tried reducing speed, acceleration, replacing G0 with G1

### Version of Marlin Firmware

Last Version

### Printer model

Custom

### Electronics

Custom

### LCD/Controller

Nextion

### Bed Leveling

None

### Your Slicer

None

### Host Software

Cura

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[CustomBuild.zip](https://github.com/user-attachments/files/18111380/CustomBuild.zip)
[Test1.txt](https://github.com/user-attachments/files/18111421/Test1.txt)
[Test3.txt](https://github.com/user-attachments/files/18111459/Test3.txt)

gokhanozcetin, 2024-12-12:
> So if I overwrite the latest bugfix files with your files in CutomBuild.zip, then I have you source base? I have an unused LPC1768 lying around (MKS SGEN L V1), so I could give it a quick try...

You may not be able to run it because I changed almost every pin :)

thinkyhead, 2024-12-12:
You need to put `M400` to wait for movement to complete before your `M11`.

gokhanozcetin, 2024-12-12:
> You need to put `M400` to wait for movement to complete before your `M11`.

Ok, Will try however the Test3 is another code which does not show same behaviour. Did you check that?

And also if it shoots out M11 by itself, it does not respond M524, M112 etc. although the emergency parser is enabled. With the Test3, I can throw M524 any time and it stops job before next move

gokhanozcetin, 2024-12-13:
> You need to put `M400` to wait for movement to complete before your `M11`.

Have tried M400 just before M11, now the device stops 7 or 8 lines before M400 and hangs. Blower output does not turn off since moves have not finished but it does not move to finish the moves

rondlh, 2024-12-13:
> > So if I overwrite the latest bugfix files with your files in CutomBuild.zip, then I have you source base? I have an unused LPC1768 lying around (MKS SGEN L V1), so I could give it a quick try...
> 
> You may not be able to run it because I changed almost every pin :)

Yes, I noticed, I'm talking about a test bench setup (nothing attached), I just want to see if the code can run without hangs and doesn't switch off the M11 fan, which I can see from the board LED status. Let me see when I have time for it...

gokhanozcetin, 2024-12-13:
> Yes, I noticed, I'm talking about a test bench setup (nothing attached), I just want to see if the code can run without hangs and doesn't switch off the M11 fan, which I can see from the board LED [...]

Yes, you can use the custom files and run a test. I have modified some other files to customize freeze function. Keep freeze pin grounded to run codes

closed 2025-03-29 by github-actions[bot]