MarlinFirmware/Marlin #27588: [FR] ADXL345 3-axis accelerometer support for input shaping measurements

rondlh, 2024-12-13:
### Is your feature request related to a problem? Please describe.

Not really, I just want to see if there is interest in adding accelerometer support to Marlin.
I have the sensor working already. The sensor is located on a tool head board (BTT EBB42 V1.2).
See #27547

Any input on this topic would be welcome... like what mode to put the sensor in, sample rates, resolution, range etc...

### Are you looking for hardware support?

No

### Describe the feature you want

Support for the ADXL345 3-axis accelerometer for input shaping measurements

### Additional context

See #27547

rondlh, 2024-12-13:
> > I just want to see if there is interest in adding accelerometer support to Marlin.
> 
> There's definitely interest and anything we can do to make calibration easier for the user and have parity with other firmware is a good thing.

Thanks, that's good to know. I just added it to #27547

Currently it uses the SD card SPI pins (SD_SS_PIN, SD_SCK_PIN, SD_MISO_PIN, SD_MOSI_PIN), so there is a conflict if you want to use the onboard SD card.

The ADXL345 device uses SPI communication "mode 3" (SPI_MODE3), while I believe the SD Card uses "mode 0". It's possible to toggle the modes while the devices are disabled (not implemented). For now I recommend to disable the SD card on the host.
The sensor I work with is located on the BTT EBB42 V1.2 head board, which doesn't have a SD card in the first place.
The MKS THR and Mellow Fly SHT tool head boards use the same acceleration sensor, but they are RP2040 based.

If somebody can tell me what requirements they have for the sensor (mode, sample rate, range etc.) then I will happily implement that.

convicte, 2025-05-06:
This may be a novice question, though since there are commits towards this capability being implemented and there is certainly general excitement for this being included in Marlin, is there perhaps a timeline for when some progress would be envisioned?