MarlinFirmware/Marlin #27589: make the date/time stamp of the 'compile' available to display.

dbyy, 2024-12-15:
### Is your feature request related to a problem? Please describe.

no, not related to a problem.

### Are you looking for hardware support?

n/a

### Describe the feature you want

It would be nice to have the date/time stamp of the last compile (from the firmware.bin file) available to display on the printers screen when running the firmware.
When testing some features in Marlin,  I may end up with several compilations in a short period of time.
To have the date/time stamp available on the display helps to keep the different versions apart.
It would be a 'nice to have'.

### Additional context

I'm not thinking of a new '#define". More of an internal variable that can be displayed on the printers screen

robherc, 2024-12-15:
Is there a reason why the compile datetimestamp from Serial Report isn't sufficient here?

ellensp, 2024-12-16:
this is almost trivial

When BUILDDATE is defined add a submenu "Build Datetime" to the About Printer menu
This menu shows the build date and time

Eg added submenu
[image]

The build date and time
[image]

```DIFF
diff --git a/Marlin/Configuration_adv.h b/Marlin/Configuration_adv.h
index 564f44df60..f992dd7884 100644
--- a/Marlin/Configuration_adv.h
+++ b/Marlin/Configuration_adv.h
@@ -1546,8 +1546,9 @@
   #endif
 
   // Include a page of printer information in the LCD Main Menu
-  //#define LCD_INFO_MENU
+  #define LCD_INFO_MENU
   #if ENABLED(LCD_INFO_MENU)
+    #define BUILDDATE
     //#define LCD_PRINTER_INFO_IS_BOOTSCREEN // Show bootscreen(s) instead of Printer Info pages
   #endif
 
diff --git a/Marlin/src/lcd/menu/menu_info.cpp b/Marlin/src/lcd/menu/menu_info.cpp
index 2db0caa805..81ca11beca 100644
--- a/Marlin/src/lcd/menu/menu_info.cpp
+++ b/Marlin/src/lcd/menu/menu_info.cpp
@@ -263,6 +263,18 @@ void menu_info_board() {
 
 #endif
 
+//
+// "Build Date" submenu
+//
+#if ENABLED(BUILDDATE)
+  void menu_info_build() {
+    if (ui.use_click()) return ui.go_back();
+    START_SCREEN();
+    STATIC_ITEM_F(F(__DATE__ " " __TIME__));                         // YYYY-MM-DD HH:MM
+    END_SCREEN();
+  }
+#endif
+
 //
 // "About Printer" submenu
 //
@@ -306,6 +318,11 @@ void menu_info() {
   }
   #endif
 
+  #if ENABLED(BUILDDATE)
+    SUBMENU(MSG_INFO_BUILD, menu_info_build);               // Printer Stats >
+  #endif
+
+
   END_MENU();
 }
 
```

Everything needs a define, so it can be removed entirely from the code.

dbyy, 2024-12-16:
> this is almost trivial
> 
> When BUILDDATE is defined add a submenu "Build Datetime" to the About Printer menu This menu shows the build date and time I didn't want to add it to the Printer info menu, as there is already a release date and time in there and it was just confusing.
> 
> Eg added submenu ![Screenshot from 2024-12-16 12-58-26](https://private-user-images.githubusercontent.com/530024/395920386-9b1e973b-984d-49aa-87f0-b784ac8bb52f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQzMDk0NDEsIm5iZiI6MTczNDMwOTE0MSwicGF0aCI6Ii81MzAwMjQvMzk1OTIwMzg2LTliMWU5NzNiLTk4NGQtNDlhYS04N2YwLWI3ODRhYzhiYjUyZi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjE2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIxNlQwMDMyMjFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT01ZWRkNGY1ODljYjljN2E1NGI2NTQ5ZWZkYjZmMWJhZmIyM2Y1MTAzMjBjNGVhNDM3OTFkMWEyODdjZTU1M2UyJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.i1IZBOsqAAito6JUwrjOleiufjWgGN9uY-NK3SKWdn0)
> 
> The build date and time ![Screenshot from 2024-12-16 12-57-11](https://private-user-images.githubusercontent.com/530024/395920180-fe5b8104-92ea-4261-abd4-acc82b9df454.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQzMDk0NDEsIm5iZiI6MTczNDMwOTE0MSwicGF0aCI6Ii81MzAwMjQvMzk1OTIwMTgwLWZlNWI4MTA0LTkyZWEtNDI2MS1hYmQ0LWFjYzgyYjlkZjQ1NC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjE2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIxNlQwMDMyMjFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1lYjFiZTViNGRjZWRiODk4ZDg0YmUxODkxMTZiMzA4OTk2NDU4ODZlOWRmN2VhNjIwOGE5MWExMzA1MTRhMzUxJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.frsTvpAu29YC29lXkOVVhXVV5pwR26yEFNWb7y9Y62U)
> 
> ```diff
> diff --git a/Marlin/Configuration_adv.h b/Marlin/Configuration_adv.h
> index 564f44df60..f992dd7884 100644
> --- a/Marlin/Configuration_adv.h
> +++ b/Marlin/Configuration_adv.h
> @@ -1546,8 +1546,9 @@
>    #endif
>  
>    // Include a page of printer information in the LCD Main Menu
> -  //#define LCD_INFO_MENU
> +  #define LCD_INFO_MENU
>    #if ENABLED(LCD_INFO_MENU)
> +    #define BUILDDATE
>      //#define LCD_PRINTER_INFO_IS_BOOTSCREEN // Show bootscreen(s) instead of Printer Info pages
>    #endif
>  
> diff --git a/Marlin/src/lcd/menu/menu_info.cpp b/Marlin/src/lcd/menu/menu_info.cpp
> index 2db0caa805..81ca11beca 100644
> --- a/Marlin/src/lcd/menu/menu_info.cpp
> +++ b/Marlin/src/lcd/menu/menu_info.cpp
> @@ -263,6 +263,18 @@ void menu_info_board() {
>  
>  #endif
>  
> +//
> +// "Build Date" submenu
> +//
> +#if ENABLED(BUILDDATE)
> +  void menu_info_build() {
> +    if (ui.use_click()) return ui.go_back();
> +    START_SCREEN();
> +    STATIC_ITEM_F(F(__DATE__ " " __TIME__));                         // YYYY-MM-DD HH:MM
> +    END_SCREEN();
> +  }
> +#endif
> +
>  //
>  // "About Printer" submenu
>  //
> @@ -306,6 +318,11 @@ void menu_info() {
>    }
>    #endif
>  
> +  #if ENABLED(BUILDDATE)
> +    SUBMENU(MSG_INFO_BUILD, menu_info_build);               // Printer Stats >
> +  #endif
> +
> +
>    END_MENU();
>  }
>  
> ```
> 
> Everything needs a define, so it can be removed entirely from the code.

This is awesome.
Just wondering, on which version of configuration/configuration_adv.h is the 'diff' based upon?
I'm using 'configs' of Marlin 2.1.2.4 with BTT-SKR-Mini-E3/BTT TF35 and can't locate the spots so easily in my files.
Thanks for the suggestions.

ellensp, 2024-12-16:
based on bugfix 2.1.x

ellensp, 2024-12-16:
this is for marlin menus, to update the touch interface would require you to update the btt tft firmware

ellensp, 2024-12-16:
I have created a PC to add this

closed 2024-12-16 by ellensp

dbyy, 2024-12-16:
> based on bugfix 2.1.x

I figured out where it has to go.
The only problem is (and I'm not a c++ guru) I'm receiving an error when compiling  `'Marlin/src/lcd/menu/menu_info.cpp: In function 'void menu_info_board()':
Marlin/src/lcd/menu/menu_info.cpp:232:5: error: a function-definition is not allowed here before '{' token  `
and later on as consequence of that
`Marlin/src/lcd/menu/menu_info.cpp:333:29: error: 'menu_info_build' was not declared in this scope; did you mean 'menu_info_board'?
  333 |     SUBMENU(MSG_INFO_BUILD, menu_info_build);               // Printer Stats >`
So at the moment I'm stuck with that. Too bad.
Any help is much appreciated even though its not really a problem of  the Marlin code.
TIA

dbyy, 2024-12-16:
Update:
I figured it out.
I positioned the changes at the wrong spot in "menu_info.cpp".
After I consulted the original file from bug-fix 2.1.x, I changed the location (to the right spot), and now it's compiling w/o any error.
Have to flash the firmware and see how it looks on the display.
Thanks again for all feedback.