MarlinFirmware/Marlin #27598: RENAME PIN/DESCRIPTIONS/DESIGNATION

sinus19, 2024-12-21:
### Is your feature request related to a problem? Please describe.

No

### Are you looking for hardware support?

no

### Describe the feature you want

I've been using Marlin for over 5 years and one thing that i have always struggles with is understanding the different name designations of the pins.  It would be really nice to name each pin by its original Pin Number per the Manufacture. 

Example 1:
There are 3 designations for the X MIN End stop socket on a BTT Octopus Pro Board.
1. Physical Socket is number J27 per BTT documentation
2. Firmware Pin Designation PG6 also shown in BTT documentation
3. Firmware Pin Description X MIN END STOP, X_MIN, etc

What I suggest is to only call a socket by #2 .. no description. Because it can get confusing trying to cross reference and track #2 & #3 as you start to customize your firmware. Keep the name consistent through out the config and pins files 

Example 2:
If i want configure a specific end stop on a board, Then i will look up #2 per BTT Docs since that is the first thing you will do. Now, when i jump in the firmware i want to say "HOME X TO PG6 (pin #) not X MIN. Similar example when setting up sensorless homing in the Pins Configuration files. Ditch #3 and just call everything by #2.

Now, I understand that Marlin has the option to use Software/Virtual end stops and for that, it needs #3 which would be X Max right? but why does it have to be X Max? why not have it defined by the bed size or a custom max travel value?  

In short, I believe X Min, X Max, FILDET, etc... are not necessary. This goes for ALL of the pins. They should be referred as #2 so that as you go through the config files, everything cross checks easier. I do like that most code lines have comments to describe that code and that is a great. each end stop pin can have a description... BUT....as a comment. i.e  PG6 //TYPICALLY USED AS X_MIN BY MOST PRINTERS

Maybe it's just me.. but I struggle trying to re-assign the pins since my printer a fully custom build and most of the predefined info is not meant for my set up. I have to write down all of the descriptions of the pins and what number they are, Just seams like an unnecessary step. I see Pin PG6 on the board, so that is what I'm expecting to see and look for in Marlin. Instead, its a combination of multiple names for that same pin. A bit frustrating and confusing.. Not to mention, some code lines prefer the value of X_MIN vs the Pin #. One time i wanted to use the FILDET as a Z4 ENDSTOP but Marlin did not recognize FILDET or the PIN #. 

### Additional context

Im totally open for other ideas or to further discuss this!

thisiskeithb, 2024-12-21:
> What I suggest is to only call a socket by https://github.com/MarlinFirmware/Marlin/pull/2 .. no description.

You should always use # 2 to reference a pin since that is the pin on the MCU and what Marlin will be controlling/watching. Sockets can contain other pins and comments / descriptions are just that.

> Now, I understand that Marlin has the option to use Software/Virtual end stops and for that, it needs https://github.com/MarlinFirmware/Marlin/pull/3 which would be X Max right? but why does it have to be X Max? why not have it defined by the bed size or a custom max travel value?

Marlin homes to an axis' min or max travel. You can't home to any endstop/pin on the fly, so that is why the MIN & MAX (or `STOP` if there's only a single endstop pin) pins are defined in the pins file.

> Maybe it's just me.. but I struggle trying to re-assign the pins since my printer a fully custom build and most of the predefined info is not meant for my set up.

We try to follow an existing standard, usually by the board OEM (as printed on the PCB), printer OEM (as the printer is built), or some kind of long-standing community setup.

If you are building a fully custom/one-off machine, having to adjust some defaults is expected.

For best results getting help with configuration and troubleshooting, please use the following resources, particularly our Discord server:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.com/servers/marlin-firmware-461605380783472640) Discord server

closed 2024-12-21 by thisiskeithb

sinus19, 2024-12-21:
> You should always use # 2 to reference a pin since that is the pin on the MCU and what Marlin will be controlling/watching. Sockets can contain other pins and comments / descriptions are just [...]

Luckily, i think i have finally figured out a flow process and know what to look for on feature builds and Marlin Versions. But man... it was brutal trying to chase down certain codes. If marlin didnt recognized FILDET as an end stop, i had to go chase down that tag in other files and change some lines so that it would accept it. ok maybe thats not the best example lol. But i had similar situations where i needed to dig deeper because Marlin wanted a Descriptive Pin as a value instead of a Pin Number. Granted, i think that has been fixed in the latest version. 

I fully understand that we are way deep in the defualt method as you mentioned it is what we have been utilizing for a while and mostly everyone is used to it for their common printer. I suppose i was thinking differently, a bit more universal while having #3 more or less as a comment instead. 

> * [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
> * [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
> * [Marlin [...]