MarlinFirmware/Marlin #27602: After homing, babystep is not working on tinybee

moriamoria, 2024-12-23:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Hello,

I have a tinybee motherboard with BLTOUCH and 2209 driver on Z

After G28 homing command, babystep command M290 Z-0.05 is not working, no effect on Z.

If you can help me, I have check my configuration, but I cant' find a reason

[Marlin.zip](https://github.com/user-attachments/files/18234111/Marlin.zip)

I have test with 2.1.3-beta2 and board_SIMULATED, and when I send the command M290 Z1 I see the printer head raise

Many thanks for your help

Regards

### Expected behavior

using M290 Z1 should raise the Z from 1mm and also add 1 to Z offset.

### Actual behavior

no Z motor activities

### Steps to Reproduce

G28
M290 Z1

### Version of Marlin Firmware

bugfix 2.1.X and 2.1.3-beta2

### Printer model

ortur

### Electronics

tinybee

### LCD/Controller

REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2024-12-24:
duplicate of https://github.com/MarlinFirmware/Marlin/issues/27322

closed 2024-12-24 by ellensp

moriamoria, 2024-12-24:
Hello,

Thanks for your answer.

Do you think I should throw away my tinybee motherboard and buy an another processor ?

Regards