MarlinFirmware/Marlin #27603: [Custom Menu: Z Probe Offset]  FLSUN Q5 - Robin_nano_v1v2 UART

yogiunde, 2024-12-25:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

The Custom Menu Item defined in Configuration_adv.h for Z Probe Offset does not work. Defined M28 GCode - 

  #define MAIN_MENU_ITEM_4_DESC "4. Z OffSet Wizard"
  #define MAIN_MENU_ITEM_4_GCODE "G28" //Modif menu_main.cpp(lig158)
  #define MAIN_MENU_ITEM_4_CONFIRM

This works with 2.1.2.5 build. Checked with Bugfix too, does not work.

### Expected behavior

I expect it to run a Z-offset wizard.

### Actual behavior

Homes all axes and nothing further.

### Version of Marlin Firmware

2.1.3-b1

### Printer model

FLSUN Q5 Robin Nano V1

### Electronics

All TMC2209 UART

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Config.zip](https://github.com/user-attachments/files/18243829/Config.zip)

ellensp, 2024-12-25:
these are not marlin provided config files and  they do not conform to 2.1.3-b1 Config format.

ellensp, 2024-12-25:
G28 Has never initiated  Z-offset wizard in Stock Marlin

yogiunde, 2024-12-25:
Appreciate the clarification.
What can I define to initiate the Z-Probe Wizard in place of G28?
FYI This is the Fork - https://github.com/Foxies-CSTL/Marlin_2.1.x/tree/bugfix-2.1.x
May be they changed something about G28.

ellensp, 2024-12-25:
You need to raise an issue on that fork, we cannot support 3rd party forks.

closed 2024-12-25 by ellensp