MarlinFirmware/Marlin #27604: Compensation of surface irregularities along the Z axis ...

Uri-34, 2024-12-25:
### Is your feature request related to a problem? Please describe.

no

### Are you looking for hardware support?

no

### Describe the feature you want

Task:
My surface on which the extrusion will be performed is not smooth.
I need to analyze the surface as the nozzle moves (X,Y) during the extrusion of the material, in front of the nozzle in the direction of its movement.
Depending on this, change the Z value.   

Question:
Is there (or is it possible to embed) a function in the firmware that would be called before each step (by X,Y). 
It would provide an opportunity to receive data about the contents of the queue of commands, as well as about the current coordinate (in X,Y). 
In which I could analyze the received data from my device (a special head) and adjust the Z value during the movement. 
That will help me solve the problem of bending the irregularities in the height of my surface.

### Additional context

I really hope for your answer!

Sincerely, Yuri.

borland1, 2024-12-27:
This is the "Issue Reporting" area of Marlin, not for technical support.  Suggest you close your report.

BTY, I don't understand your  printer's problem description, so this may not be applicable.  But there's an optional feature in Marlin called "Manual Bed Leveling", where you can create a grid of leveling points and manually enter compensation amounts for each grid point to adjust for bed flatness.  During printing, Marlin will apply your compensation amounts at each grid point to adjust for flatness, including interpolation between each grid point.

ellensp, 2024-12-27:
They seem to want real time surface measurement and compensation. for the first layer, vs probing the surface building a mesh of offsets and compensating off of that mesh like everyone else does.

Uri-34, 2024-12-28:
Yes ! 
And so it is ! 
This is compensation without creating a compensation grid ! 
The fact is that it is assumed that some elements of the image will be applied on top of the previously applied ones ! 
Re-creating a compensation grid to solve this problem (from my point of view) is not suitable at all !!!

Sincerely, Yuri.

Uri-34, 2024-12-28:
As one of the options, we can consider dynamically filling the offset grid with data coming from the head (my version) ! 
Then I need an interface to the offset grid matrix ! 
I would like to ask you to advise me on this issue !?

Sincerely, Yuri.

borland1, 2024-12-30:
Here's a link to the Marlin's G-code Reference Guide....

[G29 - Bed Leveling (Manual)](https://marlinfw.org/docs/gcode/G029-mbl.html)

Example to Modify Mesh Points and View Results:

```
> G29 S3 I2 J2 Z0.042
> G29 S3 I1 J1 Z-0.666
> G29 S0
3x3 mesh. Z offset: 0
Measured points:
       0      1       2
0 +0.011 -0.020  -0.026
1 +0.017 -0.666  -0.019
2 +0.022 -0.030  +0.042
```

Uri-34, 2024-12-31:
Can any of those present advise or suggest who could do this on the issue of the logic of working with the offset grid ? 
After getting to know https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.1.x/Marlin/src/feature/bedlevel/ubl and https://marlinfw.org/docs/gcode/M421.html I understand that there is a possibility of solving my problem through dynamic correction of the offset grid !!! 
This is probably not the most optimal way! But on the other hand, it is more logical in terms of the logic of the firmware. 
It would be great if someone could advise and provide a link to the documentation on classes and methods !

Sincerely, Yuri.

Uri-34, 2024-12-31:
After reading this URL description: https://marlinfw.org/docs/features/unified_bed_leveling.html I saw that the maximum number of grid points is 15x15 ! 
Unfortunately, this is too little for my task !!! 
Question: can I increase the grid size to 160x160 without consequences, for example ?

Sincerely, Yuri.

Uri-34, 2025-01-06:
Apparently, colleagues who understand the field I'm asking about have absolutely no time to answer my questions ?! Or it 's not accepted here ! :)
Then maybe someone can provide links to relevant forums where I could get answers ?

Sincerely, Yuri.

ellensp, 2025-01-06:
We are all volunteers, this has to appeal to someones interest before it gets any traction.
Since it involves custom hardware that no on else has, interest will be low.

This is also FR 1 of 544 some of which have been waiting for years... 

Also most of the planet is on break right now with X-Mas and New Years holidays

Uri-34, 2025-01-06:
Thanks for the answer! 
I hope that sooner or later someone will tell me ! 
Correction: I don't need any help writing code or debugging my device ! 
I only need hints on the availability of certain features in the firmware - Marlin !

Sincerely, Yuri.

Uri-34, 2025-01-18:
OK! Apparently, if I just wait for those who want to answer my question to appear, I'll never wait for it ! :))) 
Maybe one of the colleagues present can recommend me a specialist in modifying the Marlin firmware, or at least how it works at a low level ?! 
I would like to receive detailed consultations ! 
If you are interested in payment, I am ready to talk to a specific person to resolve this issue!  

With respect.