MarlinFirmware/Marlin #27606: [BUG] BTT TFT35 E3 and MKS GEN L V1.0

BlackRiverCoder, 2024-12-25:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Hello, I just got BTT TFT35 E3 V3.0.1 and wanted to use it, but I also have OctoPrint server connected to main board and this 2 cannot work together. I wanted to set SERIAL_PORT_2 for TFT display to communicate by TX and RX in AUX-1 port in MKS GEN L V1.0, but I get error when compiling: #error "Serial Port 1 pin D18 and/or D19 conflicts with another pin on the board.", which are pins for Z endstop. I tried PORT 1, 2, 3, but still same error, just another pin numbers. 

Is this wrong board pin configuration in Marlin?

Thanks
[configurations.zip](https://github.com/user-attachments/files/18247373/configurations.zip)

### Bug Timeline

When connected BTT TFT35

### Expected behavior

OoctoPrint to run on default -1 or 0 PORT (USB) and BTT TFT35 to use TX and RX (port 1, 2 or 3) , so they can work both. (set temp on display or in octoprint)

### Actual behavior

Can't do anything by OctoPrint, only by BTT TFT35

### Version of Marlin Firmware

2.1.2.5

### Printer model

KRYAL Cube

### Electronics

MKS GEN L V1.0

### LCD/Controller

BTT TFT35 E3 V3.0.1 (GD version)

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

None

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2024-12-26:
The uart pins on a atmega2560 are fixed ie you cannot move them to other pins
Serial0 on pins 0 (RX) and 1 (TX), Serial1 on pins 19 (RX) and 18 (TX), Serial2 on pins 17 (RX) and 16 (TX), Serial3 on pins 15 (RX) and 14 (TX).
so to use those you need to disable what ever is currently using the required pins

pins 0 (RX) and 1 (TX) on aux-1 are connected to the usb to serial chip for usb
pins 18 and 19 by default are z-min and z-max, so you would need to move the endstops elsewhere.  
pins 16 and 17 by default are on exp1, so conflict with a display.  
pins 14 and 15 by default are y-min and y-max, so to use these you would need to move the endstops else where. 

-1 is not a valid option on this board

closed 2024-12-26 by ellensp

ellensp, 2024-12-26:
you have

#define Y_HOME_DIR +1
#define Z_HOME_DIR -1

so you can easily move your endstops

you want to use uart 1 

then move Z_MIN to Y_MIN plug by adding 
```cpp
#define Z_MIN_PIN 14 // use Y_MIN plug
#define Y_MIN_PIN -1 // disabled
``` 

this will free up uart 1

ellensp, 2024-12-26:
you also have #define USE_ZMAX_PLUG
But what is plugged in there?

BlackRiverCoder, 2024-12-26:
Hi, many thanks, that works :)

Have a nice day