MarlinFirmware/Marlin #27610: [FR] Support for bed MAX31865

anvoice, 2024-12-27:
### Is your feature request related to a problem? Please describe.

The Marlin documentation claims support for a bed max31865 module. Such support does not exist however, and I was hoping to see it happen.

### Are you looking for hardware support?

Fully custom cartesian printer, MKS Tinybee motherboard, max31865 module for bed support.

### Describe the feature you want

Support for max31865 for the bed.

### Additional context

Although the comments in Marlin code claim support exists, it does not seem to. Setting #define TEMP_SENSOR_BED -5 leads to errors, upon fixing which I get more errors. Hoping to see actual support added.

ellensp, 2024-12-28:
This is just config options that are not in standard config files. and I added a sanity check 

```diff
diff --git a/Marlin/Configuration.h b/Marlin/Configuration.h
index 6d5ae49de8..f7ae53b1fd 100644
--- a/Marlin/Configuration.h
+++ b/Marlin/Configuration.h
@@ -599,6 +599,10 @@
   #define MAX31865_SENSOR_OHMS_2      100
   #define MAX31865_CALIBRATION_OHMS_2 430
 #endif
+#if TEMP_SENSOR_IS_MAX_TC(BED)
+  #define MAX31865_SENSOR_OHMS_BED      100
+  #define MAX31865_CALIBRATION_OHMS_BED 430
+#endif

 #if HAS_E_TEMP_SENSOR
   #define TEMP_RESIDENCY_TIME         10  // (seconds) Time to wait for hotend to "settle" in M109
diff --git a/Marlin/Configuration_adv.h b/Marlin/Configuration_adv.h
index 806bafa2d0..99437183d2 100644
--- a/Marlin/Configuration_adv.h
+++ b/Marlin/Configuration_adv.h
@@ -174,9 +174,10 @@
  * Thermocouple Options — for MAX6675 (-2), MAX31855 (-3), and MAX31865 (-5).
  */
 //#define TEMP_SENSOR_FORCE_HW_SPI                // Ignore SCK/MOSI/MISO pins; use CS and the default SPI bus.
-//#define MAX31865_SENSOR_WIRES_0 2               // (2-4) Number of wires for the probe connected to a MAX31865 board.
-//#define MAX31865_SENSOR_WIRES_1 2
-//#define MAX31865_SENSOR_WIRES_2 2
+//#define MAX31865_SENSOR_WIRES_0   2             // (2-4) Number of wires for the probe connected to a MAX31865 board.
+//#define MAX31865_SENSOR_WIRES_1   2
+//#define MAX31865_SENSOR_WIRES_2   2
+//#define MAX31865_SENSOR_WIRES_BED 2

 //#define MAX31865_50HZ_FILTER                    // Use a 50Hz filter instead of the default 60Hz.
 //#define MAX31865_USE_READ_ERROR_DETECTION       // Treat value spikes (20°C delta in under 1s) as read errors.
@@ -188,6 +189,7 @@
 //#define MAX31865_WIRE_OHMS_0              0.95f // For 2-wire, set the wire resistances for more accurate readings.
 //#define MAX31865_WIRE_OHMS_1              0.0f
 //#define MAX31865_WIRE_OHMS_2              0.0f
+//#define MAX31865_WIRE_OHMS_BED            0.0f

 /**
  * Hephestos 2 24V heated bed upgrade kit.
diff --git a/Marlin/src/inc/SanityCheck.h b/Marlin/src/inc/SanityCheck.h
index 9c21c4fd29..b7edc34ade 100644
--- a/Marlin/src/inc/SanityCheck.h
+++ b/Marlin/src/inc/SanityCheck.h
@@ -2121,6 +2121,13 @@ static_assert(NUM_SERVOS <= NUM_SERVO_PLUGS, "NUM_SERVOS (or some servo index) i
     #error "MAX31865_SENSOR_OHMS_2 and MAX31865_CALIBRATION_OHMS_2 must be set if TEMP_SENSOR_2/TEMP_SENSOR_REDUNDANT is MAX31865."
   #endif
 #endif
+#if TEMP_SENSOR_BED_IS_MAX31865
+  #if !defined(MAX31865_SENSOR_WIRES_BED) || !WITHIN(MAX31865_SENSOR_WIRES_BED, 2, 4)
+    #error "MAX31865_SENSOR_WIRES_BED must be defined as an integer between 2 and 4."
+  #elif !defined(MAX31865_SENSOR_OHMS_BED) || !defined(MAX31865_CALIBRATION_OHMS_BED)
+    #error "MAX31865_SENSOR_OHMS_BED and MAX31865_CALIBRATION_OHMS_BED must be set if TEMP_SENSOR_BED is MAX31865."
+  #endif
+#endif

 /**
  * Redundant temperature sensor config
```

ellensp, 2024-12-28:
related PR https://github.com/MarlinFirmware/Marlin/pull/27511

ellensp, 2024-12-28:
PR created to add this to marlin.

anvoice, 2024-12-28:
Thank you, but for my motherboard (MKS tinybee, an esp32-based board), even after adding everything according to your instructions, I get the following error: `Marlin/src/libs/../HAL/shared/Delay.h:220:58: error: 'DELAY_CYCLES_VAR' was not declared in this scope`. Apparently DELAY_CYCLES_VAR is not defined for my platform. I attempted to add it manually in Delay.h:
```
#elif defined(ESP32) || defined(__PLAT_LINUX__) || defined(__PLAT_NATIVE_SIM__)

  // DELAY_CYCLES specified inside platform

  // Delay in microseconds
  #define DELAY_US(x) DELAY_CYCLES((x) * ((F_CPU) / 1000000UL))
  #define DELAY_CYCLES_VAR DELAY_CYCLES                         // attempt to get rid of error
#else

  #error "Unsupported MCU architecture"

#endif
```
but while that compiles, the printer does not respond to commands if the bed temp sensor is defined as the max31865.

ellensp, 2024-12-28:
Is a esp32 specific issue

```diff
diff --git a/Marlin/src/HAL/shared/Delay.h b/Marlin/src/HAL/shared/Delay.h
index a6795a78ea..4751d7a5e2 100644
--- a/Marlin/src/HAL/shared/Delay.h
+++ b/Marlin/src/HAL/shared/Delay.h
@@ -174,6 +174,9 @@ void calibrate_delay_loop();
 
   // Delay in microseconds
   #define DELAY_US(x) DELAY_CYCLES((x) * ((F_CPU) / 1000000UL))
+
+  #define DELAY_CYCLES_VAR DELAY_CYCLES
+
 #else
 
   #error "Unsupported MCU architecture"

```

ellensp, 2024-12-28:
please attach your config files.

ellensp, 2024-12-28:
add   #define DEBUG_MAX31865  to Configuration.h 
It may tell you more over serial

ellensp, 2024-12-28:
Please test the [`bugfix-2.1.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to see where it stands.

anvoice, 2024-12-29:
Apologies for late response, ran into some issues. The bugfix-2.1.x branch results in same behaviour: printer connects but does not communicate. Here are my configuration files and changed source from the 2.1.2.5 branch. I'm not sure I got everything but hope I did: the code with AlexB in the comments was code that I changed. The changes not related to the bed max31865 include an additional of serial3 to allow the TFT screen to communicate with the printer, and I also inverted the behaviour of a fan pin to accommodate the CPAP fan. Please let me know if any other details are needed. One (possible) issue I see with the code is that `BED` is defined as 0 somewhere else in the source, perhaps that is causing a problem. I could be totally wrong though.

I'm currently using the pins on the exp2 connection of the mainboard. The max31865 version I have is a stepstick variant on a breadboard. I tried using it directly in the slot for the unused E1 stepstick but got the same results. The MKS tinybee does not seem to support SPI mode for the steppers due to I think low pin count, so I thought maybe using the unused exp2 connection could be an alternative.
[Configuration_AlexB.zip](https://github.com/user-attachments/files/18270920/Configuration_AlexB.zip)
[changes_AlexB.zip](https://github.com/user-attachments/files/18270921/changes_AlexB.zip)

anvoice, 2025-01-04:
Please let me know if anything else is needed.

anvoice, 2025-01-09:
> add #define DEBUG_MAX31865 to Configuration.h It may tell you more over serial

Forgot to mention, I did try that as well to no avail, no serial output, just an unresponsive (albeit connecgted) printer.

anvoice, 2025-06-22:
Is there any news on this front? My printer build has been on the backburner but I'd like to get back to it. I am not very familiar with C and macro-heavy code, which makes modifying this myself very difficult.

ellensp, 2025-06-22:
the above PR may help,  you have

```cpp
  #define TEMP_BED_MISO_PIN 19                 // AlexB
  #define TEMP_BED_SCK_PIN 18                  // AlexB
  #define TEMP_BED_MOSI_PIN 23                 // AlexB
  #define TEMP_BED_CS_PIN 14                   // AlexB: Custom code: attempt to make stepstick MAX31865 work in place of 
```

but marlin was ignoring all but the CS define.

thinkyhead, 2026-10-08:
Thanks for your patience! The missing bed MAX31865 options and the ESP32 `DELAY_CYCLES_VAR` build error are fixed in bugfix-2.1.x (#27611), and #27938 fixed Marlin ignoring the TEMP_BED_MISO/SCK/MOSI pins. A TinyBee build with TEMP_SENSOR_BED -5 now compiles cleanly. Since pins 19/18/23 on EXP2 are the same hardware SPI bus the onboard SD card uses, please define only `TEMP_BED_CS_PIN 14` (or enable `TEMP_SENSOR_FORCE_HW_SPI`) so the sensor shares that bus instead of bit-banging it. Then test a clean current bugfix-2.1.x with `DEBUG_MAX31865` enabled and post the serial boot log, plus whether it still hangs with SDSUPPORT disabled.

ellensp, 2026-11-04:
I tested this on an MKS TinyBee with a real MAX31865 module and a 3-wire PT100, wired to the same EXP2 pins you used (MISO 19, SCK 18, MOSI 23, CS 14). It works, and I think I know why your printer stopped responding.

**Why it stopped responding**

Pins 18, 19 and 23 are also the TinyBee's SD card SPI bus. With `TEMP_BED_MISO_PIN`, `TEMP_BED_SCK_PIN` and `TEMP_BED_MOSI_PIN` defined and SD support on, Marlin tries to drive the sensor in software on pins the SD card's hardware SPI has already taken. The sensor then reads as a fault, Marlin stops with `MINTEMP`, and the printer looks connected but doesn't answer. I got exactly this on my board. With SD support off, the same setup read fine.

**What to do**

Keep the same wiring, but let the sensor share the SD card's bus. In your pins or configuration, remove these three lines and keep only the CS pin:

```cpp
//#define TEMP_BED_MISO_PIN 19
//#define TEMP_BED_SCK_PIN 18
//#define TEMP_BED_MOSI_PIN 23
#define TEMP_BED_CS_PIN 14
```

Then set these to match your sensor, the wires in `Configuration_adv.h` and the rest in `Configuration.h`:

```cpp
#define TEMP_SENSOR_BED -5
#define MAX31865_SENSOR_OHMS_BED      100   // 100 for a PT100
#define MAX31865_CALIBRATION_OHMS_BED 430   // The reference resistor on your module
#define MAX31865_SENSOR_WIRES_BED 3         // 2, 3 or 4, matching your sensor
```

On a module like mine, 3-wire or 4-wire sensors may also need its solder jumpers set to match.

GPIO 14 is also `BTN_EN1` when a wired LCD is enabled on this board, so if you use one, pick a different free pin for CS.

**What I saw on my board**

- About 19.4 °C at room temperature, 26.6 °C held in my fingers, then falling steadily after letting go.
- A 1000-line SD print ran in order while the sensor was being read.
- With one sensor wire pulled out, Marlin stopped with `MAXTEMP`, which is what you want.

I haven't heated anything with it.

**Serial errors you may see**

On current `bugfix-2.1.x`, this setup prints `gpio_set_level(226): GPIO output gpio_num error` lines on the serial port about 4 times a second. The sensor still reads correctly, and commands still got through in my test. #28797 fixes that. It also turns the setup that caused your problem into a clear build error, so nobody else has to guess.

If you try it, please let us know how it goes.

closed 2026-11-06 by ellensp