MarlinFirmware/Marlin #27613: [FR] Creality Ender 3 S1 Plus Touch Screen Support

AlexanderPirvu, 2024-12-29:
### Is your feature request related to a problem? Please describe.

Marlin (as far as I can tell/search) does not have support for the touch screen that comes with the Ender 3 S1 Plus

### Are you looking for hardware support?

Yes, the touch screen [Creality firmware here](https://github.com/CrealityOfficial/Ender-3S1/tree/3S1_Plus_Screen) is not currently supported

### Describe the feature you want

I would like to use Marlin on the S1 Plus with the touch screen if possible

### Additional context

Let me know if there is any way I can help :)

ThomasToka, 2024-12-29:
Hello, 
while the pro is „basically supported“ up to Crealities F26 release. The Plus afaik lacks support in upstream marlin.
You can have look in my fork which supports the Plus fully and has extended capabilities. 
https://github.com/ThomasToka/MarlinFirmware

thisiskeithb, 2024-12-29:
@ThomasToka: We look forward to some PRs from you to bring support for this printer & TFT upstream :)

ThomasToka, 2024-12-29:
Meanwhile the problem is that my whole Marlin is dynamic and i am so far away from upstream with all the bells and whistles… and naturally with the screen also. But i will try to „generalize“ my dynamic functions in the next months. So that all the dynamic things are in one dynamic config file and could be included by other implementations on demand. And then maybe PR it again..

AlexanderPirvu, 2024-12-29:
Awesome stuff! Thanks guys :)