MarlinFirmware/Marlin #27623: [BUG] Cannot compile the firmware when using floats in X2_MAX_POS and/or X2_HOME_POS on DUAL_X_CARRIAGE setup

Queldaras, 2025-01-05:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Marlin cannot compile when using floats in X2_MAX_POS and/or X2_HOME_POS on DUAL_X_CARRIAGE setup. It gives the following errors. If I use an integer it compiles just fine, and then I have to go to tool offsets and change the value from the LCD screen to the appropriate accurate value. This still exists in the latest bugfix, but doesn't exist in the version 2.1.2.5 that I currently used before I installed the 2.1.3-b1 

The compile time error:

Compiling .pio\build\BTT_SKR_PRO\src\src\HAL\STM32\MarlinSPI.cpp.o
In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfigPre-4-adv.h:33,
                 from Marlin\src\HAL\STM32\../../inc/Conditionals-4-adv.h:32,
                 from Marlin\src\HAL\STM32\../../inc/MarlinConfigPre.h:30,
                 from Marlin\src\HAL\STM32\../../inc/MarlinConfigPre-5-post.h:24,
                 from Marlin\src\HAL\STM32\../../inc/MarlinConfigPre-6-type.h:24,
                 from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:28,
                 from Marlin\src\HAL\STM32\HAL.cpp:26:
Marlin\src\HAL\STM32\../../inc/../../Configuration_adv.h:828:24: error: floating constant in preprocessor expression
  828 |   #define X2_MAX_POS   288.80        // The max position of the X2 carriage, typically also the home position
      |                        ^~~~~~
Marlin\src\HAL\STM32\../../inc/../../Configuration_adv.h:829:23: note: in expansion of macro 'X2_MAX_POS'
  829 |   #define X2_HOME_POS X2_MAX_POS  // Default X2 home position. Set to X2_MAX_POS.
      |                       ^~~~~~~~~~
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:1938:10: note: in expansion of macro 'X2_HOME_POS'
 1938 |   #elif (X2_HOME_POS <= X1_MAX_POS) || (X2_MAX_POS < X1_MAX_POS)
      |          ^~~~~~~~~~~
Marlin\src\HAL\STM32\../../inc/../../Configuration_adv.h:828:24: error: floating constant in preprocessor expression
  828 |   #define X2_MAX_POS   288.80        // The max position of the X2 carriage, typically also the home position
      |                        ^~~~~~
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:1938:41: note: in expansion of macro 'X2_MAX_POS'
 1938 |   #elif (X2_HOME_POS <= X1_MAX_POS) || (X2_MAX_POS < X1_MAX_POS)
      |                                         ^~~~~~~~~~
In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:49,
                 from Marlin\src\HAL\STM32\HAL.cpp:26:
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:1939:6: error: #error "DUAL_X_CARRIAGE will crash if X1 can meet or exceed X2 travel."
 1939 |     #error "DUAL_X_CARRIAGE will crash if X1 can meet or exceed X2 travel."
      |      ^~~~~
In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfigPre-4-adv.h:33,
                 from Marlin\src\HAL\STM32\../../inc/Conditionals-4-adv.h:32,
                 from Marlin\src\HAL\STM32\../../inc/MarlinConfigPre.h:30,
                 from Marlin\src\HAL\STM32\../../inc/MarlinConfigPre-5-post.h:24,
                 from Marlin\src\HAL\STM32\../../inc/MarlinConfigPre-6-type.h:24,
                 from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:28,
                 from Marlin\src\HAL\STM32\HAL_SPI.cpp:26:
Marlin\src\HAL\STM32\../../inc/../../Configuration_adv.h:828:24: error: floating constant in preprocessor expression
  828 |   #define X2_MAX_POS   288.80        // The max position of the X2 carriage, typically also the home position
      |                        ^~~~~~
Marlin\src\HAL\STM32\../../inc/../../Configuration_adv.h:829:23: note: in expansion of macro 'X2_MAX_POS'
  829 |   #define X2_HOME_POS X2_MAX_POS  // Default X2 home position. Set to X2_MAX_POS.
      |                       ^~~~~~~~~~
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:1938:10: note: in expansion of macro 'X2_HOME_POS'
 1938 |   #elif (X2_HOME_POS <= X1_MAX_POS) || (X2_MAX_POS < X1_MAX_POS)
      |          ^~~~~~~~~~~
Marlin\src\HAL\STM32\../../inc/../../Configuration_adv.h:828:24: error: floating constant in preprocessor expression
  828 |   #define X2_MAX_POS   288.80        // The max position of the X2 carriage, typically also the home position
      |                        ^~~~~~
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:1938:41: note: in expansion of macro 'X2_MAX_POS'
 1938 |   #elif (X2_HOME_POS <= X1_MAX_POS) || (X2_MAX_POS < X1_MAX_POS)
      |                                         ^~~~~~~~~~
In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:49,
                 from Marlin\src\HAL\STM32\HAL_SPI.cpp:26:
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:1939:6: error: #error "DUAL_X_CARRIAGE will crash if X1 can meet or exceed X2 travel."
 1939 |     #error "DUAL_X_CARRIAGE will crash if X1 can meet or exceed X2 travel."
      |      ^~~~~
In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfigPre-4-adv.h:33,
                 from Marlin\src\HAL\STM32\../../inc/Conditionals-4-adv.h:32,
                 from Marlin\src\HAL\STM32\../../inc/MarlinConfigPre.h:30,
                 from Marlin\src\HAL\STM32\../../inc/MarlinConfigPre-5-post.h:24,
                 from Marlin\src\HAL\STM32\../../inc/MarlinConfigPre-6-type.h:24,
                 from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:28,
                 from Marlin\src\HAL\STM32\HardwareSerial.cpp:32:
Marlin\src\HAL\STM32\../../inc/../../Configuration_adv.h:828:24: error: floating constant in preprocessor expression
  828 |   #define X2_MAX_POS   288.80        // The max position of the X2 carriage, typically also the home position
      |                        ^~~~~~
Marlin\src\HAL\STM32\../../inc/../../Configuration_adv.h:829:23: note: in expansion of macro 'X2_MAX_POS'
  829 |   #define X2_HOME_POS X2_MAX_POS  // Default X2 home position. Set to X2_MAX_POS.
      |                       ^~~~~~~~~~
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:1938:10: note: in expansion of macro 'X2_HOME_POS'
 1938 |   #elif (X2_HOME_POS <= X1_MAX_POS) || (X2_MAX_POS < X1_MAX_POS)
      |          ^~~~~~~~~~~
Marlin\src\HAL\STM32\../../inc/../../Configuration_adv.h:828:24: error: floating constant in preprocessor expression
  828 |   #define X2_MAX_POS   288.80        // The max position of the X2 carriage, typically also the home position
      |                        ^~~~~~
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:1938:41: note: in expansion of macro 'X2_MAX_POS'
 1938 |   #elif (X2_HOME_POS <= X1_MAX_POS) || (X2_MAX_POS < X1_MAX_POS)
      |                                         ^~~~~~~~~~
In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:49,
                 from Marlin\src\HAL\STM32\HardwareSerial.cpp:32:
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:1939:6: error: #error "DUAL_X_CARRIAGE will crash if X1 can meet or exceed X2 travel."
      |      ^~~~~
*** [.pio\build\BTT_SKR_PRO\src\src\HAL\STM32\HAL_SPI.cpp.o] Error 1
*** [.pio\build\BTT_SKR_PRO\src\src\HAL\STM32\HAL.cpp.o] Error 1
*** [.pio\build\BTT_SKR_PRO\src\src\HAL\STM32\HardwareSerial.cpp.o] Error 1

### Bug Timeline

New bug. Started since version 2.1.3-b1 and still exists in latest bugfix

### Expected behavior

I expected to be able to compile the firmware after using floats in X2_MAX_POS and/or X2_HOME_POS on DUAL_X_CARRIAGE setup

### Actual behavior

I cannot compile the firmware when using floats in X2_MAX_POS and/or X2_HOME_POS on DUAL_X_CARRIAGE setup

### Steps to Reproduce

1. Download and extract the version 2.1.3-b1 or the latest bugfix release.
2. Replace the configuration files with the ones that I currently attach in this report.
[configuration.zip](https://github.com/user-attachments/files/18310611/configuration.zip)
3. Try to compile using VSCode, PlatformIO and AutoBuildMarlin

### Version of Marlin Firmware

2.1.3-b1 and the latest bugfix

### Printer model

Anet A6 modded as a dual X carriage printer.

### Electronics

BTT SKR PRO V1.2

### LCD/Controller

BTT LCD Touch Screen V3

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

rondlh, 2025-01-06:
I can confirm this issue, it's caused by some sanity checks in the form of compiler directives that cannot handle floats.
The easiest way to work around this is to just comment out the relevant lines in SanityCheck.h

Comment out the following lines:
```
  #elif (X2_HOME_POS <= X1_MAX_POS) || (X2_MAX_POS < X1_MAX_POS)
    #error "DUAL_X_CARRIAGE will crash if X1 can meet or exceed X2 travel."

```

Queldaras, 2025-01-06:
My workaround was to just not use a float in this variable. So the firmware compiled just fine.
Then I installed the firmware on the printer and changed the tool offset from the LCD screen to the appropriate value and saved the changes to EEPROM.

rondlh, 2025-01-07:
Great job, that will do the trick. I think this topic can be closed...

closed 2025-04-23 by github-actions[bot]

thisiskeithb, 2025-04-23:
Fixed in https://github.com/MarlinFirmware/Marlin/pull/27701