MarlinFirmware/Marlin #27624: [BUG] X_MIN_ENDSTOP_HIT_STATE LOW without any other result than HIGH

smarthomenewbee81, 2025-01-05:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

My endstops are NC, so I configure #define X_MIN_ENDSTOP_HIT_STATE LOW, but it's still the same result as HIGH.
Same for other axis Y.

### Bug Timeline

bug

### Expected behavior

I would expect #define X_MIN_ENDSTOP_HIT_STATE LOW is the same as old X_MIN_ENDSTOP_INVERTING TRUE

### Actual behavior

Homing does not work, due too printer always think axis is at point zero.
Display: "Homing Failed: / PRINTER HALTED / Please Reset"

### Steps to Reproduce

compile marlin with both configs:
#define X_MIN_ENDSTOP_HIT_STATE LOW &
#define X_MIN_ENDSTOP_HIT_STATE HIGH

Both will be the same result.

### Version of Marlin Firmware

2.1.3 Beta 1

### Printer model

Ender 3 Pro

### Electronics

SKR MINI E3 V3.0

### LCD/Controller

CR10 stock

### Other add-ons

BL touch

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin.zip](https://github.com/user-attachments/files/18311076/Marlin.zip)

ellensp, 2025-01-05:
Do some basic diagnosis

Use M119 to see endstop status   https://marlinfw.org/docs/gcode/M119.html

Or enable LCD_ENDSTOP_TEST and watch them on the LCD

My guess is you simply have the endstops plugged into the wrong ports

ellensp, 2025-01-05:
Tested on my hardware
X_MIN_ENDSTOP_HIT_STATE and Y_MIN_ENDSTOP_HIT_STATE  are wroking as expected

closed 2025-01-05 by ellensp

ellensp, 2025-01-05:
Your initial statement "My endstops are NC, so I configure #define X_MIN_ENDSTOP_HIT_STATE LOW" is wrong

NC is the switch is normally closed, when switch is not activated
The BTT E3 SKR MINI V3 endstop has two pins. GND and SIGNAL. Closed keeps the signal IO line connected to GND, ie LOW
When the switch is pressed the connection to GND is cut  and the signal is pulled HIGH by pullup resistor. 

IE NC endstops should be configured as {X|Y|Z}_MIN_ENDSTOP_HIT_STATE HIGH

smarthomenewbee81, 2025-01-06:
There were some errors in my wiring to fix:
1. Y-axis endstop has other pinning than X-axis endstop, so LCD_ENDSTOP_TEST was the hint I needed. This is line is not available in Marlin documentation.
2. Mainboard seemed to have a problem with common ground, so I removed it from the metal bracket.

And now: I works
Thanks @ellensp