MarlinFirmware/Marlin #27625: [BUG] Settings loaded CRC message is broken

rondlh, 2025-01-06:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

The message when load settings is broken:

`echo:V90 stored settings retrieved (952 bytes; crc 6475841`

The CRC is a bit long for a 16-bit CRC, that is because the 41 at the end is the character ')'.

In settings.cpp
`        DEBUG_ECHOLN(version_str, F(" stored settings retrieved ("), eeprom_total, F(" bytes; crc "), working_crc, ')');
`

That should probably be:
`        DEBUG_ECHOLN(version_str, F(" stored settings retrieved ("), eeprom_total, F(" bytes; crc "), working_crc, ")");
`

### Bug Timeline

At least 3 months ago

### Expected behavior

Correct CRC message.

### Actual behavior

Wrong CRC message.

### Steps to Reproduce

Just check the message when setting are loaded at restart, or run M501

### Version of Marlin Firmware

Lastest bugfix

### Printer model

Custom

### Electronics

STM32H7xx based custom motherboard

### LCD/Controller

BTT TFT35 V3.0

### Other add-ons

CAN toolhead (EBB42 V1.2)

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

Not relevant, but here are the configuration files.
[Configuration.zip](https://github.com/user-attachments/files/18318390/Configuration.zip)

ellensp, 2025-01-10:
this was changed in https://github.com/MarlinFirmware/Marlin/pull/27199 
https://github.com/MarlinFirmware/Marlin/pull/27199/commits/6cf923e25bc0edf6615af4a08cd095a6157c9646

Yes it should be fixed.

ellensp, 2025-01-10:
PR created, closing issue

closed 2025-01-10 by ellensp