MarlinFirmware/Marlin #27626: [BUG] [Bedleveling] Z dont rise on deploy with fixed probe and homed to max

KillerInk, 2025-01-06:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Hi
My setup on Z have max endstop and on min side an inductive sensor as FIX_MOUNTED_PROBE.
It homes to max side because its on bottom. No good idea to home to min after a print is done^^
After a G28 and everything is homed, i do a G29 to start the bedleveling.

`
>>> G29  X0.00 Y0.00 Z122.00
Machine Type: Cartesian
Probe: FIX_MOUNTED_PROBE
Probe Offset X0.00 Y-80.00 Z0.00 (Middle-Front & Same Z as Nozzle)
Auto Bed Leveling: BILINEAR (disabled)
remember_feedrate_scaling_off: fr=66.67 100%
>>> set_bed_leveling_enabled  X0.00 Y0.00 Z122.00
<<< set_bed_leveling_enabled  X0.00 Y0.00 Z122.00
  current_position= X0.00 Y0.00 Z122.00 : Probe::set_deployed
deploy=1 no_return=0
Raise Z to 10.00
do_z_clearance(10.00 [122.00 to 10.00], 0)
`

at that point the bed should start to rise, but it do not^^

after looking through the code i ended up here

https://github.com/MarlinFirmware/Marlin/blob/4eb8a871930ef152e4fa73fc1782e2ed22348c85/Marlin/src/module/probe.cpp#L526-L530

wich brought me there

https://github.com/MarlinFirmware/Marlin/blob/4eb8a871930ef152e4fa73fc1782e2ed22348c85/Marlin/src/module/motion.cpp#L1298

so we start at 122 and the bed should go to 10. but lower_allowed is by default false it returns silent and z never rise.

changing the code inside the probe line 529 to
`
do_z_clearance(zdest,true,true); 
`
rises Z finaly.

its equal if i use Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN or Z_MIN_PROBE_PIN, acts both the same.
also that affects every auto bedleveling

due the fact that i have no idea, if that is the correct way, nor how that affects other probes, i opend that issue and not a pull request.

if you wonder why the slow probs triggered, it hit the max endstops^^

happy new year and thx for marlin. it never went boring with it^^

[marlin_probe log.txt](https://github.com/user-attachments/files/18319770/marlin_probe.log.txt)
[Configurations.zip](https://github.com/user-attachments/files/18319836/Configurations.zip)

### Bug Timeline

I dont know 

### Version of Marlin Firmware

latest bugfix-2.1.x

### Printer model

be3D DeeGreen(only orginal are the axis and the case)

### Electronics

mks tinybee tmc2209-standalone

### LCD/Controller

no lcd

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

None

### Host Software

Pronterface

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

Danez-DK, 2025-01-07:
I can confirm this is a problem with anycubic kossel plus (Marlin-2.1.3-b1) using BD Sensor.

KillerInk, 2025-04-12:
push