MarlinFirmware/Marlin #27627: Marlin compilation problem

AleSan78, 2025-01-06:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Marlin compilation problem, I have this error   "arm-none-eabi-g++: error: D:/3D/Artillery Sidewinder X2/Firmware/Carte Mère/Marlin-release-2.1.3-beta2/buildroot/share/PlatformIO/variants/MARLIN_ARTILLERY_RUBY/ldscript.ld: No such file or directory
*** [.pio\build\Artillery_Ruby\firmware.elf] Error 1"

While when I compile version Marlin 2.1.2.4 it works

### Version of Marlin Firmware

Marlin-release-2.1.3-beta2

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2025-01-06:
you cannot have non English characters such as 'è' in the path, it breaks the linker. this is outside of marlin control

AleSan78, 2025-01-08:
> vous ne pouvez pas avoir de caractères non anglais tels que 'è' dans le chemin, cela casse l’éditeur de liens. C’est hors du contrôle de Marlin

But it works when I compile the old version of marlin 2.1.2.4 with the same path which has the accent "è"

ellensp, 2025-01-08:
confirmed.. It is path related though  moving to D:\Marlin and it builds

AleSan78, 2025-01-10:
> confirmed.. It is path moving to D:\Marlin and it builds
> 
> It is the è character, what I haven't worked out is what 2.1.2.4 did differently, lots has changed...

This worked, thanks, but strange that with the old version of marlin it worked.

closed 2026-09-28 by thinkyhead