MarlinFirmware/Marlin #27636: [BUG] Marlin Simulator SD Card does not work.

alexgrach, 2025-01-11:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

SD card emulation does not work. I've tried to use existing fs.img and generating empty in sim. No success.
[image]

### Expected behavior

SD icon become active.

### Actual behavior

SD icon is inactive.

### Version of Marlin Firmware

bugfix-2.1.x

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration.zip](https://github.com/user-attachments/files/18385819/Configuration.zip)

p3p, 2025-02-03:
This is caused by a conflict in the default simulation pins file,
```
#define EXP2_08_PIN                           41  // KILL
...
#if ANY(TFT_COLOR_UI, TFT_CLASSIC_UI, TFT_LVGL_UI)
...
  #define SD_DETECT_PIN                       41
...
#ifndef KILL_PIN
  #define KILL_PIN                   EXP2_08_PIN
#endif
```

ellensp, 2025-02-04:
If you set #define KILL_PIN                   11  there is no conflict, and it still fails to work. 

Under linus something very odd is going on.  
If you click on SD Card present then it stops taking gcodes

thisiskeithb, 2025-02-04:
On the topic of SD card issues in the Sim...

I found that I cannot copy `TFT_LVGL_UI` assets to `fs.img` because it's too small. Assets are ~5MB and the image is 100MB, so that doesn't add up:

[image]

p3p, 2025-02-04:
@ellensp  I can't reproduce that, once I set the KILL_PIN to resolve the conflict the sd card woks as expected on linux, when there is still a conflict clicking SD Card Present will cause Marlin to stop responding as it triggers kill.

@thisiskeithb at least for me my OS (Linux Mint) mounts the image as read only unless I manually mount it writable from the command line.

ellensp, 2025-02-04:
Seems to have been a corrupted sdcard image

ellensp, 2025-02-04:
Pr created, closing

closed 2025-02-04 by ellensp