MarlinFirmware/Marlin #27638: [BUG] (bug summary)The hotend fan is not working

Sevich07, 2025-01-12:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

The mks Tinybee v 1.0 firmware was recorded without any problems. When the printer is turned on, the hotend fan does not work. Even when the extruder reaches the required 50 degrees Celsius, the fan power lamp lights up on the board. I connect a fan that is obviously serviceable. The firmware files are attached. 

Please help me figure out what the problem is.

### Expected behavior

Я включал и отключал #define E0_AUTO_FAN_PIN -1, Менял  на #define E0_AUTO_FAN_PIN FAN_PIN, перепрошивал. Проблема не ушла
[Configuration_adv.zip](https://github.com/user-attachments/files/18391124/Configuration_adv.zip)

### Version of Marlin Firmware

2.0.9.2

### Bed Leveling

None

### Your Slicer

Cura

### Host Software

Repetier Host

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

thisiskeithb, 2025-01-13:
> Version of Marlin Firmware
> 2.0.9.2

Please download [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code and let us know if you're still having this issue.

ellensp, 2025-01-13:
In Marlin the FAN_PIN is always the part cooling fan controlled by gcodes M106 and M107

You CANNOT set E0_AUTO_FAN_PIN to FAN_PIN.  As that cause two conflicting systems to try and use the same FAN at teh same time.

closed 2025-01-13 by ellensp