MarlinFirmware/Marlin #27641: [BUG] Extremely slow movement when ussing TB6560 & TB6600 driver types

PanosPetrou, 2025-01-14:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I have a mostly homemade printer using an MKS TinyBee board and MKS SERVO42C motors on X & Y axes. It was running an older version of Marlin bugfix 2.1x. The motors were skipping steps from time to time but worked correctly without issues when I set #define X_DRIVER_TYPE & #define Y_DRIVER_TYPE to TB6560. So far so good, I guess they needed a slow driver type because they have a weird closed loop system.

However about a week ago I decided to upload the latest Marlin bugfix to the printer and noticed the X & Y axes moving extremely slow.
The same thing happens if I set #define X_DRIVER_TYPE & #define Y_DRIVER_TYPE to TB6560 and/or TB6600, but the motors work at normal speed if I set them to anything else (for example A4988 or DRV8825) but they skipped steps again on one occasion, so we are back to skipping steps again.

Did you change something regarding stepper movement in the latest bugfix?
I will attach my configuration files and few files I edited in a zip file bellow.
I also have a backup of a fully working previous 2.1x bugfix version (I keep a backup of a previous version when I'm updating the printer, it's a habit of mine), but I can't attach it due to size (411MB).

[Talos_Configuration.zip](https://github.com/user-attachments/files/18403603/Talos_Configuration.zip)

### Bug Timeline

It started with the latest bugfix which I tested a week ago

### Expected behavior

I expected no issues when using: 
#define X_DRIVER_TYPE TB6560
#define Y_DRIVER_TYPE TB6560

### Actual behavior

Extremely slow movement, perhaps less than 1 millimeter per second

### Steps to Reproduce

1. Use an MKS TinyBee board and MKS SERVO42C motors on X & Y axes.
2. Set #define X_DRIVER_TYPE TB6560 & #define Y_DRIVER_TYPE TB6560
3. Upload the latest Marlin Bugfix to the board

### Version of Marlin Firmware

Latest Marlin Bugfix 2.1x

### Printer model

Talos, a homemade printer based on an aluminum frame & bed of a Geeetech A10M

### Electronics

MKS TinyBee

### LCD/Controller

MKS Mini 12864 V3

### Other add-ons

MKS SERVO42C closed loop motors on X & Y axes

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2025-01-14:
TB6560 are extremely slow devices

the driver type selects the timing presets

for TB6560

    #define MINIMUM_STEPPER_POST_DIR_DELAY 15000
    #define MINIMUM_STEPPER_PULSE_NS 30000
    #define MAXIMUM_STEPPER_RATE 15000

vs A4988

    #define MINIMUM_STEPPER_POST_DIR_DELAY 200
    #define MINIMUM_STEPPER_PULSE_NS 1000
    #define MAXIMUM_STEPPER_RATE 500000

All of which can be over ridden in Configuration_adv.h

PanosPetrou, 2025-01-14:
@ellensp yes, I'm aware that TB6560 are slow, that's precisely why I set  X_DRIVER_TYPE & Y_DRIVER_TYPE to TB6560. Apparently MKS SERVO42C motors are using an extremely slow controller that gets overwhelmed very easily, however I didn't have a problem with that setting when I was using an older version of Marlin Bugfix 2.1x. I could use any driver type I wanted, just the fast ones were causing skipped steps from time to time. 

The timestamps of they files say March 20th 2024, so the working bugfix version is a bit after Marlin 2.1.2.2 release.